evcc-io/evcc #224: Solaredge Inverter kann über SunSpec nicht ausgelesen werden.

schenlap, 2020-07-19:
**Describe the bug**
Inverter wird scheinbar nicht als SunSpec erkannt.

**To Reproduce**
Steps to reproduce the behavior:
1. Solaredge Inverter benötigt
```
- name: pvse1 # pv Ertrag from solaredge via modbus
  type: default
  power:
    type: modbus
    uri: 10.0.0.207:502
    id: 1
```
ergibt:
```
./evcc -l trace | grep modb
[modb] TRACE 2020/07/19 08:48:57 Measurement(0): 0.00
```

**Expected behavior**
es sollten mehr als 200W angegeben werden

**EVCC details:**
Show output of `evcc -v`
`evcc version dev (HEAD)`

Show evcc log output with `--log trace`

Wenn ich SunSpec als Model angebe bekomme ich einen Trace.
```
- name: pvse1 # pv Ertrag from solaredge via modbus
  type: default
  power:
    type: modbus
    uri: 10.0.0.207:502
    id: 1
    model: SunSpec
```
```
./evcc_pi -l trace | grep modb
[modb] TRACE 2020/07/19 08:51:08 modbus: send 00 01 00 00 00 06 01 03 9c 40 00 02
[modb] TRACE 2020/07/19 08:51:08 modbus: recv 00 01 00 00 00 07 01 03 04 53 75 6e 53
[modb] TRACE 2020/07/19 08:51:08 modbus: send 00 02 00 00 00 06 01 03 9c 42 00 02
[modb] TRACE 2020/07/19 08:51:08 modbus: recv 00 02 00 00 00 07 01 03 04 00 01 00 41
[modb] TRACE 2020/07/19 08:51:08 modbus: send 00 03 00 00 00 06 01 03 9c 85 00 02
[modb] TRACE 2020/07/19 08:51:08 modbus: recv 00 03 00 00 00 07 01 03 04 00 65 00 32
[modb] TRACE 2020/07/19 08:51:08 modbus: send 00 04 00 00 00 06 01 03 9c b9 00 02
[modb] TRACE 2020/07/19 08:51:08 modbus: recv 00 04 00 00 00 07 01 03 04 ff ff 00 00
[modb] TRACE 2020/07/19 08:51:08 modbus: send 00 05 00 00 00 06 01 03 9c 44 00 42
[modb] TRACE 2020/07/19 08:51:08 modbus: recv 00 05 00 00 00 87 01 03 84 53 6f 6c 61 72 45 64 67 65 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 53 45 32 30 30 30 4d 2d 52 57 4b 30 32 42 4e 4e 34 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 30 30 30 34 2e 30 30 30 39 2e 30 30 33 33 00 00 37 43 30 30 31 35 31 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 65
```

ev. auch interessant, die Ausgabe von mbmb:
`./mbmd inspect -d SOLAREDGE:1@10.0.0.207:502`
```
2020/07/19 08:46:43 config: creating TCP connection for 10.0.0.207:502
--------- Model 1 common ---------
Mn         SolarEdge     string
Md   SE2000M-RWK02BNN    string
Opt                      string
Vr           0004.000    string
SN           7C00151A    string
DA               1.00    uint16
Pad            0x0065    pad
--------- Model 101 inverter ---------
A                    2.10    uint16
AphA                 2.10    uint16
AphB                  n/a    uint16
AphC                  n/a    uint16
A_SF                   -2    sunssf
PPVphAB            231.30    uint16
PPVphBC               n/a    uint16
PPVphCA               n/a    uint16
PhVphA                n/a    uint16
PhVphB                n/a    uint16
PhVphC                n/a    uint16
V_SF                   -1    sunssf
W                  292.95    int16
W_SF                   -2    sunssf
Hz                  49.98    uint16
Hz_SF                  -3    sunssf
VA                 488.30    int16
VA_SF                  -1    sunssf
VAr                390.60    int16
VAr_SF                 -1    sunssf
PF                  59.99    int16
PF_SF                  -2    sunssf
WH             2614862.00    acc32
WH_SF                   0    sunssf
DCA                  4.25    uint16
DCA_SF                 -3    sunssf
DCV                 69.90    uint16
DCV_SF                 -1    sunssf
DCW                297.41    int16
DCW_SF                 -2    sunssf
TmpCab                n/a    int16
TmpSnk              46.39    int16
TmpTrns               n/a    int16
TmpOt                 n/a    int16
Tmp_SF                 -2    sunssf
St                      4    enum16
StVnd                   0    enum16
Evt1                  n/a    bitfield32
Evt2                  n/a    bitfield32
EvtVnd1        0x00000000    bitfield32
EvtVnd2               n/a    bitfield32
EvtVnd3               n/a    bitfield32
EvtVnd4        0x00000000    bitfield32

```

Ich denke dass hier der Wechselrichter nicht ganz konform ist und daher nicht richtig erkannt wird.
`mbmd run` liefert einen Fehler, da werde ich auch noch ein Ticket öffnen.

premultiply, 2020-07-19:
Müsste die ID in der Config dann nicht 101 sein? 🤔

schenlap, 2020-07-19:
ja, das habe ich probiert, mit selben Ergebnis. Da es mit id=1 aber mit mbmd geht, hatte ich gedacht das paßt so.
Als modbus ID habe ich im Wechselrichter 1 eingestellt. Was da jetzt wirklich hingehört kann ich nicht sagen, es geht jedenfalls mit Beiden nicht.

premultiply, 2020-07-19:
Hier gibt es auch noch ein paar Tipps und Bugs wohl dazu:

https://github.com/jbuehl/solaredge/issues/80

Vielleicht hilft's ja.

andig, 2020-07-19:
Gleiches Problem wie #207? Du hast keinen konfiguriert, den Du auslesen willst 😳

andig, 2020-07-19:
Da fehlt noch Default oder Fehlermeldung.

andig, 2020-07-19:
Fixed in 985a16a91d681da4b54bad066e7149c260e105ed

closed 2020-07-19 by andig