evcc-io/evcc #225: Zielladen und dynamische Stromtarife

premultiply, 2020-07-19:
**Zielladen** stellt sicher, dass zu einem vom Nutzer vorgegebenen Zeitpunkt die vorgegebene Energiemenge mindestens im Fahrakku bereitsteht.

Ein typisches Anwendungszenario dürfte es z. B. sein Montagmorgens um 7:00 Uhr für den Weg zur Arbeit ein sicher ausreichend für den geplanten Tagesbedarf geladenes Fahrzeug anzutreffen.

Dies wird so implementiert, dass erst zum spätmöglichsten Zeitpunkt (dieser wird permanent anhand der aktuellen SOC-Info und max. Ladeleistungsvorgabe inkl. Reserven für spezielle Ladecharakteristik neu berechnet) bis zur vorgegebenen Mindestmenge mit maximaler Leistung nachgeladen wird. Im Zeitraum davor wird selbstverständlich anfallender Überschuss auch schon geladen.

-     Im Idealfall wird also überhaupt nicht mehr gesondert (nach)geladen, da Ziel-SOC schon durch Überschuss erreicht.
-     Im Normalfall wird eine vglw. kleine verbleibende Differenzenergiemenge nachgeladen.
-     Im Worst-Case wird ausschließlich mit Netzbezug geladen.

Dies erfordert eine Eingabemöglichkeit des Ziel-SOC und Zielzeitpunkt (Datum+Uhrzeit) pro Fahrzeug.

premultiply, 2020-07-19:
Eine weitere Ausbaustufe wäre dies mit einer zeitlichen Dynamik unter Berücksichtigung von dynamischen Stromtarifen wie z. B. Awattar oder aber auch festen Hoch/Niedertarif kostenmäßig zu optimieren. Dafür wird dann die Phase der Endladung zum Ziel-SOC zeitlich so vorverlegt, dass sie in den günstigsten Zeitraum des Vorherssagezeitraums der den Zielzeitpunkt enthält fallen.

andig, 2020-07-19:
Das ist *unglaublich* komplex. Wie soll der Algorithmus dazu aussehen?

premultiply, 2020-07-19:
Auf was beziehst du dich? Phase 1 finde ich eigentlich recht einfach. Die dynamischen Stromtarife sind zweifellos recht komplex aber ich halte sie im Rahmen der Zielladen-Funktion als Grundlage für durchaus machbar.
Im Prinzip sucht man 24 Stunden vorher ein Preisminimum für den notwendigen Zeitraum und lädt dann da die benötigte Energie. Ich glaube anders macht es der go-e-Charger auch nicht.

andig, 2020-07-19:
Das ist mehrdimensionale Optimierung: Stromtarife, Solarforecast (woher bekommen wir den?) etc. Wirklich *einfach* wäre nur solange mit PV zu laden bis nix mehr geht um dann im letzten Moment mit voller Leistung reinzuhauen. Das wäre einfach.

premultiply, 2020-07-19:
Ja, das ist Zielladen.
Danach sehen wir dann mal weiter... ;)

premultiply, 2020-09-17:
Böser Bot! 😁

premultiply, 2020-09-24:
Och joah...

andig, 2020-09-24:
Oh yeah 😂😂😂

Optic00, 2020-10-06:
Vom Prinzip hätte ich auch gern so eine Funktion, die wenn das Auto angeschlossen ist und die Sonne schon unten ist, mindestens 50% SoC (oder einen einstellbaren Wert) lädt. Sinn wäre, dass man im Notfall immer genug Strom in der Batterie hat. Wenn es dann am nächsten Tag wieder einen Überschuss gibt, die 50-100% halt nur PV oder min+pv (je nachdem was eingestellt war)

andig, 2020-10-06:
@Optic00 genau das wird in #370 doch gebaut?

andig, 2020-12-16:
Der anhängige PR ist jetzt schon uralt, viel Nachfrage gab es bisher nicht. Ich überlege ob wir das Feature wirklich benötigen oder hier zu machen sollten?

premultiply, 2020-12-16:
...Warten auf eine UI-Lösung...

andig, 2020-12-16:
> ...Warten auf eine UI-Lösung...

...wenn ich das mit @naltatis mal diskutieren kann, ok

PR #370 (2021-01-02): Add target charging
Fix #225, experimental implementation, no UI
closed 2021-01-02 by andig