evcc-io/evcc #231: Simple LPs (on/off)

premultiply, 2020-07-19:
Wer Fahrzeuge oder andere größere akkubetriebene Gerätschaften besitzt die nur über normale Schuko oder CEE-Steckdosen geladen werden können wünscht sich dafür häufig ebenfalls eine Überschusslademöglichkeit in Form einer passend ein- und ausgeschalteten Steckdose.

Solche Fahrzeuge sind z. B. Lastenfahrräder, Roller, E-Scooter, Twizy, Gabelstapler, Rasenmäher uvm. oder aber auch Standard-E-PKWs mit Ladeziegel statt Wallbox.

Diese "Simple LPs" sollen von der Logik her genau wie EVSE-LPs funktionieren mit dem Unterschied, dass dabei der maximal mögliche/erwartete Ladestrom fest vorkonfiguriert werden muss und und es keine Stromregelung gibt bzw. dass dabei immer gilt
Ladestrom = Maximalstrom = Minimalstrom.
Für Schukodosen würde man vielleicht einen Defaultwert von 1P 10A hinterlegen.
Es müssen aber auch sowohl z. B. 3P 32A für Gabelstapler wie auch wesentlich kleinere Ströme für kleinere Ladesysteme von Rollern oder Lastenfahrrädern möglich sein.

Typischerweise schaltet man die Energiezufuhr bei solchen Steckdosen dann direkt über ein Relaisboard bzw. Installationsschütz. Ein individuelles Metering ist natürlich auch optional möglich.

andig, 2020-07-19:
Hast du so ein Ding?

premultiply, 2020-07-19:
Du meinst sowas? Ja.

https://www.antratek.de/8-relaismodule-ethernet-eth008?gclid=EAIaIQobChMIps7KjKPa6gIVxuN3Ch3EqwkCEAQYCiABEgI9hPD_BwE

premultiply, 2020-07-20:
Optimal wäre natürlich ein preiswertes Hutschienengerät was ein ordentliches Relais bzw. Schütz, eine IP- oder Modbus-Kommunikationsschnittstelle und optional einen Energiezähler enthielte.
Wer kennt sowas bzw. den richtigen Fachbegriff dafür? Sowas gibt es bestimmt schon.

Natürlich gibt es diesen ganzen Tasmota-China-Kram für Heimautomatisierer aber da findet man ja schon kaum ein Gerät an das man freiwillig einen größeren Verbraucher wie z. B. eine Waschmaschine anschließen möchte.

andig, 2020-07-20:
>  Sowas gibt es bestimmt schon.

FritzDECT Steckdosen kenne ich- geht aber nur bis 1500W siw.

premultiply, 2020-07-20:
Einigermaßen leistungsstabil sollen wohl Sonoff POW R2 sein.

Bei den FritzDECT-Teilen sind nur die Aussensteckdosenzwischenstecker voll belastbar aber die sind teuer und benötigen ohnehin alle zwingend eine Fritzbox.

Caibuk, 2020-07-20:
Das Beschriebene nennt sich (Hutschienen-) Schaltaktor (bspw. hier https://www.voltus.de/hausautomation/knx-eib/aktoren/schaltaktoren/).
Das spricht kein Modbus oder irgendetwas Proprietäres, sondern [KNX](https://de.wikipedia.org/wiki/KNX-Standard).

Damit kann man _echtes_ Smarthome machen. Langlebig, zukunftssicher und herstellerunabhängig.

Leider nicht unbedingt etwas für die Nachrüstung ...

premultiply, 2020-07-20:
Oha. 9,6 kbit/s über Klingeldraht zu Apothekenpreisen und 6 bis 16A Schaltleistung. Süß.
Gibt es da auch wenigstens "Männerversionen" die wenigstens mal 3P 32A schalten können?

Da ist es ja wesentlich preiswerter und einfacher ne extern beschaltete EVSE, Schütz und Zähler vor jede Dose zu bauen...

andig, 2020-07-20:
> Gibt es da auch wenigstens "Männerversionen" die wenigstens mal 3P 32A schalten können?

Das dürfte eher eine spezielle Anforderung sein :)

premultiply, 2020-08-20:
Es gibt auch sowas: https://de.aliexpress.com/item/4000218727784.html?spm=a2g0x.12010610.8148356.5.13f55e2186AoaJ

blawusel, 2020-08-24:
Ich nehme für sowas ein Sonoff Basic mit Tasmota Firmware
https://www.amazon.de/dp/B07CLDDY2G/ref=cm_sw_em_r_mt_dp_Mt-qFb9EV4FMQ
und schalte damit einen ABB ESB25-40N-06, Installationsschütz
https://www.amazon.de/dp/B07H966LXK/ref=cm_sw_em_r_mt_dp_-u-qFbZYV5S1A

premultiply, 2020-08-24:
Ja, da fehlt dann nur die direkte Energiemessung

blawusel, 2020-08-24:
Hi,
 ich stehe zwar auch nicht so auf den China Kram , aber zum Energiemessen würde es ein Shelly EM
 https://www.conrad.de/de/p/shelly-20210-schalt-messaktor-2246359.html?hk=SEM&WT.srch=1&WT.mc_id=google_pla&s_kwcid=AL%21222%213%21434983517412%21%21%21g%21%21&ef_id=CjwKCAjwyo36BRAXEiwA24CwGSHYCbu6nbgvPOM7V_htgctyHxnt2RfX6K9ghX91Nl52jHxa4kj6JhoCu1sQAvD_BwE%3AG%3As&gclid=CjwKCAjwyo36BRAXEiwA24CwGSHYCbu6nbgvPOM7V_htgctyHxnt2RfX6K9ghX91Nl52jHxa4kj6JhoCu1sQAvD_BwE

bestimmt auch tun

premultiply, 2020-08-24:
Mindestens 1P 16A Dauerlast und 16A auch mehrfach unter Volllast schaltbar?

Ich schätze mal das machen die Dinger nicht lange mit.

Von den 22 kW reden wir lieber noch garnicht...

blawusel, 2020-08-25:
Schalten sollen die einen richtigen Schütz (Installationsschütz von ABB), der kann auch die Last also 40 A schalten , messen machen sie mit Hilfe von Klemmen ,

premultiply, 2020-08-25:
Da die Strommessung über über einen Shunt-Widerstand (oder externen Wandler) pro Phase erfolgen muss ist das extern bei diesem Teil nicht möglich. Damit kann man nur direkt schalten und messen.

andig, 2020-10-29:
Ich mache hier mal zu mangels akutem Bedarf. Bei Notwendigkeit können wir wieder aufmachen.

closed 2020-10-29 by andig