evcc-io/evcc #232: Multilanguage Support

Tscherno, 2020-07-20:
Wenn ich es richtig sehe, sind die Frontendtexte momentan statisch und nur auf Deutsch verfügbar. Eine Auslagerung in eine Sprachdatei und die Unterstützung mehrerer Sprachen wäre aus offensichtlichen Gründen wichtig.

Tscherno, 2020-07-21:
Warum sollte es die nicht geben? Unterstützung von Englisch wäre für den Anfang sicher mehr als ausreichend.

stepir, 2021-04-24:
Hi

I'd like to have the front-end in English. If possible I'm happy to help with the translation although I can see you Guys have a better English than mine :-D

andig, 2021-04-24:
This would need a PR for the frontend part. I don‘t have muxh use for it as we‘re pretty much focused on the German market.

stepir, 2021-04-24:
that's a shame as it's basically few terms in two files?

```
Bezug / Import -> assets/js/components/SiteDetails.vue
Einspeisung / Export -> assets/js/components/SiteDetails.vue
Erzeugung / Production -> assets/js/components/SiteDetails.vue
Batterie / Battery -> assets/js/components/SiteDetails.vue
Sofort / Max -> assets/js/components/Mode.vue
Leistung / Power -> assets/js/components/LoadpointDetails.vue
Geladen / Charged -> assets/js/components/LoadpointDetails.vue
Reichweite / Range  -> assets/js/components/LoadpointDetails.vue
Restzeit / Left -> assets/js/components/LoadpointDetails.vue
```

any chance you could consider set them as variable that a user can configure in evcc.yaml?

andig, 2021-05-02:
> any chance you could consider set them as variable that a user can configure in evcc.yaml?

No, that's not where it belogs. Would accept a proper PR though. 

/cc @naltatis regarding multi-language UI

naltatis, 2021-05-02:
Shouldn't be to complicated to build. I can make the initial implementation and add English.

stepir, 2021-05-02:
Happy to help

naltatis, 2021-05-02:
@stepir I've created a PR with i18n support. I hope I cached all German text. Please review the PR.

andig, 2021-05-04:
Great work, thank you @naltatis

andig, 2021-05-04:
@stepir where are you coming from to take an interest in an english version?

closed 2021-05-04 by andig

stepir, 2021-05-04:
> @stepir where are you coming from to take an interest in an english version?

absolutely!! 

evcc is great, look how it's following my production curve even on cloudy days with constant drops/gains

[image]

I reviewed @naltatis PR and looks good to me. How can I help?

thierolm, 2021-05-04:
> How can I help?

How about providing an Italian language file (it.js), based on https://github.com/andig/evcc/blob/master/assets/js/i18n/en.js ?

;-)

stepir, 2021-05-04:
absolutely - although I'm a believer of using EN everywhere :-)

language file here
https://github.com/andig/evcc/compare/master...stepir:patch-1

thierolm, 2021-05-04:
Cool! Don't forget to press the create pull request button.

stepir, 2022-11-28:
> @stepir Hey, are you still around? Italian translation is way behind the evcc progress now, could you update it ?
> EN https://github.com/evcc-io/evcc/blob/master/assets/i18n/en.toml
> IT https://github.com/evcc-io/evcc/blob/master/assets/i18n/it.toml

hey @RTTTC it's done here https://github.com/evcc-io/evcc/compare/master...stepir:evcc:patch-4

andig, 2022-11-28:
@stepir feel free to open PR when you're ready

andig, 2022-11-28:
> Mir ist der Anwenderkreis nicht klar. Gibt es internationale Nutzer?

We sure have come a long way since... 😻