evcc-io/evcc #233: Grafische Anzeige des Ladeverlaufs mit Forecast

Tscherno, 2020-07-20:
Als Anwender würde ich gerne den aktuellen Ladeverlauf grafisch visualisiert bekommen und auch eine Abschätzung der restlichen Ladedauer sehen. Die voraussichtliche Ladedauer soll mit der aktuellen und maximal möglichen Ladeleistung berechnet werden - es sollen die beiden Werte dargestellt werden. Falls es sinnvoll möglich ist (Speicherung der bisherherigen Zeiten, Datenbank, Konfiguration) sollen das Ladeverhalten des jeweilgen Autos berücksichtigt werden (Ladeprofil). 

Als Idee dient die Batterieanzeige des iPhones. Man könnte vielleicht noch die Forceast-Verläufe als Linie mit darüber legen.
[image]

premultiply, 2020-07-20:
Die Prognose der Ladedauer ist ganz schwierig, da sie individuell vom Ladezustand, vom Ladestrom, von der Akkutemperatur, vom Verbrauch der sonstigen Fahrzeugsysteme usw. abhängt. Das kann bei konstanten Strömen nur das BMS einigermaßen gut, bei variablen Strömen kann man das quasi ganz vergessen.

I. d. R. bekommt man die Ladezeitprognose auch über die jeweilige SOC-API des Fahrzeugs. Die könnte man ggf. anzeigen.

Den Ladestromverlauf könnte man natürlich auch anhand der lokalen Messwerte grafisch aufbereiten.

andig, 2020-07-20:
Charting ist tatsächlich ein ziemlich schwieriges Thema. Hierfür bräuchte ich die Unterstützung eines erfahrenen Frontend Entwicklers, anderenfalls wontfix mangels Kapazität.

TheRealKerni, 2020-07-27:
- Charting: Das wäre schön aber imho keine Priorität so lange sich an den Grundlagen noch einiges ändert. Würde ich auf ne Ideenliste legen
- Prognose: Die Ladeströme sind meist so klein dass man Dinge wie Akkutemperatur etc. ignorieren kann. Größere Auswirkungen haben der vorhanden Überschuss da dieser vom Hausverbrauch und von der dem zu erwarteten Ertrag abhängen. Auch der Lademodus, also Sofort, Min+PV, PV sind entscheidend und eine einigermaßen gute Richtschnur zu geben. Würde ich auch auf ne Ideenliste legen.