evcc-io/evcc #235: Integration in Home Assistant als Addon

Tscherno, 2020-07-20:
Home-Assistant (genau genomme Hass.io) bietet die Möglichkeit Addons ins Frontend zu integrieren und vollständig über HA zu verwalten. Dazu müssen die Applikationen als speziell angepasste Dockercontainer bereitgestellt werden. Es wäre genial, wenn das für evcc klappen würde.

Beispiel für Grafana:
https://github.com/hassio-addons/addon-grafana/

[image]

andig, 2020-07-20:
Da ich kein hassio verwende müsste dieser Beitrag aus der Community kommen.

Tscherno, 2020-07-20:
OK, war erstmal nur als Ideensammlung gedacht. Ich schaue mir das Thema an. Wie funktioniert dein Dockerimage? Zieht sich das automatisch das aktuellste Release von Github?

andig, 2020-07-20:
Ein Dockerimage ist ein Release. Ein neues ziehen musst Du selber. Aber im Ernst: was ist der Nutzen eines Hassio Dockers? Ich habs nicht rausgefunden 😳

Tscherno, 2020-07-20:
1. Zugriffschutz über HA
2. Integration in HA - eine UI für alles rund ums smarte Home
3. Zugang über HA, wenn ein Reverse Proxy / Cloud eingerichtet, klappt es von überall
4. Installation und Updates über Webinterface
5. Riesige Benutzerbasis
6. Editieren des Configfiles über Webinterface
7. Integration in HA Apps

andig, 2020-07-21:
Spannend- die ganzen Infrastrukturthemen sind gute Argumente. Die Zielgruppe- EV Nutzer die evcc einsetzen und dann auch noch HA haben dürfte sehr klein sein :O. Der Aufwand allerdings auch- anscheinend braucht es nicht mehr als eine `config.json`. Falls Du dafür einen Vorschlag machen möchtest könnte ich Dir helfen wir ein entsprechendes Docker Image gebaut werden könnte.

Tscherno, 2020-07-21:
> Die Zielgruppe- EV Nutzer die evcc einsetzen und dann auch noch HA haben dürfte sehr klein sein

Anders rum wird ein Schuh draus: Es gibt sehr viele HA Nutzer die auch EV haben und evcc gar nicht kennen. Ich bin auch nur zufällig über das tolle Projekt gestolpert. 

Ich schaue auf jeden Fall mal, wie weit ich komme. Ich habe schon ein paar Container gebaut, aber noch nie spezifisch für HA.

Ich denke der Knackpunkt für die eigentliche Veröffentlichung ist dann die (fehlende) Mehrsprachigkeit des Frontend. Oder habe ich da was übersehen?

andig, 2020-07-27:
@Tscherno wie startest Du das eigentlich? Mir fehlt das "Supervisor" control unten wenn ich `homeassistant/home-assistant` auf dem Mac launche?

Tscherno, 2020-07-27:
Das geht nur wenn Du Hass.io verwendest. Das ist ein fertiges OS mit Supervisor, der eben die Addons verwaltet. Bei mir läuft das virtualisiert auf der Synology. Ursprünglich war es als fertiges Raspberry-Image gedacht.

Tscherno, 2020-07-27:
Mittlerweile heisst HASS.io auch nur noch Home Assistant und ist die bevorzugte Variante.

Home Assistant vs. Home Assistant Core

Home Assistant Core is a Python program, in simple words. It can be run on various operating systems and provide the ability to track, control and automate your devices. When people talking about Home Assistant Core they usually refer to a standalone installation method.

Home Assistant is a combination of Home Assistant Core and tools which allows one to run it easily on a Raspberry Pi and other platforms without setting up an operating system first. Home Assistant is an all-in one-solution and has a management user interface that can be used from the Home Assistant frontend. This interface is not present in a Home Assistant Core setup.

Be aware that add-ons are only available in regular Home Assistant installations.

andig, 2020-07-27:
Wie installierst Du denn das virtualisierte Hass.io? Ich kann dazu nix finden, es sei denn für Ubuntu. Scheint jedenfalls ein tieferer Eingriff zu werden den ich mir nicht antun möchte.

Tscherno, 2020-07-27:
https://www.home-assistant.io/hassio/installation/

bei Virtual Appliance gibt's fertige Disk-Images.

henning89, 2020-09-27:
@Tscherno Arbeitest du noch an dem Home Assistant Addon?
Ich habe Home Assistant bereits am laufen (auf einem Raspi 4b) und würde gegebenenfalls noch etwas abwarten, wenn ich mir dafür eine separate EVCC Installation ersparen kann.

cathiele, 2020-10-25:
Moin @henning89 kannst du mal versuchen ob du das addon in hass.io mit diesem Custom Repository https://github.com/cathiele/evcc installieren kannst? ich habe das gerade mal versucht aber mein test hass.io ist nur ein alter RPI und der hat nicht genug Speicher um evcc zu bauen. dein Raspi 4b sollte das aber hinbekommen

andig, 2020-10-25:
@cathiele ich hab mir Deinen Fork angeschaut- damit ginge ja nicht mehr als das UI aufzurufen. Ich hatte irgendwie die Hoffnung, dass sich mit Integration mehr erreichen ließe, z.B. dass auch /api direkt in hassio einbindbar wäre. Gibt die Integration das prinzipiell her?

Caibuk, 2020-10-25:
Das UI in Home Assistant aufzurufen ist genau der Sinn dieser Integrationsmöglichkeit. Damit kann man nämlich die Sicherheit der Home-Assistant-Anmeldung für den Aufruf von unterwegs nutzen.
Für eine Funktionsintegration könnte man eine Custom Integration so ähnlich wie diese hier https://www.home-assistant.io/integrations/ bauen.

henning89, 2020-10-25:
Hallo @cathiele ich hab es eben versucht. Leider ist er relativ lange (3-4min) mit der Installation beschäftigt und bricht dann mit folgender Fehlermeldung ab:
„Failed to install addon

The command '/bin/sh -c AUDI_HASH_SECRET={{ env "AUDI_HASH_SECRET" }} make build' returned a non-zero code: 127“

Mein Raspi 4b hat allerdings nur 2GB und keine 4GB RAM, könnte hier ein Problem liegen?

cathiele, 2020-10-25:
@henning89 danke für den test. dann schaue ich erstmal weiter

cathiele, 2020-10-25:
> @cathiele ich hab mir Deinen Fork angeschaut- damit ginge ja nicht mehr als das UI aufzurufen. Ich hatte irgendwie die Hoffnung, dass sich mit Integration mehr erreichen ließe, z.B. dass auch /api direkt in hassio einbindbar wäre. Gibt die Integration das prinzipiell her?

Wie @Caibuk sagt, ist die UI-Integration der Sinn eines Addons. Wenn wir das im sinne einer "integration" in Home Assistant haben wollen, dann kann ich es so wie meine go-eCharger integration bauen, aber was wäre denn der Mehrwert evcc aus Home Assistant heraus steuern zu können?

andig, 2020-10-25:
Der Hashkram war mein Fehler. Ich habe git 2 commits zurück gesetzt und gehoben. Sorry für die Verwirrung.

Caibuk, 2020-10-25:
Ich setze in HA mit einigen Logiken die Lademodi in Abhängigkeit von "dies und das". Durch die Webservices von evcc ist das sehr einfach zu integrieren. Eine custom integration würde es "nur" noch einfacher machen.

andig, 2020-10-25:
As sowas hatte ich gedacht. Den Nutzen des Logins hatte ich bisher nicht gesehen, dafür nutze ich Traefik Forward Auth in Docker, allerdings nicht ganz trivial...

andig, 2020-10-25:
Ich denke in der Version von @cathiele fehlen die weiteren Ports aus dem Dockerfile- TCP wie UDP- damit z.B. auch die SHM Integration von außen funzt. Ich vermute, dass hass dann entsprechende Portregeln für Docker aufsetzt?

cathiele, 2020-10-27:
@henning89  bitte einmal mit dem Repo https://github.com/cathiele/hassio-addons testen. 

Die Konfiguration muss unter /config/evcc.yaml in der hass.io installation liegen (z.B. mit dem "File editor" Addon zu bearbeiten).

Wenn beim ersten Start keine Datei gefunden wird, wird eine einfache default-config dort hinkopiert.

die von hass.io vorgesehene Konfiguration über den entsprechenden Tab bei den Addons geht leider nicht, da dort nur zwei ebenen der Verschachtelungstiefe möglich sind und es eine json und keine yaml Datei sein muss.

andig, 2020-10-27:
Json geht auch- evcc sollte das Format egal sein ;)

cathiele, 2020-10-27:
``

> Json geht auch- evcc sollte das Format egal sein ;)

ok das ist schonmal gut 👍 Aber es ist leider weiter mehr als zwei Ebenen tief verschachtelt.

Tscherno, 2020-10-27:
Servus,

danke an @cathiele  für die Mühe :) Bei mir war das Thema leider eingeschlafen, aber die Updates hier im Thread haben mich wieder geweckt. Ich habe es eben ausprobiert. Der Ansatz mit der ausgelagerten Config ist auf jeden Fall sinnvoll. Ich habe mir damals auch den Kopf zerbrochen wie man die dynamische Config von evcc in HA config abbilden kann - IMO gar nicht sinnvoll. Da würde ich nur sowas wie logging reinnehmen.

Nach der Installation und Ablage von vorhandener config sieht alles erstmal gut aus im Log:
```
[core  ] INFO 2020/10/27 08:09:11 site config:
[core  ] INFO 2020/10/27 08:09:11   grid ✓
[core  ] INFO 2020/10/27 08:09:11   pv ✓
[core  ] INFO 2020/10/27 08:09:11   battery ✓
[core  ] INFO 2020/10/27 08:09:11   grid meter config:
[core  ] INFO 2020/10/27 08:09:11     power ✓
[core  ] INFO 2020/10/27 08:09:11     energy ✓
[core  ] INFO 2020/10/27 08:09:11     currents —
[lp-1  ] INFO 2020/10/27 08:09:11 loadpoint 1 config:
[lp-1  ] INFO 2020/10/27 08:09:11   vehicle ✓
[lp-1  ] INFO 2020/10/27 08:09:11   charge ✓
[lp-1  ] INFO 2020/10/27 08:09:11   charge meter config:
[core  ] INFO 2020/10/27 08:09:11     power ✓
[core  ] INFO 2020/10/27 08:09:11     energy —
[core  ] INFO 2020/10/27 08:09:11     currents ✓
[lp-1  ] INFO 2020/10/27 08:09:11   charger config:
[lp-1  ] INFO 2020/10/27 08:09:11     power ✓
[lp-1  ] INFO 2020/10/27 08:09:11     energy —
[lp-1  ] INFO 2020/10/27 08:09:11     currents ✓
[lp-1  ] INFO 2020/10/27 08:09:11     timer —
[core  ] DEBUG 2020/10/27 08:09:11 ----
[core  ] DEBUG 2020/10/27 08:09:12 pv power: 223W
[core  ] DEBUG 2020/10/27 08:09:12 grid power: 910W
[core  ] DEBUG 2020/10/27 08:09:13 battery power: -157W
[core  ] DEBUG 2020/10/27 08:09:13 site power: 753W
[lp-1  ] DEBUG 2020/10/27 08:09:13 charge power: 0W
[lp-1  ] DEBUG 2020/10/27 08:09:16 vehicle soc: 34%
[lp-1  ] DEBUG 2020/10/27 08:09:16 charger status: B
[lp-1  ] DEBUG 2020/10/27 08:09:16 max charge current: 0A = 0A + -2A (753W @ 3p)
[lp-1  ] DEBUG 2020/10/27 08:09:16 target charge current: 0A
```

Leider klappt der Zugriff dann nicht. Beim Aufruf über HA bekommt man:
"Request failed with status code 401" und "server unavailable". Die Oberfläche ist dann ziemlich leer. Bei Konfiguraiton ist auch nichts sichtbar.

Der fehlschlagende Aufruf ist
GET | http://hass.fritz.box:8123/api/config

[image]

henning89, 2020-10-27:
Ich bekomme die gleichen Fehlermeldungen wie @Tscherno

cathiele, 2020-10-28:
Durch das ingress mapping von hass wird evcc mit der Url (bei meinem TestPi):

http://homeassistant.fritz.box:8123/api/hassio_ingress/fkcZnkj5OdOzGmQi0sviYfRSnDxp3OyzwXqgnp6Qtow

aufgerufen. Hieran schließt sich dann die relative evcc url für die API-Zugriffe an, also z.B. für health:

http://homeassistant.fritz.box:8123/api/hassio_ingress/fkcZnkj5OdOzGmQi0sviYfRSnDxp3OyzwXqgnp6Qtow/api/health

evcc greift aber mit absolutem Pfad per axios auf die api zu. Somit kommt dann am hass-Server das hier an:

http://homeassistant.fritz.box:8123/api/health

@andig kannst du die API-Aufrufe in evcc relativ zu dem Pfad unter dem evcc erreichbar ist durchführen?

cathiele, 2020-10-28:
An der Stelle müsste das dann angepasst werden:

`axios.defaults.baseURL = loc.protocol + "//" + loc.hostname + (loc.port ? ":" + loc.port : "") + "/api";
`

andig, 2020-10-30:
Das Problem ist aus dem Weg, siehe https://github.com/andig/evcc/commit/e52f68798874669458186176feb4f73e2aa5a1a2. Bin aber nicht sicher, ob das mit dem Websocket auch klappt, käme auf einen Test an. Vmtl. abhängig davon, wie Hassio das durch routet.

cathiele, 2020-10-30:
@henning89 @Tscherno  bitte nochmal testen.

henning89, 2020-10-30:
@cathiele 
Ich bekomme leider immer die Fehlermeldung "502 bad gateway", sowohl mit der default evcc.yaml als auch mit meiner customized Variante. Ich greife per https auf Home Assistant zu, ist das möglicherweise ein Problem?

cathiele, 2020-10-30:
Ich mit http ging es bei mir. Dann richte ich auf dem Test Raspberry https zum testen ein.

henning89, 2020-10-30:
Vielleicht hab ich auch die URL falsch zusammengebastelt?
`uri: https://xyz.duckdns.org:8123/hassio/ingress/08a511dd_evcc # uri for ui`

andig, 2020-10-30:
Sinnvoll wäre mal die Sachen einzeln zu testen. Ohne duckdns, ohne Zertifikat...

cathiele, 2020-10-30:
- [X] Ohne Zertifikat geht bei mir
- [X] Mit Zertifikat geht bei mir

duckdns und externen Zugriff von außen habe ich nicht, daher kann ich das nicht testen. @henning89 Kannst du mal die genaue Fehlermeldung senden?

cathiele, 2020-10-30:
achso: und was ist das für eine URI konfiguriation? bei mir steht in der evcc.yaml das hier so wie in der Beispielkonfig:

`uri: 0.0.0.0:7070 # uri for ui`

henning89, 2020-10-30:
Ich hab nochmal die default evcc.yaml geladen und das addon neu installiert.
Leider stoppt das addon ein paar Sekunden nach dem Start.

Im Log steht:
starting evcc --config /config/evcc.yaml
[main  ] FATAL 2020/10/31 00:04:42 cannot create renault vehicle: missing personID

Ich habe leider noch kein Elektroauto (kommt in einer Woche), kann daher keine richtige Konfiguration für ein Fahrzeug hinterlegen.

Wenn ich direkt nach dem Start (bevor es wieder stoppt) das UI öffne, dann kommt nur die Meldung "502: bad gateway".

cathiele, 2020-10-30:
versuche mal diese Datei:

https://github.com/cathiele/hassio-addons/blob/master/evcc/evcc.dist.yaml

andig, 2020-10-30:
Nimm das Auto einfach raus...

Tscherno, 2020-10-31:
Update auf neuestes Addon hat bei mir nicht funktioniert:
"404 Client Error: Not Found ("No such image: 08a511dd/amd64-addon-evcc:latest")"

Deinstallieren und Neuinstallieren hingegen hat funktioniert.

Mit der aktuellen Version läuft es für mich perfekt. Sowohl lokal ohne HTTPS als auch über HA-Cloud-Reverse-Proxy und die App klappt es wie es soll.

Danke für die Mühe!

henning89, 2020-10-31:
Jetzt läuft es auch bei mir, hatte noch einen Fehler in meiner evcc.yaml.
Vielen Dank für die Integration @cathiele !

cathiele, 2020-10-31:
@andig jetzt wo das addon funktioniert bleibt die Frage wo wir das dauerhaft hinterlegen. Der Punkt ist, dass bei einem evcc-Update auch die Version in dem addon hochgezogen werden muss. Aktuell geht es einfach auf latest. Daher denke ich, dass es wohl besser ist das aus meinem Report in dein Repo zu verschieben und in Deinen Releaseprozess zu integrieren. Was denkst du?

Tscherno, 2020-10-31:
Wenn die Versionsnummer im config.json hochgezogen würde, würde auch die Autoupdate-Funktion vom HA-Supervisor funktionieren.

Es sieht jetzt übrigens so aus:
[image]

Das Icon im Menü gibt es leider in der HA Version nicht. Ich ändere es gerade auf mdi:ev-station passt eh besser.

Ansonsten würde ich gerne den Config-File Pfad als Konfigurationsoption hinterlegen. Damit ist erstens klar wo es standardmäßig liegen muss und zweitens kann man leicht zwischen mehreren Configs wechseln.

cathiele, 2020-10-31:
config path ist eine gute Idee. Baue ich ein. Und dann auch gleich loglevel

Tscherno, 2020-10-31:
Ich habe es mal versucht einzubauen. Ich kann es leider nicht von meinem Fork aus testen. Den nimmt HA warum auch immer nicht als gültiges Repository. Ich kann dir einen Pull-Request schicken, dann könnte man es ausprobieren.

https://github.com/Tscherno/hassio-addons/blob/master/evcc/config.json
https://github.com/Tscherno/hassio-addons/blob/master/evcc/scripts/run.sh

cathiele, 2020-10-31:
Kannst gerne einen PR machen. In config.json ist aber bei dir options doppelt drin und Schema sollte dann auch definiert werden.

Tscherno, 2020-10-31:
@cathiele Das Startupscript scheint nicht richtig zu funktionieren. Komischerweise werden nicht mal die beiden Echos am Anfang geloggt. Es steht nur "standard_init_linux.go:211: exec user process caused "no such file or directory"" drin.

Ich denke es liegt an bashio. Das Image hat das vermutlich nicht drin, korrekt? Eventuell gibt es auch eine andere Möglichkeit an die Konfigurationswerte zu kommen?

https://github.com/hassio-addons/bashio

Tscherno, 2020-10-31:
es geht auch ohne, ich bau das mal um.

andig, 2020-10-31:
Nabend. Wenn ihr eine finale Lösung habt gerne PR an evcc. Vmtl ist es einfacher, das dort immer im neuen Build mit auszuliefern- oder müsste auch dann eine config Nummer hochgezählt werden? Hinweis in der README wäre noch gut, die weiterführenden Installationsinstruktionen könnten wir gerne in ein doc/HASSIO.md auslagen mit Link dorthin aus der README.

cathiele, 2020-10-31:
@Tscherno zu spät gesehen, habe schon von bashio auf jq umgebaut :) kannst du mal testen?

cathiele, 2020-10-31:
https://github.com/cathiele/evcc/

Ich habe nun das addon-repo in das evcc gemerged. Somit könnte das dann zu einem PR für evcc werden. Vorher sollten wir noch klären wie wir das mit der Versionierung machen. Ich schaue mir dazu mal an wie bei einem evcc-Release die Version gesetzt wird. oder @andig hast du einen Tipp wie wir in dem Dockerfile zum Zeitpunkt des Publish eines neuen evcc-images die Version in der FROM-Zeile anpassen können?

cathiele, 2020-10-31:
achso, und der Support link geht noch nicht, weil github keine einbettung in iframes erlaubt @andig wäre es ok für dich das  mit target blank in einem neuen Fenster/tab zu öffnen?

andig, 2020-10-31:
@cathiele ich verstehe die Frage nicht. Bei jedem commit bauen ich ein neues evcc Image, das würde die addon-config ja mit winschliessen?

Tscherno, 2020-11-01:
Mit dem aktuellen Build über https://github.com/cathiele/evcc/ scheint der Pfad wieder falsch zu sein (http://hass.fritz.box:8123/api/config) - ich bekomme ich wieder 401 /config beim Öffnen.

Tscherno, 2020-11-01:
Übrigens, wegen der Versionsnummer: jetzt wird in HA bereits 0.30 angezeigt. Ich glaube nicht, dass da noch etwas zu tun ist.

cathiele, 2020-11-01:
@Tscherno lösche nochmal das repo aus hassio und trage es neu ein, wenn du da die 0.30 siehst ist es nicht die aktuelle version

cathiele, 2020-11-01:
hassio verwendet die versionsnummer in der config.json (hassio-spezifisch) um zu erkennen, ob es eine neue version gibt. wenn da "latest" steht wird nie ein update gemacht. Daher wäre es gut, wenn wir in der config.json von dem plugin so etwas wie  evcc-version.<aufsteigende build nummer> anstelle "latest" hätten. 

Ich habe nur aktuell keine Idee ob das mit dem evcc-setup einfach realisierbar wäre.

andig, 2020-11-01:
D.h. Es ist nicht in der Lage zu erkennen, ob es im Docker Hub ein neues Image gibt? Wäre eigentlich schwer zu glauben da das das Standardverfahren ist?

cathiele, 2020-11-01:
Das Image wird ja nicht direkt genommen, sondern nur als Basis für das hassio-image verwendet. 

Bei meinen Tests hat das nicht geklappt obwohl ich es auch gehofft habe. Evtl. prüft hassio aber zu selten, so dass es deswegen bei mir nicht geklappt hat.

Vorschlag: wir nutzen erstmal latest, und beobachten ob es ggf. Doch mit dem Autoupdate klappt wenn lange genug wartet wird. Wenn das nicht klappt gibt es den Workaround löschen und Neuinstallation und wir müssen dann nachbessern.

[102 more replies]

PR #408 (2020-11-08): Add Hass.io add-on
Fix #235. Replaces https://github.com/andig/evcc/pull/406/files.

@cathiele Dein Branch ist immer noch kaputt, ich habs deshalb nochmal neu aufgesetzt.
closed 2020-11-22 by andig