evcc-io/evcc #241: Updatehinweis schließen blockiert evcc

schenlap, 2020-07-25:
**Describe the bug**
Wenn man den orangen Updatehinweis wegklickt kann man UI nicht mehr bedienen.
Log wird nicht mehr upgedated, es hängt. Wenn man dann die Seite neu lädt wird dann ein Default Setting Angezeigt (SOC-Werte wieder 30%/50%/80%/100%, ich habe aber eigentlich 30%/50%/100% konfiguriert.)
features/target_soc

**To Reproduce**
Steps to reproduce the behavior:
1. Auf x bei Updatehinweis klicken
2. SOC kann nicht mehr eingestellt werden, Mode kann nicht mehr ungeschaltet werden

**Expected behavior**
Es geht weiterhin, Updatehinweis (beim selbstkomplieren) soll garnicht erst kommen. Derzeit steht da Installiert: 0.0.1-alpha. Verfügbar 0.13.0

**EVCC details:**
features/target_soc c142e7c54664e83bf5793dd4b88d36e1977d6cf0

andig, 2020-07-25:
Ich kanns nicht nachvollziehen. Aufpoppen des Hinweises falls Du von Hand ohne make gebaut ist ist behoben, schöner wäre aber zu verstehen warum das UI blockiert. Gibts Fehler in der Browser Konsole? Im Logile (trace)?

schenlap, 2020-07-25:
Im Log steht nichts.
Aufhängen (und dann Level Default Werte konnte ich jetzt nicht mehr machen, am Vormittag habe ich es mehrfach geschafft)
Dass es nicht bedienbar ist, ist am Handy weiterhin so. Ich verwende Opera falls du es nachstellen möchtest. Am Hndy habe ich leider keine Konsole...

andig, 2020-07-25:
Geht in kurzem Test auch mit Opera:
- evcc öffnen
- Versionshinweis schließen
- Target Soc ändern
Alles ohne Fehlermeldung oder einzufrieren. Opera auf dem Handy habe ich nicht.

Cache hattest Du gelöscht?

schenlap, 2020-07-25:
Ja, chache ist gelöscht, ansonsten geht garnichts

andig, 2020-07-26:
Da Du das Problem auch mit anderen Webseiten hattest mache ich hier mal zu- vmtl. nicht spezifisch für evcc.

closed 2020-07-26 by andig