evcc-io/evcc #244: Es wird "Willkommen bei evcc" angezeigt, obwohl evcc.yaml ohne Fehler eingelesen wird und Werte berechnet werden

Caibuk, 2020-07-28:
_*** ich schreibe auf Deutsch, da das Projekt momentan nicht lokalisiert/internationalisiert werden soll und mir das auf die Schnelle leichter fällt ;-) ***_

Nach Update auf 0.14.0 wird die "Willkommen bei evcc"-Seite angezeigt (die auf den unkonfigurierten Status hinweist), obwohl die evcc.yaml ohne Fehler eingelesen wird und meiner Meinung nach alle Objekte instantiiert werden (grid, pv, battery, charger).

evcc ist dadurch über Oberfläche nicht mehr bedienbar.

**To Reproduce**
Steps to reproduce the behavior:
1. Update auf 0.14.0
2. Start des Dockercontainers
3. Aufruf des WebIF unter x.x.x.x:7070
4. Oberfläche startet im unkonfigurierten Modus

**Expected behavior**
Anzeige der "normalen" Oberfläche mit konfiguriertem Loadpoint

**EVCC details:**
**If using Docker:**
Show output of `docker run andig/evcc -v`
`evcc version 0.14.0 (d58eb15)`

Show evcc log output with `docker logs <container id>`
```
[main  ] INFO 2020/07/28 09:27:23 evcc 0.14.0 (d58eb15)
[main  ] INFO 2020/07/28 09:27:23 using config file /etc/evcc.yaml
[main  ] INFO 2020/07/28 09:27:23 listening at 0.0.0.0:7070
[mqtt  ] INFO 2020/07/28 09:27:23 connecting evcc-1609481699 at 10.0.1.15:1883
[lp-1  ] INFO 2020/07/28 09:27:24 charger enabled
[lp-1  ] DEBUG 2020/07/28 09:27:24 set charge current: 6A
[core  ] INFO 2020/07/28 09:27:24 site config:
[core  ] INFO 2020/07/28 09:27:24   grid ✓
[core  ] INFO 2020/07/28 09:27:24   pv ✓
[core  ] INFO 2020/07/28 09:27:24   battery ✓
[lp-1  ] INFO 2020/07/28 09:27:24 loadpoint 0 config:
[lp-1  ] INFO 2020/07/28 09:27:24   vehicle —
[lp-1  ] INFO 2020/07/28 09:27:24   charge ✓
[lp-1  ] INFO 2020/07/28 09:27:24   charger config:
[lp-1  ] INFO 2020/07/28 09:27:24     power —
[lp-1  ] INFO 2020/07/28 09:27:24     energy —
[lp-1  ] INFO 2020/07/28 09:27:24     timer ✓
[core  ] DEBUG 2020/07/28 09:27:24 ----
[core  ] DEBUG 2020/07/28 09:27:24 pv power: 2276W
[core  ] DEBUG 2020/07/28 09:27:24 grid power: -10W
[core  ] DEBUG 2020/07/28 09:27:24 battery power: -1695W
[core  ] DEBUG 2020/07/28 09:27:24 site power: -1705W
[lp-1  ] DEBUG 2020/07/28 09:27:35 charge power: 0W
[lp-1  ] DEBUG 2020/07/28 09:27:35 charger status: B
[lp-1  ] DEBUG 2020/07/28 09:27:35 max charge current: 7A = 0A + 7A (-1705W @ 1p)
[lp-1  ] DEBUG 2020/07/28 09:27:35 target charge current: 6A
[core  ] DEBUG 2020/07/28 09:27:35 ----
[core  ] DEBUG 2020/07/28 09:27:35 pv power: 2276W
[core  ] DEBUG 2020/07/28 09:27:35 grid power: 0W
[core  ] DEBUG 2020/07/28 09:27:36 battery power: -1690W
[core  ] DEBUG 2020/07/28 09:27:36 site power: -1690W
[lp-1  ] DEBUG 2020/07/28 09:27:36 charge power: 0W
[lp-1  ] DEBUG 2020/07/28 09:27:36 charger status: B
[lp-1  ] DEBUG 2020/07/28 09:27:36 max charge current: 7A = 0A + 7A (-1690W @ 1p)
[lp-1  ] DEBUG 2020/07/28 09:27:36 target charge current: 6A
[core  ] DEBUG 2020/07/28 09:27:44 ----
```

andig, 2020-07-28:
@TheRealKerni behebt das gerade, danke!

TheRealKerni, 2020-07-28:
Ist in #245 gefixt

closed 2020-07-28 by andig