evcc-io/evcc #247: Empfohlenes setup

friflo, 2020-07-28:
Hi,
habe bisher noch nicht mit Docker und Raspberry Pi zu tuen gehabt.
Daher frage eine eventuell blöde Frage.

Ist meine Annahme richtig: evcc läuft als service in einem Docker Image auf einem Raspberry Pi. korrekt?
Wenn ja, welches Gerät empfiehlst du?

andig, 2020-07-28:
Du kannst es in Docker oder als normales Binary betriben, ganz nach Wunsch. Was meinst Du mit empfehlen?

friflo, 2020-07-28:
okay, danke!
Mit "empfehlen" meinte ich z.B.: Raspberry Pi 4 Modell B, 4GB RAM.
Reicht der, oder gibt es was geeigneteres.

andig, 2020-07-29:
Reicht völlig. Im Prinzip reicht da aber auch eine alte "Gurke", selbst der Raspi1 sollte genügen!

friflo, 2020-07-29:
Danke!
Werde mir am WE das Projekt anschauen. Go fehlt noch in meiner Liste. Mir gefällt das Projekt!

andig, 2020-07-30:
Super, vielen Dank! Dann mache ich hier erstmal zu 👍🏻

closed 2020-07-30 by andig