evcc-io/evcc #255: SOC Interpolation

premultiply, 2020-07-30:
Bei vielen Fahrzeug-SOC-APIs kann der SOC nur in größeren Zeitabständen während der Ladung aktualisiert werden.

Derzeit änder sich der angezeigte SOC ausschließlich wenn ein neuer SOC von der API abgefragt werden konnte.

Um dennoch ein besseres Feedback des Ladestands in der GUI zu bekommen wäre es hilfreich den SOC-Fortschritt anhand der letzten beiden SOC-Meldungen sowie der dazwischen und anschließend geladenen Energiemenge zu interpolieren und kontinuierlich zu aktualisieren.
Auf diese Art wäre das System selbstjustierend und die Abweichung zwischen externer Prognose und tatsächlichem Stand minimal.

andig, 2020-07-30:
Mhhhm.... hättest Du einen Vorschlag für den Algorithmus? So völlig trivial ist das anbetrachts aller Randbedinungen eigentlich nicht?

premultiply, 2020-07-30:
Ja.

Ich würde bei aktivierter Ladung jeweils immer die letzten zwei SOC-Werte **S**_1_, **S**_2_ bei Änderung sowie den dann zugehörigen Chargemeter-Zählerstand **E**_1_, **E**_2_ zu diesen Zeitpunkten cachen. Wobei **S**_2_ und **E**_2_ jeweils die aktuelleren Daten sind. Dann daraus jeweils bei Aktualisierungen die lineare Steigung "SOC-Prozent pro Wh" berechnen

**m** = (**S**_2_-**S**_1_) / (**E**_2_-**E**_1_).

Den interpolierten "Echtzeit-SOC" für die Anzeige bekommt man dann zu jeder Zeit mit

**S**_gui_ = **S**_2_ + ((**E**_aktuell_ - **E**_2_) * **m**)

Glaube ich :D
Aber zumindest so in der Art.

Irrtümer und Fehler vorbehalten. Ich lasse mich gerne korrigieren!

schenlap, 2020-07-31:
Eine andere, einfachere Möglichkeit wäre es die Akkugröße (**A**) in der Config einzustellen.
So macht es auch evse-wifi.
**S**_neu = **S**_letzter + **E**_letzte / **A**

**E**_letzte .. geladene Energie seit letztem SOC Update
**A** .. Akkugröße (nutzbare Kapazität)
**S** .. SOC

Jedenfalls würde ich die Funktion super finden. Der geschätzte Wert sollte dann auch für das Ladeziel verwendet werden.

Zusätzlich könnte auch gleich noch ein durchschnittlicher Verbrauch vom Fahrzeug in der Config aufgenommen werden. z.B 18kWh/100km, damit könnte man dann nicht nur die geladene Energie sondern die geladenen km anzeigen, ebenso beim SOC). Macht evse-wifi so und finde ich sehr praktisch.
Wenn durchschnittlicher Verbrauch nicht konfiguriert -> keine km Anzeige in der GUI
Wenn Akkugröße nicht konfiguriert -> keine SOC Interpoaltion

andig, 2020-07-31:
Die Akkugröße findest Du heute schon in der Konfig- heisst „capacity“. Ohne die wäre es auch nicht möglich die verbleibende Ladezeit zu berechnen!

premultiply, 2020-07-31:
Das würde es wesentlich ungenauer machen und vor allem nicht der schwankenden und unbekannten Speicherkapazität der Akkus nach Temperatur und Alter Rechnung tragen. Je weiter man von der letzten SOC-Übermittlung weg ist desto weiter läuft das dann auseinander. Ausserdem würde man dafür eine fehlerträchtige benutzerseitige Konfiguration benötigen. Die **richtige** Neu-Kapazität des Akkus kennt kaum ein Nutzer. Die landläufigen Marketinggrößen stimmen kaum mit der tatsächlichen Kapazität überein. Daher und wegen der nicht berücksichtigten Umgebungsbedingungen sind die darauf bezogenen Prognosen quasi unbrauchbar.

andig, 2020-07-31:
> Ausserdem würde man dafür eine fehlerträchtige benutzerseitige Konfiguration benötigen

S.o.- die braucht es heute schon

premultiply, 2020-07-31:
Ja und ist auch dabei schon weit daneben.
Die Schätzung und Fortschreibung auch der Restzeit über die SOC-API (wo möglich) wäre genauer da sie vom BMS kommt welches die genaueren Rahmenbedingungen kennt.

premultiply, 2020-07-31:
> Nachteil wäre dass ggf. Der Fortschritt auch mal um ein paar Prozent zurück springen kann- das müssten wir vllt einfach testen?

S.o.
Daher mein Vorschlag welches diesen Fehler nicht macht und sich konfigurationsfrei und automatisch korrigiert.

andig, 2020-07-31:
Für die Restzeit wäre das nicht anwendbar da dafür die verbleibende Grösse bekannt sein muss. 

Falls Du soc apis kennst die Restdauer liefern bitte Info.

premultiply, 2020-07-31:
z. B. Renault ZE API

mclane, 2020-07-31:
Für Fahrzeuge, deren API (noch) nicht unterstützt wird, wünsche ich mir die Möglichkeit, den SoC bei Beginn der Ladung eintragen zu können. Mir ist klar, dass die Vorausrechnung durch die o.g. Problematik (Beispiel Zoe oben) ungenau ist, aber das wäre besser als gar nix.

andig, 2020-07-31:
> Time remaining 4:40

Ups 😬. Na dann bauen wir das doch mal als Erstes und Offensichtlichstes ein 😀

premultiply, 2020-07-31:
Je genauer man die Messzeitpunkte der via API übermittelten SOC-Werte bestimmen kann, z. B. durch eine hohe Aktualisierungsrate oder wie z. B. bei Renault über einen präzisen Zeitstempel deste mehr kann man auch anderen Schätzungen daraus gut ableiten.
Im Prinzip kann man auch aus jeweils mindestens zwei Messpunkten durch oben genannte lineare Interpolation von Fahrzeug-SOC und lokalem Charger-Zählerstand auch eine virtuelle Batteriekapazität berechnen und diese dann wiederum für Prognosen hinsichtlich SOC und verbleibender Ladedauer (über die momentante Charger-Leistung) hinzuziehen.

Ich will damit sagen mit ein bisschen einfacher Mathematik kann man da schon viel herzaubern wenn die Grundlagen da sind.
Einziger Nachteil: Es braucht dazu immer wie gesagt immer erst mindestens zwei zeitlich unterschiedliche Messpunkte.
Je nach Implementierung und verfügbaren Messwerten sind die Vorhersagen dann mehr oder weniger präzise.

schenlap, 2020-07-31:
Natürlich ist das Verfahren über 2 SOC Werte genauer. Man könnte natürlich auch beides kombinieren, dann wird's aufwändiger.
Beim anstecken bis zum zweiten SOC-Wert: Schätzung über Akkukapazität, danach Berechnung über die 2 SOC Werte.
Sofort beim Anstecken & Ladestart eine Restzeitschätzung zu haben ist sicher sinnvoll, auch wenn sie nicht so genau ist.

andig, 2020-08-09:
Nach einigem Nachdenken gefällt mir das nicht so gut. Gerade für die Fahrzeuge die nur zufällig und in langen Intervallen Werte liefern dürfte es durch dieses Verfahren eher schlechter werden da die Werte nicht zum gemessenen Zeitpunkt passen?

premultiply, 2020-08-09:
Was meinst du mit DAS?

premultiply, 2020-08-17:
Done by #277

closed 2020-08-17 by premultiply