evcc-io/evcc #256: Help for Config of Fronius Smyo

StefanSchoof, 2020-08-01:
**Is your feature request related to a problem? Please describe.**
I operate a Fronius Symo Inverter for my PV. I try to get the data into evcc, but I fail to create a config

**Describe the solution you'd like**
Help to get my config working.

**Describe alternatives you've considered**
At this time none.

**Additional context**
My current config
```yml
meters:
- type: modbus
  uri: pv.fritz.box:502
  model: FRONIUS1
  id: 1
  power: DC Power
```
Output
```shell
$ docker run -v $(pwd)/evcc.yaml:/etc/evcc.yaml -p 7070:7070 andig/evcc evcc meter
[modbus] FATAL 2020/08/01 16:25:20 invalid measurement for power: DC Power
```

Running mbmd is working and results into data in the browser
```shell
$ ./mbmd run -a pv.fritz.box:502 -d SUNS:1
2020/08/01 17:01:48 mbmd 0.10 (e83718d)
2020/08/01 17:01:48 config: creating TCP connection for pv.fritz.box:502
2020/08/01 17:01:48 config: declared device SUNS:1
2020/08/01 17:01:48 httpd: starting api at 0.0.0.0:8080
2020/08/01 17:01:51 initialized device Fronius1.1: {Fronius Symo 7.0-3-M 3.16.6-1 0.3.24.1 27331464}
```
[image]

andig, 2020-08-01:
Interessant. Sind diese 3 Werte die einzigen die Dein WR anbietet? Problem ist vmtl. dass Du hier nur Stringwerte hast, aber keinen Gesamtwert. Ums genauer zu sagen müsste ich testen- melde Dich gerne per Mail bei mir (cpuidle@gmx.de) falls Du mir remote Zugang geben möchtest.

StefanSchoof, 2020-08-02:
Ich habe ein Lösung gefunden.

Bei den Einstellungen des Wechselrichter gibt es die Option `Sunspec Model Type`. Dies war auf `float` gestellt. Ich habe das jetzt auf `int + SF` gestellt.
[image]

Jetzt zeigt mdmb auch mehr Werte und Totals an:
[image]

und `evcc meter

closed 2020-08-02 by StefanSchoof