evcc-io/evcc #260: Bug: SOC-Begrenzung schaltet Lademodus nicht auf Stop

premultiply, 2020-08-02:
Die frei definierbare SOC-Begrenzung beendet zwar nach erreichen des gewünschten Ladzustands die Ladung in dem die EVSE gesperrt wird. In der GUI wechselt der Lademodus aber nicht auf Stop, so dass weder ersichtlich ist dass die Ladung beendet wurde und sie auch nicht mehr durch Rückschalten auf den entsprechenden Lademodus doch noch weiter fortgesetzt werden kann.
Dieses Verhalten ist mindestend verwirrend und nicht erwartet.

andig, 2020-08-02:
Naja- der Lademodus bleibt ja PV. Wenn pv 100% erreichts ist geht ja auch mix auf stop?

premultiply, 2020-08-02:
In deinem Fall schaltet das Fahrzeug ab und der Charger bleibt enabled.

Ich hoffe nicht dass EVCC bei 100% auch zusätzlich immer den Charger abschaltet. Das hab ich jetzt nicht getestet.
Das wäre u. U. ungünstig, da man dann nicht mehr am Netz vorklimatisieren könnte.
Sollte das aber so beabsichtigt sein müsste man dann noch eine "unbegrenzt"-Vorauswahl hinzufügen die bei Ladeende nicht abschaltet bzw. auch jederzeit den Ladevorgang fortsetzen kann.

Hilfreich wäre jedenfalls immer eine grafische Signalisierung dass der gewählte Ziel-SOC erreicht ist und deshalb nicht weiter geladen wird. Ob der Modus auf Stop wechselt oder dies anders gemacht wird ist mir in dem Fall wirklich egal. Hauptsache man sieht was Sache ist und man kann zuverlässig manuell fortfahren.

andig, 2020-08-02:
> Ich hoffe nicht dass EVCC bei 100% auch zusätzlich immer den Charger abschaltet.

Jetzt nicht mehr: 569e1e89389b4f654ccf03108ee62de354c738a6

> Hilfreich wäre jedenfalls immer eine grafische Signalisierung dass der gewählte Ziel-SOC erreicht ist und deshalb nicht weiter geladen wird. 

Das zeigt die SoC Anzeige des Autos an. Das Verhalten ist analog zum PV Modus. Warum reicht das nicht oder ist verwirrend?

andig, 2020-08-04:
ping @premultiply fehlt hier noch etwas?

closed 2020-08-04 by premultiply