evcc-io/evcc #265: evcc schaltet spontan von Sofortladen auf PV Laden

mclane, 2020-08-04:
**Describe the bug**
Ich hatte die Ladung mit "Sofortladen" gestartet, da ich trotz heute fehlender Sonne morgen einen vollen Akku brauche. Trotzdem schaltete sich evcc spontan von Sofortladen (bei mir mit 20 A /ca. 4.3 kW) auf PV Überschussladen (Modusumschaltung wird nicht angezeigt!)

**Expected behavior**
Spontanladen sollte mit maximal zulässigem Strom laden bis das Fahrzeug die Ladung beendet.

**EVCC details:**

[core  ] DEBUG 2020/08/04 14:28:01 pv power: 2580W
[core  ] DEBUG 2020/08/04 14:28:01 grid power: 2296W
[core  ] DEBUG 2020/08/04 14:28:01 site power: 2296W
[lp-1  ] DEBUG 2020/08/04 14:28:01 charge power: 4370W
[lp-1  ] DEBUG 2020/08/04 14:28:01 vehicle soc: 50%
[lp-1  ] DEBUG 2020/08/04 14:28:01 charger status: C
[core  ] DEBUG 2020/08/04 14:28:11 ----
[core  ] DEBUG 2020/08/04 14:28:12 pv power: 2620W
[core  ] DEBUG 2020/08/04 14:28:12 grid power: -97W
[core  ] DEBUG 2020/08/04 14:28:12 site power: -97W
[lp-1  ] DEBUG 2020/08/04 14:28:12 charge power: 2403W
[lp-1  ] DEBUG 2020/08/04 14:28:12 vehicle soc: 50%
[lp-1  ] DEBUG 2020/08/04 14:28:12 charger status: C

evcc Vers. 0.19 aus dem tarball

andig, 2020-08-04:
Vermutung: Duplikat zu #264

andig, 2020-08-04:
Bitte beobachten und mit log: trace erneut melden.

andig, 2020-08-04:
Hast Du evtl noch den Crash dokumentiert?

mclane, 2020-08-04:
leider nein

andig, 2020-08-05:
@mclane ich mache den Fehler mangels Nachvollziehbarkeit hier zu. Falls Du neue Informationen hast, Logfile etc bitte hier melden und ich mache wieder auf. Vielen Dank dass Du evcc probierst!

closed 2020-08-05 by andig

andig, 2020-08-06:
Evtl. mit behoben https://github.com/andig/evcc/releases/tag/0.21