evcc-io/evcc #270: Fragen zu Wallbox etc.

userjp123, 2020-08-07:
Erst mal: Konzept gefällt mir sehr gut !

Meine Konstellation:
Ich habe noch kein Auto (geplant:ZOE) und keinen Lader.
Ich habe einen Server laufen und lese sowieso schon den Einspeisewert (pos/neg) über Lesekopf aus dem Zähler.
Ich kann den Einspeisewert mittels script/bash zur Verfügung stellen. (klappt auch schon)
Ansonsten würde ich nichts zur Verfügung stellen (PV etc)

Ich will Überschuss-Laden realisieren. (10 kWp auf dem Dach, durch Ost/West max. 6 kW)
ZOE soll man ja nicht ein/dreiphasig umschalten, während sie angesteckt ist.

Fragen:
Welche Wallbox lässt sich am besten durch EVCC steuern ?
	- kann die Wallbe alles, was man braucht ?
	
Wie kann man mit der Wallbe einphasig laden ?
	- über Sicherungen im Verteiler ? 
	- funktioniert die Wallbe dann noch richtig ?

Wird ein Ladeleistungzähler benötigt ?
	- welche Wallbox hat das integriert ?
	- oder reicht da die SOC Abfrage vom Auto ?

Danke !

andig, 2020-08-07:
Da müssen die Experten kommentieren /cc @premultiply 

> Welche Wallbox lässt sich am besten durch EVCC steuern ? kann die Wallbe alles, was man braucht ?

Was braucht man? Leistungssteuerung ja.

> Wie kann man mit der Wallbe einphasig laden ?

Ja

> über Sicherungen im Verteiler ?
> - funktioniert die Wallbe dann noch richtig ?

?

> Wird ein Ladeleistungzähler benötigt ?

Nein, optional

> - welche Wallbox hat das integriert ?

Da gibts verschiedene

> - oder reicht da die SOC Abfrage vom Auto ?

Hat nichts miteinander zu tun ;)

premultiply, 2020-08-07:
Wenn dir 16A / 3,6 kW (1P) PV-Überschussladeleistung reichen nimm die gute und günstige Wallbe Eco 2.0s (11 kW).
Ansonsten brauchst du ne "22 kW"-Wallbox (32A) die du dann bis 20 A (4,6 kW) für PV-Überschussladung nutzen kannst.
Alle Wallboxen versorgen sich selber immer nur von L1. L2 und L3 sind optional.
Wenn du eine günstige Lösung willst um bei Dringlichkeit L2+L3 zuzuschalten mach eine Lasttrennschalter (z. B. Hager 304) davor, der nur mit L2 und L3 beschaltet (L1 dran vorbei) ist. Von dem selektiven Abschalten von Leitungsschutzschaltern würde ich abraten und da einen 3er-Block einbauen, damit alle Phasen auch im Falle eines Falles zusammen abgeschaltet werden.

userjp123, 2020-08-07:
Vielen Dank für die Antworten !
Dann werde ich wahrscheinlich demnächst die Wallbe 16A bestellen.

Allgemeine Frage: Wozu benötige ich einen Ladeleistungzähler ?

Zu EVCC: 
- Config-File: loadpoints - phases  -> dient das nur zur Berechnung ?
Falls ich die Phasen mechanisch umschalten will, muss ich das dann auch anpassen ?

- Hat jemand Erfahrung mit der (neuen)ZOE und PV-Laden (ein/dreiphasig) ?
Was macht mehr Sinn bei 16A: ein- oder dreiphasig. (reichen die 6A/Phase für die ZOE bei 3-phasig)

- Besteht die Gefahr, dass das Auto beim Laden einschläft ? Wie kann man das verhindern ? (OpenWB-AddonPlatine)

Danke schon mal.

andig, 2020-08-07:
> Wozu benötige ich einen Ladeleistungzähler ?

S.o.

> Besteht die Gefahr, dass das Auto beim Laden einschläft ? Wie kann man das verhindern ? (OpenWB-AddonPlatine)

Hier gibts nur Software- wenn Du Owb Hardware brauchst solltest Du im Owb Forum fragen- da sitzen die Experten ;)

andig, 2020-08-07:
@userjp123 issue geschlossen da wir hier kein allgemeines Supportforum bereitstellen.

closed 2020-08-07 by andig

userjp123, 2020-08-07:
...wie kommst du darauf, dass ich OpenWB-Hardware benötige ?
Ich bin Programmierer und die Frage zielte darauf, ob man das mit der Software auch realisieren kann.
Mir geht es darum zu entscheiden, ob ich auf OpenWB oder EVCC oder ganz was anderes setze.

andig, 2020-08-07:
> OpenWB-AddonPlatine