evcc-io/evcc #273: json Input negieren

blawusel, 2020-08-09:
Hi, ich benötige Hilfe bei der Konfiguration meiner Sonnenbatterie . 
In der yaml Datei steht folgendes:

- name: sonnenGrid
  type: default
  power:
    type: http
    uri: http://192.168.1.75:8080/api/v1/status
    jq: .GridFeedIn_W

leider wird jetzt .GridFeedIn_W falsch angezeit ( Bezug und Einspeisung )wird vertauscht .
Was kann man da tun ?

andig, 2020-08-09:
Du kannst einfach scale: -1 mit anfügen

andig, 2020-08-10:
Ich glaub die Frage wurde beantwortet? Falls noch etwas fehlt gerne wieder aufmachen. Dein Beispiel wurde jetzt auch entsprechend dokumentiert: https://github.com/andig/evcc-config#meter-sonnenbatterie-eco-grid-meter-http

closed 2020-08-10 by andig