evcc-io/evcc #280: evsewifi Verbesserungen

schenlap, 2020-08-13:
**Is your feature request related to a problem? Please describe.**
die EVSE Wifi bietet einen eigenen Remote Mode, dieser sollte von evcc verwendet werden. Dieser Mode hat den Vorteil dass über die Webseite die Einstellungen nicht mehr verändert werden können da sie ja über REST API vorgegeben werden.

Die Einstellung dazu:
[image]

Zur Erklärung:
AlwaysActive und remote Controlled Mode unterscheiden sich dadurch, dass im RemoteMode keine Bedienung über GUI möglich ist sondern nur über API.
Im NormalMode (wenn nicht _setStatus_  über API gesteuert wird) muß man die Ladung über den Taster an der Wallbox händisch starten.

**Describe the solution you'd like**
**1) AlwaysActive Mode verwenden **
```
	if params.AlwaysActive {
		evse.HTTPHelper.Log.WARN.Println("cannot control evse- alwaysactive is on")
	}
```
diese Abfrage gehört dann umgedreht.

**2) AlwaysActive Mode verwenden **
```
// Enable implements the Charger.Enable interface
func (evse *EVSEWifi) Enable(enable bool) error {
	url := fmt.Sprintf("%s?active=%v", evse.apiURL(evseSetStatus), enable)
	return evse.checkError(evse.Get(url))
}
```
Enable / Disable wird dann über die Stromvorgabe gemacht, `setStatus` wird dann garnicht mehr benötigt. Stromvorgabe: 0 => AUS, >=6 => EIN

**3) Modbus Fehler **
```
	switch params.VehicleState {
	case 1: // ready
		return api.StatusA, nil
	case 2: // EV is present
		return api.StatusB, nil
```
Es gibt einen Zustand 0 (undokumentiert): Modbus Fehler. Ich bekomme den sporadisch immer wieder mal, behebt sich kurz darauf von selbst wieder.

**4) Meter einbinden **
Das Vorhandensein den Meters ist über API leider nicht abrufbar. Ticket dazu gibt es https://github.com/CurtRod/SimpleEVSE-WiFi/issues/49 , aber bisher keine Reaktion. Dann wäre es schön wenn es in evcc konfigurierbar wäre.

**Describe alternatives you've considered**
es geht auch so ;-)

**Additional context**
Wenn implementiert, kann ich gerne die Doku zur Integration in evcc anpassen.

andig, 2020-08-14:
>  im RemoteMode keine Bedienung über GUI möglich ist sondern nur über API.

Das heisst im "always active mode" ist Bedienung über das GUI möglich? Nicht sehr intuitiv. Bedeutet "Normal Mode"  dann  "API inaktiv"?

> diese Abfrage gehört dann umgedreht.

Warum? Uns sollte der Modus doch egal sein, Hauptsache wir können die EVSE kontrollieren?

> Enable / Disable wird dann über die Stromvorgabe gemacht, setStatus wird dann garnicht mehr benötigt.

D.h. je nach Modus hat die EVSE für die gleiche Funktion ein anderes API? Oder steckt dahinter auch eine andere Funktion?

M.E. sollten wir das so gestalten dass der Anwender nichts tun muss- auch nicht in der EVSE UI- sondern evcc einfach funktioniert. Magst Du evtl. einen PR erstellen? Meter könnten wir nachreichen.

schenlap, 2020-08-14:
>Das heisst im "always active mode" ist Bedienung über das GUI möglich? Bedeutet "Normal Mode" dann "API inaktiv"?
Ja, im always Active Mode ist Bedienung über GUI möglich (aber man braucht den Taster an der Wallbox nicht). API ist aber trotzdem aktiv :-)

>Warum? Uns sollte der Modus doch egal sein, Hauptsache wir können die EVSE kontrollieren?
Eigentlich ja. Die Entwickler empfehlen für OpenWB auch den Controlles Mode zu verwenden. https://openwb.de/forum/viewtopic.php?t=1204
Spricht was dagegen beide Modis zu unterstützen? Da geht es dann unabhängig von der Konfig der evse?

>D.h. je nach Modus hat die EVSE für die gleiche Funktion ein anderes API? Oder steckt dahinter auch eine andere Funktion?
Ob da unterschieldiche Funktionen dahinter sind kann ich dir nciht sagen. Ergebnis ist jedenfalls das gleiche, das Fahrzeug wird geladen.

>Magst Du evtl. einen PR erstellen? Meter könnten wir nachreichen.
Ich kann's mir mal anschauen. Zuvor müssen wir uns aber einig sein welchen Mode du unterstützen willst.

Wäre es auch möglich am Beginn des Regelablaufs einmalig die evse abzufragen und daraus dann alle Daten zu parsen? Ist das in evcc vorgesehen und gibt es  ein Beispiel bei einer anderen Wllbox? Die Statusabfrage (ist glaube ich die erste) dafür zu missbrauchen ist ja nicht sauber.
Für Energy, Strom, Status jedesmal abzufragen ist ja verschwenderisch, die Antwort ist eh immer die selbe und enthält alle Werte.

andig, 2020-08-14:
Das API gibt das nicht direkt her, aber man könnte ein Caching einbauen und einfach Werten jünger als X Sekunden vertrauen?

premultiply, 2020-08-17:
Ebenso ist zu Berücksichtigen, dass die EVSEWifi derzeit regelmäßig den Ladezähler zurücksetzt.
Das hat u. a. auch Auswirkungen auf die SOC-Interpolation #255

Unter https://github.com/CurtRod/SimpleEVSE-WiFi/issues/54 gibt es schon ein passendes Ticket welches bestimmt etwas Untersützung gebrauchen könnte.

andig, 2020-08-17:
Hab mich eingeklingt- das ist ja wirklich merkwürdig.

andig, 2020-08-30:
Ich glaube den haben wir per #302 erledigt.

closed 2020-08-30 by andig