evcc-io/evcc #281: Daten der letzten Ladung (erst) bei connect zurücksetzen

premultiply, 2020-08-14:
Die Ladedaten (geladene Energie, Ladedauer, ...?) werden aktuell offensichtlich direkt nach Ende des Ladevorgangs bzw. bei Moduswechsel genullt.

Dies führt u. a. zu "leeren" disconnect-Push-Benachrichtigungen und Merkwürdigkeiten in der SOC-Interpolation (#257) nach Ladeende.

Ich halte hier ein Verhalten analog der Tankstellenanzeige für sinnvoll wo der Reset erst beim connect erfolgt.

andig, 2020-08-14:
Hättest Du ein Logfile? Es *wird* erst bei Connect zurück gesetzt, keine Ahnung was hier passiert :O

premultiply, 2020-08-14:
Gerade nichts parat. Vielleicht kann @schenlap sein vollständiges Log aus dem SOC-Test bereitstellen?

premultiply, 2020-08-14:
Habs auch gerade nochmal kontrolliert. Der Code ist eigentlich korrekt und es sollte so funktionieren wie gewünscht.
Offensichtlich liegt das Problem an den charge meter-Eigenschaften.

Ich mache dann hier mal wieder zu.

closed 2020-08-14 by premultiply