evcc-io/evcc #298: Restladezeit optional von Fahrzeug-API

premultiply, 2020-08-18:
Die Restladezeitschätzung sollte optional von der Fahrzeug-API übernommen werden können da diese weitere Faktoren in die Berechnung einfließen lassen kann (Ladekurve, Zelltemperaturen, ...) als nur der momentane Ladestand und die Ladegeschwindigkeit und damit eine genauere Vorhersage liefern.

Bekannterweise liefern mindestens die Renault- und Kia-Schnittstellen #297 solche Prognosen.
Andere Vermutlich auch. Rückfallebene ist immer die vorhandene interne Restzeitschätzung. #296

andig, 2020-08-18:
Interessante Idee- wollen wir denn diesen APIs mehr vertrauen oder alternativ vielleicht die SoC Schätzung auch auf die Restladeprognose ausdehnen?

premultiply, 2020-08-18:
Aus Erfahrung und Beobachtung ist der Eigenschätzung des Fahrzeugs (zumindest bei der ZOE) sehr viel mehr als der einfachen internen Berechnung in EVCC zu trauen. Bei ständig schwankenden Ladeleistungen sind beide natürlich Verfahren ziemlich machtlos.

Genau an der Verbesserung der internen Restladeprognose arbeite ich schon unter #296. ;)

Trotzdem finde ich den API-Weg optional und wo vorhanden sehr begrüßens- und wünschenswert.

mclane, 2020-08-20:
Beim Kia kann man auch den TargetSoC vom Fahrzeug holen. Wenn vorhanden, sollte man diesen nutzen.

andig, 2020-08-20:
Macht das Sinn oder sollte man- wenn evcc eingesetzt- den nicht immer in der App auf 100 stellen? Oder lässt sich der auch Remote ändern?

mclane, 2020-08-20:
Naja, wenn ich in der App den TargetSoC auf 100% stelle das Auto aber bei 90% aufhört stimmt die Restladezeit in der WebUI nicht. Den TargetSoC kann man in der UVO connect app einstellen, es müsste demnach auch remote gehen. Dazu müsste man die Kommunikation der UVO connect app aber mal belauschen, damit man sieht, wie das geht.

premultiply, 2020-08-21:
Ja, es macht bei EVCC absolut Sinn das Fahrzeug immer auf 100% zu stellen.

Trotzdem muss man u. U. für die Restzeitkalkulation berücksichtigen dass jemand trotzdem fahrzeugseitig einen niedrigeren End-SoC gewählt hat.

andig, 2020-08-24:
Wenn #310 gemerged ist wird es deutlich einfacher optionale Erweiterungen im Code einzubauen. Ich warte eigentlich nur auf Bestätigung am Beispiel EVSE ob das gut funktioniert.

andig, 2020-08-29:
So, #310 ist drin. Welche Fahrzeuge bieten das und machen es besser als wir selbst?

premultiply, 2020-08-31:
Ich fange mal an:
Renault
Hyundai/Kia

andig, 2020-08-31:
Ich bräuchte jeweils den API Aufruf und die JSON Struktur, idealerweise einfach als PR gegen die entsprechende Datei. Einbinden kann ich es schnell ;)

premultiply, 2020-08-31:
Renault:
Kommt schon im vorhandenen _battery-status_ Abruf mit (**timeRequiredToFullSlow**).

`{'lastUpdateTime': '2019-07-12T00:38:01Z', 'chargePower': 2, 'instantaneousPower': 6600, 'plugStatus': 1, 'chargeStatus': 1, 'batteryLevel': 28, 'rangeHvacOff': 64, 'timeRequiredToFullSlow': 295}`

premultiply, 2020-09-01:
Denk bitte dran dass sich die RemainingTime auf den unter `timestamp` angegenemen Zeitpunkt bezieht. Also im Prinzip wird damit eine Enduhrzeit definiert. Dann nur noch schnell die aktuelle Zeit von der Enduhrzeit abziehen und schon hat man die aktuelle Restzeit.

andig, 2020-09-01:
> Denk bitte dran dass sich die RemainingTime auf den unter timestamp angegebenen Zeitpunkt bezieht.

Guter Hinweis- PR aktualisiert. Du kannst das gerne mittels `evcc vehicle` testen, die remaining time wird jetzt mit ausgegeben.

premultiply, 2020-09-02:
Mist, wir verwenden hier ja noch V1 der API.
Da heisst der Parameter noch **timeRequiredToFullSlow**.

https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/____/kamereon/kca/car-adapter/v1/cars/VF1AG___/battery-status?country=DE

```
{
  "data": {
    "type": "Car",
    "id": "VF1AG___",
    "attributes": {
      "batteryLevel": 55,
      "rangeHvacOff": 140,
      "chargeStatus": 1,
      "lastUpdateTime": "2020-09-02T13:18:58+02:00",
      "plugStatus": 1,
      "chargePower": 1,
      "instantaneousPower": 900,
      "batteryTemperature": 16,
      "timeRequiredToFullSlow": 920
    }
  }
}
```

Wollen wir auf V2 umstellen?

premultiply, 2020-09-11:
Funktioniert prinzipiell (wenn Daten von der API kommen), hat aber ggf. noch einen Rundungsfehler. Hat beim Test immer nur bei x:30 die angezeigte Ladezeit um eine Minute reduziert.
Problem ist halt wenn keine Schätzung von der API kommt wird nichts angezeigt.

closed 2020-09-13 by andig