evcc-io/evcc #299: Zusätzlicher Zustand "SoC unknown"

mclane, 2020-08-18:
Grund:
Bei den Kia UVO Fahrzeugen wird der SoC nur zurückgemeldet wenn das Fahrzeug im Vodafone Netz eingebucht ist. Leider funktioniert UVOconnect (derzeit?) nur über LTE, nicht aber über Wifi (obwohl ein Wifi Modul eingebaut ist!) Wenn am Ladeplatz also kein sicherer Vodafone LTE Empfang möglich ist, kommt keine SoC Rückmeldung vom Fahrzeug. 
Feature request:
Entsprechende Darstellung im  WebUI, z.B. Ladebalken hinterlegen mit "unbekannt", Restladedauer kann dann natürlich auch nicht berechnet werden (ebenfalls unbekannt)

premultiply, 2020-08-18:
Ist bereits so implementiert.

mclane, 2020-08-19:
Wird das auch in der WebUI angezeigt? Was passiert, wenn kein SoC Wert vom Fahrzeug kommt? Da hätte ich gern ein Popup Fenster, bei dem ich den Start-SoC händisch eintragen kann

andig, 2020-08-19:
Halte ich für keine gute Idee. Ich möchte dass mein Auto lädt, nicht dass eine Software mich mit Popups belästigt. Aus meiner Sicht wontfix, gerne zur Diskussion.

premultiply, 2020-08-19:
Derzeit wird dann einfach nur "laden" im Balken angezeigt. Mit meinem PR #296 wird bei temorären Ausfällen weiter der letzte Wert angezeigt bzw. die Schätzung einfach fortgeführt. Vorausgesetzt es wurde mal mindestens einmal ein SOC-Wert übermittelt. Sonst bleibt es bei "laden" - ebenso wie bei Fahrzeugen ohne Vehice-API.

mclane, 2020-08-19:
ok. dann aber auch keinen SoC Ladebalken und Restladezeit anzeigen wenn kein SoC vom Fahrzeug kommt

premultiply, 2020-08-19:
Wie gesagt. Das funktioniert bereits schon genau so.

mclane, 2020-08-20:
Dann mach ich den mal zu.

closed 2020-08-20 by mclane