evcc-io/evcc #313: evcc auf Webserver

userjp123, 2020-08-26:
Hallo.

Ich will evcc auf meinem Server laufen lassen.
Konfiguration:
- Ubuntu Server 18.04.
- Apache Webserver

In Apache-virtualhost schon mit den beiden folgenden Einträgen rumgespielt, ohne Erfolg.
     ProxyPass        /wb1    http://127.0.0.1:7070 
     ProxyPassReverse /wb1    http://127.0.0.1:7070

Im Browser wird der Mittelteil **nicht** dargestellt. ( plus Fehlermeldung rechts erscheinen: _Request failed with status code 404_  gefolgt von einem _Server unavailable_)
Was man sieht:
**evcc Laden Konfiguration Support**
**Home**
_Linie
hier fehlt was 
Linie_
**0.25** 

Hat das schon jemand auf einem externen Webserver am Laufen ?
Wäre ja schön, wenn man im lokalem Netz zB mit dem Handy drauf zugreifen könnte.

Danke schon mal.

andig, 2020-08-26:
Ich vermute ein Problem mit Deiner Proxykonfiguration bzw. Deinem `uri` setting für EVCC. Du wirst nicht darum kommen zu debuggen wo der Request nicht weiter kommt (Apache Log etc). Mit evcc hat das vmtl. wenig zu tun.

closed 2020-08-26 by andig

andig, 2020-08-26:
Was helfen könnte wäre ein Blick auf https://www.serverlab.ca/tutorials/linux/web-servers-linux/how-to-reverse-proxy-websockets-with-apache-2-4/. evcc benötigt Websockets für die Kommunikation mit dem Backend- evtl. sind dafür weitere Proxyregeln notwendig.

obel1x, 2022-06-30:
@userjp123 hast du das hinbekommen? ich hänge da auch - obwohl ich einen anderen service schon damit zum laufen bekommen habe.
Unterschied, den ich sehen kann: ich brauche eine unterseite für den anderen service
also die doma
in heisst da nicht
servername.de:7070/
sondern
servername.de:7070/servicename/

dann kann ich im apache mit location /servicename
einen proxypass localhost:7070/servicename
machen mit reverse, der dann geht

bei evcc kriege ich das nicht hin...