evcc-io/evcc #318: Modbus TCP Solaredge funktioniert nicht mehr

schenlap, 2020-08-29:
**Describe the bug**
```
./evcc_amd 
[main  ] INFO 2020/08/29 11:00:23 evcc 0.0.1-alpha (HEAD)
[main  ] INFO 2020/08/29 11:00:23 using config file /home/steve/go/src/github.com/andig/evcc/evcc.yaml
[main  ] INFO 2020/08/29 11:00:23 listening at 0.0.0.0:7070
[main  ] FATAL 2020/08/29 11:00:23 cannot create default meter: dial tcp 10.0.0.207:502: connect: connection refused
```

**To Reproduce**
Seit commit 6b622b7df76ea9fca0abadd2be4d98c8271709d2 tritt der Fehler auf.

**Expected behavior**
geht.

**evcc.yaml**
```
- name: pvse1 # pv Ertrag from solaredge via modbus
  type: default
  power:
    type: modbus
    uri: 10.0.0.207:502
    id: 1
    model: sunspec
    value: Power
```

andig, 2020-08-29:
Connection refused sieht egtl eindeutig aus. Wieviele Verbindungen akzeptiert der SE gleichzeitig- evtl nur eine? Dann müsste ich den Connection Cache doch wieder einbauen 😳

schenlap, 2020-08-29:
ja, leider nur eine Verbindung. Ist #320 bereit zum Testen?

andig, 2020-08-29:
Ja, mit SimpleEVSE getestet, sollte gehen!

schenlap, 2020-08-30:
geht, danke!

closed 2020-08-30 by andig

schenlap, 2020-09-03:
Ich habe doch noch ein Problem damit. Ich habe 2 Solaredge Wechselchrichter, die Leistung zähle ich mit dem Calc Plugin zusammen. Es wird aber die Leistung von der ersten Abfrage * 2 gerechnet anstatt die beiden Ergebnisse zu addieren.

```
[modbus] TRACE 2020/09/03 12:14:29 modbus: send 00 0b 00 00 00 06 02 03 9c 87 00 32                                    
[modbus] TRACE 2020/09/03 12:14:29 modbus: recv 00 0b 00 00 00 67 02 03 64 01 65 01 65 ff ff ff ff ff fe 09 13 ff ff ff
ff ff ff ff ff ff ff ff ff 1f 3a ff ff c3 5a ff fd 20 79 ff ff 59 1a ff fe 25 90 ff fe 00 05 8d 7d 00 00 26 03 ff fd 03
42 ff ff 1f b4 ff ff 80 00 15 a8 80 00 80 00 ff fe 00 04 00 00 ff ff ff ff ff ff ff ff 00 00 00 00 ff ff ff ff ff ff ff
ff 00 00 00 00
[modbus] TRACE 2020/09/03 12:14:29 Power: 799.40W
[modbus] TRACE 2020/09/03 12:14:29 modbus: send 00 0c 00 00 00 06 02 03 9c 87 00 32                                    
[modbus] TRACE 2020/09/03 12:14:29 modbus: recv 00 0c 00 00 00 67 02 03 64 01 65 01 65 ff ff ff ff ff fe 09 13 ff ff ff
ff ff ff ff ff ff ff ff ff 1f 3a ff ff c3 5a ff fd 20 79 ff ff 59 1a ff fe 25 90 ff fe 00 05 8d 7d 00 00 26 03 ff fd 03
42 ff ff 1f b4 ff ff 80 00 15 a8 80 00 80 00 ff fe 00 04 00 00 ff ff ff ff ff ff ff ff 00 00 00 00 ff ff ff ff ff ff ff
ff 00 00 00 00
[modbus] TRACE 2020/09/03 12:14:29 Power: 799.40W
Power: 1599W
```
**Power: 1599W**

mbmd zum fast selben Zeitplunkt liefert:
```
./mbmd inspect -d SOLAREDGE:1@solaredge:502 | grep W
2020/09/03 12:14:44 config: creating TCP connection for solaredge:502
Md   SE2000M-XXXXXXX    string
W                 1596.40    int16
W_SF                   -1    sunssf

./mbmd inspect -d SOLAREDGE:2@solaredge:502 | grep W
2020/09/03 12:14:45 config: creating TCP connection for solaredge:502
Md   SE1500M-XXXXXXX    string
W                  797.70    int16
W_SF                   -1    sunssf
```
also in Summe **2390W**

Die Config schaut dazu so aus:
```
- name: pvse
  type: default
  power:
    type: calc
    add:
     - type: modbus
       uri: 10.0.0.207:502
       model: sunspec
       value: Power
       id: 1
     - type: modbus
       uri: 10.0.0.207:502
       model: sunspec
       value: Power
       id: 2
```

premultiply, 2020-09-03:
Und die haben beide die gleiche IP?
Oder gehst du da via RS485-IP-Adapter dran wo beide am gleichen Bus hängen?

schenlap, 2020-09-03:
behoben mit https://github.com/andig/evcc/pull/328

PR #320 (2020-08-30): Re-add ModBus connection cache and wrap error and slave id handling
Fix #318 by re-adding the connection cache removed during https://github.com/andig/evcc/commit/6b622b7df76ea9fca0abadd2be4d98c8271709d2