evcc-io/evcc #330: cannot create renault vehicle - certificate signed by unknown authority

JoshSteinmann, 2020-09-05:
Hey all,

I put the following in my config:

vehicles:
- name: Zoe
  type: renault
  title: Zoe
  capacity: 45 # kWh
  user: ******
  password: ******
  region: de_DE
  vin: WREN...
  cache: 5m

and I get the following in the docker log:

[main ] FATAL 2020/09/05 15:03:11 cannot create renault vehicle: Get "https://renault-wrd-prod-1-euw1-myrapp-one.s3-eu-west-1.amazonaws.com/configuration/android/config_de_DE.json": x509: certificate signed by unknown authority

is there a simple way to fix it or is it a proper bug/change in the renault api?

Thanks in advance

andig, 2020-09-05:
How are you running the application? Using docker? Do *NOT* map an entire folder into `/etc` or you're essentially removing the installed certificates...

JoshSteinmann, 2020-09-05:
That was it! Thank you, maybe I misread something. Big thank you for the fast and friendly answer!

andig, 2020-09-05:
Glad I had seen this particular error before. Took me hours to figure out what was wrong...

closed 2020-09-05 by andig