evcc-io/evcc #336: charger phoenix-evcc not working

22kW, 2020-09-12:
Die Statusinterpretation des Chargers (EV-CC-AC1-M3) ist nicht korrekt implementiert.
Gestecktes Ladekabel wird nicht erkannt:

Kein Kabel gesteckt:

[modbus] TRCE 2020/09/12 09:52:23 modbus: send 02 04 5d c0 00 01 22 69
[modbus] TRCE 2020/09/12 09:52:23 modbus: recv 02 04 02 41 31 0c b4
[evcc  ] TRCE 2020/09/12 09:52:23 read status (24000): 41 31
-- Anmerkung: A1 = Kabel nicht gesteckt --
[lp-1  ] DBUG 2020/09/12 09:52:23 charger status: 1
[modbus] TRCE 2020/09/12 09:52:23 modbus: send 02 01 4e 20 00 01 eb 1b
[cache ] TRCE 2020/09/12 09:52:23 chargeCurrent: 6
[cache ] TRCE 2020/09/12 09:52:23 connected: false
[cache ] TRCE 2020/09/12 09:52:23 charging: false
[modbus] TRCE 2020/09/12 09:52:23 modbus: recv 02 01 01 00 51 cc
[evcc  ] TRCE 2020/09/12 09:52:23 read charge enable (20000): 00

Kabel gesteckt:
[modbus] TRCE 2020/09/12 09:56:56 modbus: send 02 04 5d c0 00 01 22 69
[modbus] TRCE 2020/09/12 09:56:56 modbus: recv 02 04 02 42 31 0c 44
[evcc  ] TRCE 2020/09/12 09:56:56 read status (24000): 42 31
-- Anmerkung B1 = Kabel gesteckt, PWM nicht aktiv --
[lp-1  ] DBUG 2020/09/12 09:56:56 charger status: 1
[modbus] TRCE 2020/09/12 09:56:56 modbus: send 02 01 4e 20 00 01 eb 1b
[cache ] TRCE 2020/09/12 09:56:56 connected: false
[cache ] TRCE 2020/09/12 09:56:56 charging: false
[modbus] TRCE 2020/09/12 09:56:56 modbus: recv 02 01 01 00 51 cc
[evcc  ] TRCE 2020/09/12 09:56:56 read charge enable (20000): 00

ich habe mal eine Ladung provoziert (modbus enable mit einer phoenix software aktiviert):
[modbus] TRCE 2020/09/12 09:46:52 modbus: send 02 04 5d c0 00 01 22 69
[modbus] TRCE 2020/09/12 09:46:52 modbus: recv 02 04 02 43 32 4d d5
[evcc  ] TRCE 2020/09/12 09:46:52 read status (24000): 43 32
-- Anmerkung C2 = connected and charging, pwm aktiv --
[lp-1  ] DBUG 2020/09/12 09:46:52 charger status: 2
[modbus] TRCE 2020/09/12 09:46:52 modbus: send 02 01 4e 20 00 01 eb 1b
[cache ] TRCE 2020/09/12 09:46:52 connected: false
[cache ] TRCE 2020/09/12 09:46:52 charging: false
[modbus] TRCE 2020/09/12 09:46:52 modbus: recv 02 01 01 01 90 0c
[evcc  ] TRCE 2020/09/12 09:46:52 read charge enable (20000): 01
[lp-1  ] DBUG 2020/09/12 09:46:52 charger disable - contactor delay 9s

Es gibt laut manual folgende Statusmeldungen, relevant sind aber nur A1 (kein Kabel), B2(Kabel gesteckt) und C2(Ladevorgang aktiv)

Manual:
A1 (12 V DC): Kein Fahrzeug angeschlossen
A2 (12 V PWM): Nur temporärer Übergangszu- stand, geht in A1 über

B1 (9 V DC): EVSE** noch nicht bereit
B2 (9 V PWM): EVSE bereit††

C1 (6 V DC): EVSE nicht bereit, Ladevorgang wird abgebrochen. Übergangszustand, nur bei ver- einfachtem Control Pilot als dauerhafter Zu- stand möglich.
C2 (6 V PWM): Ladevorgang aktiv

D1 (6 V DC): EVSE nicht bereit, Ladevorgang wird abgebrochen. Übergangszustand, nur bei ver- einfachtem Control Pilot als dauerhafter Zu- stand möglich.
D2 (6 V PWM): Ladevorgang aktiv

andig, 2020-09-12:
ping @22kW scheint als hätte ich das falsche Byte ausgewählt- funktioniert es so? Deine WB ist Eigenbau?

22kW, 2020-09-12:
Versuche es morgen zu testen (muss noch rausfinden wie ich go auf dem raspberrypi kompiliere).
Ja ist ein Eigenbau mit USR-DR302 als RS485 Ethernet Converter.
Du prüfst auf das erste ASCII Zeichen des Statuscodes (Ax,Bx,Cx) ? (return api.ChargeStatus(string(b[0])), nil)
A (0x41)=nicht verbunden, B (0x42)=verbunden, C (0x43)=charging ?

andig, 2020-09-13:
> Du prüfst auf das erste ASCII Zeichen des Statuscodes (Ax,Bx,Cx) ?

Ja, das ist die Idee. Compilieren kanns du einfach auf dem PC:

    GOOS=linux GOARCH=arm make

22kW, 2020-09-13:
Kompilieren klappt nicht, bekomme:

martin@iMac-110 evcc % GOOS=linux GOARCH=arm make
rm -rf dist/
go install github.com/mjibson/esc
go install github.com/golang/mock/mockgen
Generating embedded assets
go generate ./...
prog.go:14:2: cannot find package "github.com/andig/evcc/api" in any of:
	/usr/local/go/src/github.com/andig/evcc/api (from $GOROOT)
	/Users/martin/go/src/github.com/andig/evcc/api (from $GOPATH)
prog.go:12:2: cannot find package "github.com/golang/mock/mockgen/model" in any of:
	/usr/local/go/src/github.com/golang/mock/mockgen/model (from $GOROOT)
	/Users/martin/go/src/github.com/golang/mock/mockgen/model (from $GOPATH)
2020/09/13 12:37:05 Loading input failed: exit status 1
api/api.go:5: running "mockgen": exit status 1
fork/exec /var/folders/pm/q2g_c0vn1hb414nqwdcsv4fm0000gn/T/go-build741389459/b001/exe/decorate: exec format error
charger/evsewifi.go:57: running "go": exit status 1
prog.go:14:2: cannot find package "github.com/andig/evcc/core" in any of:
	/usr/local/go/src/github.com/andig/evcc/core (from $GOROOT)
	/Users/martin/go/src/github.com/andig/evcc/core (from $GOPATH)
prog.go:12:2: cannot find package "github.com/golang/mock/mockgen/model" in any of:
	/usr/local/go/src/github.com/golang/mock/mockgen/model (from $GOROOT)
	/Users/martin/go/src/github.com/golang/mock/mockgen/model (from $GOPATH)
2020/09/13 12:37:07 Loading input failed: exit status 1
core/chargerhandler.go:14: running "mockgen": exit status 1
fork/exec /var/folders/pm/q2g_c0vn1hb414nqwdcsv4fm0000gn/T/go-build892585922/b001/exe/decorate: exec format error
meter/meter.go:16: running "go": exit status 1
make: *** [assets] Error 1

andig, 2020-09-13:
Darf ich mal fragen welche Go Version Du hast?

andig, 2020-09-13:
> fork/exec /var/folders/pm/q2g_c0vn1hb414nqwdcsv4fm0000gn/T/go-build892585922/b001/exe/decorate: exec format error
meter/meter.go:16: running "go": exit status 1

Autsch, ich sehs gerade: an dieser Stelle funktioniert das cross-compile nicht. Ich kann Dir ein Binary bauen oder Du müsstest es derzeit doch auf dem Raspi ausführen.

andig, 2020-09-13:
Bzw: mach bitte mal ein `make build` anstatt `make`

22kW, 2020-09-13:
Das hat geklappt, läuft !

andig, 2020-09-13:
Wenn Du magst würde ich mich über Teilnahme an der Umfrage freuen: https://airtable.com/shrgPyZ58Y74SK2Kk

closed 2020-09-13 by andig

22kW, 2020-09-13:
Vielen Dank, super Projekt, Umfrage mache ich gerade!

PR #339 (2020-09-13): Pick relevant status byte
Fix #336