evcc-io/evcc #344: Normal- / Speed Charging Switch

22kW, 2020-09-14:
Ich möchte normalerweise meine Zoe auch bei "Sofort" nur mit 11kW (3 x 16A) laden.
Für ausserplanmässige Notfälle würde ich gerne auf 32A (22kW) für die nächste / aktuelle Ladung umschalten können.
Ein Umschalter per MQTT/API oder Web wäre daher super.

premultiply, 2020-09-14:
Gegenidee: Wie wäre es mit vom Benutzer in der Konfig frei definierbaren und benennbaren Lade-Presets die die Eigenschaften für den eigenen Bedarf festlegen statt der derzeitigen 4 festen Modi.
Darin könnte man dann darin bspw. den minimalen und maximalen Ladestrom festlegen, die Anzahl der zu verwendenden Phasen (1P3P-Umschaltung), die Nutzung von PV-Überschuss, das SOC-Ziel usw.

Ich hab da nämlich so ein ähnliches Anliegen: Ich würde gerne bei PV-Überschuss bis 20A (1P) aber bei Sofortladen nur mit 16A (3P) laden.

andig, 2020-09-14:
Warum sollte man sofort nur mit halber Kraft laden? Statt dieser Optionen erscheint mir Zielladen sinnvoller.

22kW, 2020-09-14:
Ist vielleicht ein Grenzfall. Ich habe festgestellt, dass PV bei mir nicht wirklich gut funktioniert, da meine 3.5kW PV nicht mit den min 5kW Zoe (3Phasen) kompatibel sind. Finde aber das Ladeziel (%) Feature wichtig um den Akku nicht immer auf 100% zu laden (schnellere Alterung) - d.h. ich kann "pv" nicht nutzen, min+pv habe ich für langsames Laden mit hohem PV Anteil, und sofort für "Notladen" mit 22kW. Da fehlt mir eben noch ein "normal" Laden mit 11kW.
22kW ist nett wenn man es ein paarmal im Jahr braucht, sonst aber schlecht für das lokale Netz, den Hausanschluss (bei gleichzeitigem Herd und Wasserkocher könnte die Hauptsicherung gefährdet sein) und evtl. den Akku unnötig belasten. Ausserdem wird das Auto laut bei 22kW (Zoe) und verbraucht im Sommer Kühlenergie.

andig, 2020-09-14:
Ich würde das Ding auf 1p umbauen per 2p Lasttrennschalter. 2x im Jahr kannst Du dann händisch schalten, ohne lässt sich mit 1p wenigstens halbwegs regeln. Aber 3,5kWp ist arg knapp...

Caibuk, 2020-09-15:
Um noch einen (meinen) Anwendungsfall für zwei (oder mehrere) Sofortladenprofile hier einzubringen:
Ich habe eine 8,9 kWp PV-Anlage, einen ziemlich großen Heimspeicher und einen zeitvariablen Stromtarif. Den Speicher kann ich nur mit 4 kW entladen. Ich habe oft den Fall, dass ich abends noch ein paar kWh ins Auto laden muss. Wenn ich hier mein normales Sofortladen (1ph, 16 A) nutze, beziehe ich einiges an HT-Strom (wir haben eine ziemlich hohe Grundlast). Min+PV möchte ich aus religiösen Gründen (😉, hohe Ladeverluste bei 1,3 kW und dauert zu lange) nicht nutzen.
Ein zweites Sofortladen mit 12 oder 13 A wäre mir da eine große Hilfe. Momentan behelfe ich mir, indem ich auf Sofortladen stelle und im Phönix-Controller per Hand dann auf 12 oder 13 A gehe. Zukünftig möchte ich nach etwas Umbau mit 20 A 1ph laden (um mehr bei Sonnenschein laden zu können). Da würde das geschilderte Problem noch größer.
Mir würden also entweder anpass- und auswählbare Ladeprofile helfen, oder die Möglichkeit den maximalen Ladestrom in der GUI zu ändern (und natürlich gern per API).
Weiterhin 👍 für die tolle Software und Eure Arbeit!

andig, 2020-09-15:
Danke für den Anwendungsfall. *Ein* weiterer Modus (sagen wie mal „Mittel“) wäre kein Problem, mehr sollte es erstmal nicht werden. Ich hänge- vor einer Änderung immer noch am Naming. Wie wäre es mit „Max“, „Med“, „Min“, „Eco“ und „Off“?

schenlap, 2020-09-15:
wie wäre es mit "User"?

Ich hätte den dann gerne so konfigurierbar dass ich mir einen PV mode machen kann mit anderen Ein-/Ausschaltschwellen. Dann hätte ich PV mit 100%PV Strom (z.B am Wochenende wenn es egal ist wie lange das Auto lädt) und "User" mit > 70% PV Strom (der soll nur abschalten wenn die Waschmaschine / Geschirrspüler den ganzen PV Strom verbraucht).
Den zweiten PV Mode könnte ich dann auch im Winter gut gebrauchen wenn der 100% PV Mode nie starten würde, min+PV aber zu wenig PV Strom verwendet und zwischendrinn nicht abschalten kann.

Wie immer sollte alles möglichst flexibel konfigurierbar sein.  :-)

premultiply, 2020-09-15:
User1, User2, User3, ... 😁

Vielleicht doch lieber direkt komplett nutzerkonfigurierbare Presets?

Ich vermisse auch einen >70%-Modus der nur lädt wenn abgeregelt werden müsste.

andig, 2020-09-25:
Interessante Idee. Wäre sinnvoll das mit dem Thema Zielladen zu integrieren. Ein API für das externe Setzen von min/max Strom wäre vmtl. auch eine gute Idee.

22kW, 2020-09-25:
Das mit der API (MQTT?) für min/max würde meiner Problemstellung genügen.

andig, 2020-09-25:
> Das mit der API (MQTT?) für min/max würde meiner Problemstellung genügen.

Ich habs mal abgesplittet: https://github.com/andig/evcc/issues/358

andig, 2020-11-22:
Dieses Thema hat sich zu einem Sammelissue für unterschiedlichste Anforderungen entwickelt. Mein Vorschlag wäre Diskussion in Slack, konkrete Einzelanforderungen wieder als Issues. In diesem Sinne erstmal wontfix.

closed 2020-11-22 by andig