evcc-io/evcc #345: Anzeige "Geladen" zeigt immer den Gesamtzählerstand (SDM630 V1)

22kW, 2020-09-14:
Leistung wird korrekt angezeigt, Gesamt (Lebensdauer) Zählerstand auch. Wie bekomme ich nur die aktuell geladenen kWh ?
Setup SDM630 (alt, V1) via RS485/Ethernet Converter

- name: charge
  type: modbus
  model: sdm # SDM630
  uri: 192.168.178.90:8899
  rtu: true # rs485 device connected using ethernet adapter
  id: 1
  power: Power # default value, optionally override
  energy: Sum # default value, optionally override

premultiply, 2020-09-15:
Anderer Effekt gleiche Anzeige: Beim NRGkick startet jede Ladephase dort wieder bei 0 obwohl der interne Energiezähler korrekt weiter hochzählt.

andig, 2020-09-15:
@22kW du hast keinen Zähler am Phoenix Controller angeschlossen, richtig?

22kW, 2020-09-15:
Der Modbus Phoenix Controller hat keinen eigenen Anschluss für einen Zähler. Der SDM630 hängt mit id1 am gleichen RS485 Bus wie der Phoenix Controller mit id2.

[cache ] TRACE 2020/09/15 20:04:49 availableVersion: 0.26
[core  ] DEBUG 2020/09/15 20:04:49 pv power: 0W
[cache ] TRACE 2020/09/15 20:04:49 pvPower: 0
[core  ] DEBUG 2020/09/15 20:04:49 grid power: 355W
[core  ] DEBUG 2020/09/15 20:04:49 site power: 355W
[cache ] TRACE 2020/09/15 20:04:49 gridPower: 354.90000000000003
[cache ] TRACE 2020/09/15 20:04:49 mode: 
[lp-1  ] DEBUG 2020/09/15 20:04:49 charge power: 0W
[cache ] TRACE 2020/09/15 20:04:49 chargePower: 0
[cache ] TRACE 2020/09/15 20:04:49 chargeCurrent: 9
[cache ] TRACE 2020/09/15 20:04:49 chargedEnergy: 0
[cache ] TRACE 2020/09/15 20:04:49 chargeDuration: 0s
[cache ] TRACE 2020/09/15 20:04:50 releaseNotes: 
[lp-1  ] DEBUG 2020/09/15 20:04:50 vehicle soc: 67%
[cache ] TRACE 2020/09/15 20:04:50 socCharge: 67
[cache ] TRACE 2020/09/15 20:04:50 chargeEstimate: -1ns
[lp-1  ] DEBUG 2020/09/15 20:04:50 charger status: A
[lp-1  ] INFO 2020/09/15 20:04:50 car disconnected
[cache ] TRACE 2020/09/15 20:04:50 chargedEnergy: 0
[push  ] ERROR 2020/09/15 20:04:50 invalid template for disconnect: wanted: [connectedDuration], got: [chargeDuration socCharge chargePower targetSoC gridPower chargeEstimate releaseNotes mode chargeCurrent availableVersion chargedEnergy pvPower]
[cache ] TRACE 2020/09/15 20:04:50 connectedDuration: 2562047h47m16.854775807s
[lp-1  ] INFO 2020/09/15 20:04:50 set target soc: 75

[core  ] DEBUG 2020/09/15 20:07:29 pv power: 0W
[cache ] TRACE 2020/09/15 20:07:29 pvPower: 0
[core  ] DEBUG 2020/09/15 20:07:29 grid power: 374W
[core  ] DEBUG 2020/09/15 20:07:29 site power: 374W
[cache ] TRACE 2020/09/15 20:07:29 gridPower: 374.5
[cache ] TRACE 2020/09/15 20:07:29 mode: minpv
[lp-1  ] DEBUG 2020/09/15 20:07:29 charge power: 0W
[cache ] TRACE 2020/09/15 20:07:29 chargePower: 0
[cache ] TRACE 2020/09/15 20:07:29 chargeCurrent: 9
[cache ] TRACE 2020/09/15 20:07:29 chargedEnergy: 0
[lp-1  ] DEBUG 2020/09/15 20:07:29 vehicle soc: 67%
[cache ] TRACE 2020/09/15 20:07:29 chargeDuration: 0s
[cache ] TRACE 2020/09/15 20:07:29 socCharge: 67
[cache ] TRACE 2020/09/15 20:07:29 chargeEstimate: -1ns
[lp-1  ] DEBUG 2020/09/15 20:07:29 charger status: B
[cache ] TRACE 2020/09/15 20:07:29 connected: true
[cache ] TRACE 2020/09/15 20:07:29 charging: false
[lp-1  ] DEBUG 2020/09/15 20:07:29 max charge current: 0A = 0A + -1A (374W @ 3p)
[lp-1  ] DEBUG 2020/09/15 20:07:29 target charge current: 9A
[core  ] DEBUG 2020/09/15 20:07:39 ----
[core  ] DEBUG 2020/09/15 20:07:39 pv power: 0W
[cache ] TRACE 2020/09/15 20:07:39 pvPower: 0
[core  ] DEBUG 2020/09/15 20:07:39 grid power: 5266W
[core  ] DEBUG 2020/09/15 20:07:39 site power: 5266W
[cache ] TRACE 2020/09/15 20:07:39 gridPower: 5265.6
[cache ] TRACE 2020/09/15 20:07:39 mode: minpv
[lp-1  ] DEBUG 2020/09/15 20:07:39 charge power: 4896W
[cache ] TRACE 2020/09/15 20:07:39 chargePower: 4896.1015625
[lp-1  ] DEBUG 2020/09/15 20:07:39 vehicle soc: 67%
[cache ] TRACE 2020/09/15 20:07:39 chargeCurrent: 9
[cache ] TRACE 2020/09/15 20:07:39 chargedEnergy: 0
[cache ] TRACE 2020/09/15 20:07:39 chargeDuration: 0s
[cache ] TRACE 2020/09/15 20:07:39 socCharge: 67
[cache ] TRACE 2020/09/15 20:07:39 chargeEstimate: -1ns
[lp-1  ] DEBUG 2020/09/15 20:07:39 charger status: C
[lp-1  ] ERROR 2020/09/15 20:07:40 charge meter error read failed: read tcp 192.168.178.47:35706->192.168.178.90:8899: i/o timeout
[lp-1  ] INFO 2020/09/15 20:07:40 start charging ->
[cache ] TRACE 2020/09/15 20:07:40 chargeCurrent: 9
[cache ] TRACE 2020/09/15 20:07:40 connected: true
[cache ] TRACE 2020/09/15 20:07:40 charging: true
[lp-1  ] DEBUG 2020/09/15 20:07:40 max charge current: 1A = 9A + -8A (5266W @ 3p)
[lp-1  ] DEBUG 2020/09/15 20:07:40 target charge current: 9A
[core  ] DEBUG 2020/09/15 20:07:49 ----
[core  ] DEBUG 2020/09/15 20:07:49 pv power: 0W
[cache ] TRACE 2020/09/15 20:07:49 pvPower: 0
[... 39 lines cut ...]
[cache ] TRACE 2020/09/15 20:10:30 connectedDuration: 3m31.040835868s
[cache ] TRACE 2020/09/15 20:10:30 chargeCurrent: 9
[cache ] TRACE 2020/09/15 20:10:30 connected: false
[cache ] TRACE 2020/09/15 20:10:30 charging: false
[lp-1  ] DEBUG 2020/09/15 20:10:30 charger disable - contactor delay 6m54s
[core  ] DEBUG 2020/09/15 20:10:39 ----
[core  ] DEBUG 2020/09/15 20:10:39 pv power: 0W
[cache ] TRACE 2020/09/15 20:10:39 pvPower: 0
[core  ] DEBUG 2020/09/15 20:10:39 grid power: 486W
[core  ] DEBUG 2020/09/15 20:10:39 site power: 486W
[cache ] TRACE 2020/09/15 20:10:39 gridPower: 485.90000000000003
[cache ] TRACE 2020/09/15 20:10:39 mode: minpv
[lp-1  ] DEBUG 2020/09/15 20:10:39 charge power: 0W
[cache ] TRACE 2020/09/15 20:10:39 chargePower: 0
[cache ] TRACE 2020/09/15 20:10:39 chargeCurrent: 9
[cache ] TRACE 2020/09/15 20:10:39 chargedEnergy: 0
[lp-1  ] DEBUG 2020/09/15 20:10:39 vehicle soc: 68%
[cache ] TRACE 2020/09/15 20:10:39 chargeDuration: 2m50s
[cache ] TRACE 2020/09/15 20:10:39 socCharge: 68
[cache ] TRACE 2020/09/15 20:10:39 chargeEstimate: -1ns
[lp-1  ] DEBUG 2020/09/15 20:10:39 charger status: A
[lp-1  ] DEBUG 2020/09/15 20:10:39 charger disable - contactor delay 6m45s
[cache ] TRACE 2020/09/15 20:10:39 connected: false
[cache ] TRACE 2020/09/15 20:10:39 charging: false
[core  ] DEBUG 2020/09/15 20:10:49 ----

andig, 2020-09-16:
Ich sehe in dem Log noch kein Problem:

    [cache ] TRACE 2020/09/15 20:10:30 chargedEnergy: 9.5725712890625e+06
    [lp-1 ] INFO 2020/09/15 20:10:30 set target soc: 75
    [cache ] TRACE 2020/09/15 20:10:30 connectedDuration: 3m31.040835868s
    [cache ] TRACE 2020/09/15 20:10:30 chargeCurrent: 9
    [cache ] TRACE 2020/09/15 20:10:30 connected: false

-> hier wird abgestöpselt

    [cache ] TRACE 2020/09/15 20:10:39 chargedEnergy: 0
    [lp-1 ] DEBUG 2020/09/15 20:10:39 vehicle soc: 68%

-> danach ist die Energie wieder Null

Könntest Du ein Log erstellen und kommentieren in dem man Dein konkretes Problem sieht?

22kW, 2020-09-16:
Dann ist einfach meine Erwartungshaltung falsch gewesen, ich dachte es sollen die aktuell (also für den laufenden Ladevorgang) geladenen kWh angezeigt werden, nicht der Wert seit Installation des Zählers...

andig, 2020-09-16:
> Dann ist einfach meine Erwartungshaltung falsch gewesen, ich dachte es sollen die aktuell (also für den laufenden Ladevorgang) geladenen kWh angezeigt werden, nicht der Wert seit Installation des Zählers...

Korrekt. Jetzt hab ichs verstanden- Du beschwerst Dich über angezeigtes Total, nicht dass der Wert auf Null geht. Ich schau mal...

andig, 2020-09-20:
ping @22kW ich habe hier möglicherweise das falsche Problem gelöst. Der von Dir bemerkte Fehler ist noch da, richtig?

22kW, 2020-09-20:
Richtig, Fehler ist noch da. Evtl. da es jeweils beim Laden Start und Stopp einen Modbus Kommunikationsfehler mit dem sdm630 gibt ?

andig, 2020-09-20:
Das könnte sein. Der Fehler sollte da ja mal weg. Wie ist Deine Topologie?

22kW, 2020-09-20:
SDM630 und Phoenix Controller am gleichen 5 cm langen RS485 Kabel zum RS485 zu Ethernet Converter (in der Garage)

[image]

andig, 2020-09-20:
> Evtl. da es jeweils beim Laden Start und Stopp einen Modbus Kommunikationsfehler mit dem sdm630 gibt ?

Passiert das immer nur bei start/stop aber nicht im normalen Verlauf? Dann würde ich fast ein elektrisches Problem mit Interferenzen vermuten? Kabel geschirmt?

22kW, 2020-09-20:
Genau, nur bei start/stop sonst nicht. Ist nur Klingeldraht ungeschirmt. Muss ich mal upgraden.

andig, 2020-09-25:
Ich mache heir mal zu während Du Deine Verkabelung checks. Meld Dich gerne wieder wenn das Problem danach immer noch existiert!

closed 2020-09-25 by andig

22kW, 2020-09-25:
Ja ist gut. Es liegt sicherlich daran. Ich hatte die letzten Ladungen sogar mit dem schlechten Kabel eine korrekte Anzeige, vermutlich weil da mal kein Kommunikationsfehler auftrat.

PR #354 (2020-09-18): Allow charge rater and timer to sum over multiple measurement intervals
Fix #345