evcc-io/evcc #348: Audi Restladezeitanzeige verschwunden

schenlap, 2020-09-15:
**Describe the bug**
Die Restladezeit wird nicht mehr berechnet, es erscheint "-" in der GUI

**To Reproduce**
Steps to reproduce the behavior:
1. anstecken
2. laden
3. warten, warten, warten, ...

**Expected behavior**
Restzeit mit aktueller Ladeleistung wird angezeigt 

**EVCC details:**
577df71d0ba40ee972c6b82f14282384e8af4cea

```
[audi  ] TRACE 2020/09/15 10:42:25 https://msg.volkswagen.de/fs-car/bs/batterycharge/v1/Audi/DE/vehicles/WAUXXXXXXXXXXX/charger                 
{"charger":{"settings":{"maxChargeCurrent":{"content":32,"timestamp":"2019-04-13T16:16:27Z"}},"status":{"chargingStatusData":{"chargingMode":{"cont
ent":"AC","timestamp":"2020-09-15T08:36:53Z"},"chargingReason":{"content":"immediate","timestamp":"2020-09-15T08:36:53Z"},"externalPowerSupplyState
":{"content":"available","timestamp":"2020-09-15T08:36:53Z"},"energyFlow":{"content":"on","timestamp":"2020-09-15T08:36:53Z"},"chargingState":{"con
tent":"charging","timestamp":"2020-09-15T08:36:53Z"}},"cruisingRangeStatusData":{"engineTypeFirstEngine":{"content":"typeIsElectric","timestamp":"2
020-09-15T08:38:07Z"},"primaryEngineRange":{"content":21,"timestamp":"2020-09-15T08:38:07Z"}},"ledStatusData":{"ledColor":{"content":"green","times
tamp":"2020-09-15T08:36:53Z"},"ledState":{"content":"blink","timestamp":"2020-09-15T08:36:53Z"}},"batteryStatusData":{"stateOfCharge":{"content":53
,"timestamp":"2020-09-15T08:38:07Z"},"remainingChargingTime":{"content":160,"timestamp":"2020-09-15T08:36:54Z"},"remainingChargingTimeTargetSOC":{"
content":"unsupported","timestamp":"2020-09-15T08:36:54Z"}},"plugStatusData":{"plugState":{"content":"connected","timestamp":"2020-09-15T08:36:53Z"
},"lockState":{"content":"locked","timestamp":"2020-09-15T08:36:53Z"}}}}}                                                                          
[cache ] TRACE 2020/09/15 10:42:25 chargeEstimate: 0s
```
```
[core  ] DEBUG 2020/09/15 10:43:03 ----                                                                                                           
[core  ] DEBUG 2020/09/15 10:43:03 pv power: 1294W                                                                                                
[cache ] TRACE 2020/09/15 10:43:03 pvPower: 1294.1000000000001                                                                                    
[core  ] DEBUG 2020/09/15 10:43:03 grid power: 1497W                                                                                              
[core  ] DEBUG 2020/09/15 10:43:03 site power: 1497W                                                                                              
[cache ] TRACE 2020/09/15 10:43:03 gridPower: 1497                                                                                                
[cache ] TRACE 2020/09/15 10:43:03 mode: minpv                                                                                                    
[wifi  ] TRACE 2020/09/15 10:43:03 http://10.0.0.208/getParameters                                                                                
{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":6,"actualPower":1.3,"duration":380750,"alwaysActive
":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0.14,"mileage":0.8,"meterReading":0.14,"currentP1":5.73,"currentP2":0,"current
P3":0}]}                                                                                                                                          
[lp-1  ] DEBUG 2020/09/15 10:43:03 charge power: 1300W                                                                                            
[cache ] TRACE 2020/09/15 10:43:03 chargePower: 1300                                                                                              
[cache ] TRACE 2020/09/15 10:43:03 chargeCurrent: 6                                                                                               
[wifi  ] TRACE 2020/09/15 10:43:03 http://10.0.0.208/getParameters                                                                                
{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":6,"actualPower":1.3,"duration":380769,"alwaysActive
":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0.14,"mileage":0.8,"meterReading":0.14,"currentP1":5.73,"currentP2":0,"current
P3":0}]}                                                                                                                                          
[wifi  ] TRACE 2020/09/15 10:43:03 energy total: 0.140000, energy evse: 0.140000, delta :0.010000                                                 
[lp-1  ] TRACE 2020/09/15 10:43:03 soc estimated: 53.23% (vehicle: 53.00%)                                                                        
[lp-1  ] DEBUG 2020/09/15 10:43:03 vehicle soc: 53%                                                                                               
[cache ] TRACE 2020/09/15 10:43:03 chargedEnergy: 20.000000000000018                                                                              
[cache ] TRACE 2020/09/15 10:43:03 chargeDuration: 40s                                                                                            
[cache ] TRACE 2020/09/15 10:43:03 socCharge: 53                                                                                                  
[cache ] TRACE 2020/09/15 10:43:03 chargeEstimate: 0s                                   # <--  ??                                                          
[wifi  ] TRACE 2020/09/15 10:43:03 http://10.0.0.208/getParameters                                                                                
{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":6,"actualPower":1.3,"duration":380793,"alwaysActive
":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0.14,"mileage":0.8,"meterReading":0.14,"currentP1":5.73,"currentP2":0,"current
P3":0}]}                                                                                                                                          
[lp-1  ] DEBUG 2020/09/15 10:43:03 charger status: C                                                                                              
[cache ] TRACE 2020/09/15 10:43:03 connected: true                                                                                                
[cache ] TRACE 2020/09/15 10:43:03 charging: true                                                                                                 
[wifi  ] TRACE 2020/09/15 10:43:03 http://10.0.0.208/getParameters                                                                                
{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":6,"actualPower":1.3,"duration":380813,"alwaysActive
":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0.14,"mileage":0.8,"meterReading":0.14,"currentP1":5.73,"currentP2":0,"current
P3":0}]}                                                                                                                                          
[wifi  ] TRACE 2020/09/15 10:43:03 http://10.0.0.208/getParameters                                                                                
{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":6,"actualPower":1.3,"duration":380831,"alwaysActive
":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0.14,"mileage":0.8,"meterReading":0.14,"currentP1":5.73,"currentP2":0,"current
P3":0}]}                                                                                                                                          
[lp-1  ] TRACE 2020/09/15 10:43:03 charge currents: [5.73 0 0]A                                                                                   
[lp-1  ] DEBUG 2020/09/15 10:43:03 detected phases: 1p [5.73 0 0]A                                                                                
[lp-1  ] DEBUG 2020/09/15 10:43:03 max charge current: 0A = 6A + -7A (1497W @ 1p) 
```

**If using Docker:**
Show output of `docker run andig/evcc -v`
Show evcc log output with `docker logs <container id>`

premultiply, 2020-09-15:
_"remainingChargingTime":{"content":160,"timestamp":"2020-09-15T08:36:54Z"}_

🤔

schenlap, 2020-09-15:
Ich habe jetzt noch etwas geforscht :-).
Sobald ich in der GUI den TargetSoc auswähle wird die Zeit angezeigt, davor nicht. Komisch...

schenlap, 2020-09-15:
die Verwendung der remainingChargingTime der API ist eine wesntliche Verschlechterung als bisher. Auto beginnt mit PV zu laden, anfangs mit 6A. remainingChargingTime der API wird ausgelesen, ist 160min. PV Leistung steigt, Auto aldet mit 10A, es wird aber bis zum nächsten Update der API (bei mir 30min) der alte Wert angezeigt.

Das war mit Einführung des estimatedSoc viel besser, da habe ich immer gut gesehen wie lange es mit der aktuellen Leistung noch dauern wird. Mit dem aktuellen Wert kann ich nichts anfangen. Er ist veraltet und bezeiht sich auf eine unbekannte Ladeleistung..
Kann man das konfigurieren dass remainingChargingTime der API nicht verwendet wird?

premultiply, 2020-09-15:
Eigentlich **sollte** die Restladezeit auch bei Verwendung der Vehicle-API mit dem Fortschreiten des Ladevorgangs auch weiter runterzählen. Bei mir (Renault-API) tut sie das auch jede Minute - aber nur wenn das Ladeziel auf 100% steht, sonst wird nur noch --- angezeigt. Irgendwas ist da wohl noch faul...

andig, 2020-09-15:
Bei mir ist alles da:

[image]

ich frage mich allerdings langsam ob uns diese grusseligen APIs überhaupt irgendeinen Vorteil bringen?

andig, 2020-09-15:
@schenlap kann es sein, dass Dein Auto schon voll ist? Dann wäre nämlich der Returnwert 0.

andig, 2020-09-15:
@schenlap Du könntest mal https://github.com/andig/evcc/pull/350 testen, da sind weitere Debugstatements drin.

andig, 2020-09-15:
@premultiply 

> Bei mir (Renault-API) tut sie das auch jede Minute - aber nur wenn das Ladeziel auf 100% steht, sonst wird nur noch --- angezeigt. Irgendwas ist da wohl noch faul...

Siehe https://github.com/andig/evcc/issues/348#issuecomment-692627174. Da muss es irgendeinen Zusammenhang geben- könntest Du das mit Debugging untermauern?

premultiply, 2020-09-15:
Ich grübele da gerade schon drüber.

schenlap, 2020-09-15:
Ergebnis von #350 
[image]
```
[audi  ] TRACE 2020/09/15 17:29:50 https://msg.volkswagen.de/fs-car/bs/batterycharge/v1/Audi/DE/vehicles/WAUXXXXXXXXXX/charger                 
{"charger":{"settings":{"maxChargeCurrent":{"content":32,"timestamp":"2019-04-13T16:16:27Z"}},"status":{"chargingStatusData":{"chargingMode":{"cont
ent":"AC","timestamp":"2020-09-15T15:16:28Z"},"chargingReason":{"content":"immediate","timestamp":"2020-09-15T15:16:28Z"},"externalPowerSupplyState
":{"content":"available","timestamp":"2020-09-15T15:16:28Z"},"energyFlow":{"content":"on","timestamp":"2020-09-15T15:16:28Z"},"chargingState":{"con
tent":"charging","timestamp":"2020-09-15T15:16:28Z"}},"cruisingRangeStatusData":{"engineTypeFirstEngine":{"content":"typeIsElectric","timestamp":"2
020-09-15T15:26:08Z"},"primaryEngineRange":{"content":0,"timestamp":"2020-09-15T15:26:08Z"}},"ledStatusData":{"ledColor":{"content":"green","times$
amp":"2020-09-15T15:16:28Z"},"ledState":{"content":"blink","timestamp":"2020-09-15T15:16:28Z"}},"batteryStatusData":{"stateOfCharge":{"content":10,
"timestamp":"2020-09-15T15:26:08Z"},"remainingChargingTime":{"content":300,"timestamp":"2020-09-15T15:26:08Z"},"remainingChargingTimeTargetSOC":{"c
ontent":"unsupported","timestamp":"2020-09-15T15:26:08Z"}},"plugStatusData":{"plugState":{"content":"connected","timestamp":"2020-09-15T15:16:28Z"}
,"lockState":{"content":"locked","timestamp":"2020-09-15T15:16:28Z"}}}}}                                                                           
[cache ] TRACE 2020/09/15 17:29:50 chargeEstimate: 0s  
```
```
core  ] DEBUG 2020/09/15 17:31:39 pv power: 462W                                                                                                  
[core  ] DEBUG 2020/09/15 17:31:39 grid power: 1112W                                                                                               
[core  ] DEBUG 2020/09/15 17:31:39 site power: 1112W
[cache ] TRACE 2020/09/15 17:31:39 pvPower: 461.77000000000004
[cache ] TRACE 2020/09/15 17:31:39 gridPower: 1112
[cache ] TRACE 2020/09/15 17:31:39 mode: pv
[wifi  ] TRACE 2020/09/15 17:31:39 http://10.0.0.208/getParameters
{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":6,"actualPower":1.28,"duration":925540,"alwaysActive":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0.33,"mileage":1.8,"meterReading":0.33,"currentP1":5.62,"currentP2":0,"currentP3":0}]}
[lp-1  ] DEBUG 2020/09/15 17:31:39 charge power: 1280W
[cache ] TRACE 2020/09/15 17:31:39 chargePower: 1280
[cache ] TRACE 2020/09/15 17:31:39 chargeCurrent: 6
[wifi  ] TRACE 2020/09/15 17:31:39 http://10.0.0.208/getParameters
{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":6,"actualPower":1.28,"duration":925578,"alwaysActive":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0.33,"mileage":1.8,"meterReading":0.33,"currentP1":5.62,"currentP2":0,"currentP3":0}]}
[wifi  ] TRACE 2020/09/15 17:31:39 energy total: 0.330000, energy evse: 0.330000, delta :0.000000                                                 
[lp-1  ] TRACE 2020/09/15 17:31:39 soc estimated: 10.45% (vehicle: 10.00%)                                                                        
[lp-1  ] DEBUG 2020/09/15 17:31:39 vehicle soc: 10%
[cache ] TRACE 2020/09/15 17:31:39 chargedEnergy: 40.000000000000036
[cache ] TRACE 2020/09/15 17:31:39 chargeDuration: 2m0s
[cache ] TRACE 2020/09/15 17:31:39 socCharge: 10
[cache ] TRACE 2020/09/15 17:31:39 chargeEstimate: 0s
[wifi  ] TRACE 2020/09/15 17:31:39 http://10.0.0.208/getParameters
{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":6,"actualPower":1.28,"duration":925602,"alwaysActive":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0.33,"mileage":1.8,"meterReading":0.33,"currentP1":5.62,"currentP2":0,"currentP3":0}]}
[lp-1  ] DEBUG 2020/09/15 17:31:39 charger status: C
[cache ] TRACE 2020/09/15 17:31:39 connected: true
[cache ] TRACE 2020/09/15 17:31:39 charging: true
[wifi  ] TRACE 2020/09/15 17:31:39 http://10.0.0.208/getParameters
{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":6,"actualPower":1.28,"duration":925631,"alwaysActive":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0.33,"mileage":1.8,"meterReading":0.33,"currentP1":5.62,"currentP2":0,"currentP3":0}]}
[wifi  ] TRACE 2020/09/15 17:31:39 http://10.0.0.208/getParameters
{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":6,"actualPower":1.28,"duration":925666,"alwaysActive":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0.33,"mileage":1.8,"meterReading":0.33,"currentP1":5.62,"currentP2":0,"currentP3":0}]}
[lp-1  ] TRACE 2020/09/15 17:31:39 charge currents: [5.62 0 0]A
[lp-1  ] DEBUG 2020/09/15 17:31:39 detected phases: 1p [5.62 0 0]A
[lp-1  ] DEBUG 2020/09/15 17:31:39 max charge current: 1A = 6A + -5A (1112W @ 1p)                                                                 
[lp-1  ] DEBUG 2020/09/15 17:31:39 site power 1112W >= disable threshold 500W      
```
Von was brauchst du den TRACE? Ich kann es leider nicht golbal einschalten weil dass ws alles zumüllt...
Wenn ich dann auf 100% klicke wird eine Restzeit angezeigt
[image]
und auch in der Debugausgabe erscheint was:
```
cache ] TRACE 2020/09/15 17:35:19 chargedEnergy: 40.000000000000036
[cache ] TRACE 2020/09/15 17:35:19 chargeDuration: 1m40s
[lp-1  ] TRACE 2020/09/15 17:35:19 soc estimated: 10.45% (vehicle: 10.00%)
[lp-1  ] DEBUG 2020/09/15 17:35:19 vehicle soc: 10%
[cache ] TRACE 2020/09/15 17:35:19 socCharge: 10
[cache ] TRACE 2020/09/15 17:35:19 chargeEstimate: 4h56m17.198971525s

```

schenlap, 2020-09-15:
```
[wifi  ] TRACE 2020/09/15 17:47:39 energy total: 1.000000, energy evse: 0.670000, delta :0.010000
[lp-1  ] TRACE 2020/09/15 17:47:39 soc estimated: 17.54% (vehicle: 10.00%)
[cache ] TRACE 2020/09/15 17:47:39 chargedEnergy: 670.0000000000003
[cache ] TRACE 2020/09/15 17:47:39 chargeDuration: 14m0s
[lp-1  ] DEBUG 2020/09/15 17:47:39 vehicle soc: 17%
[cache ] TRACE 2020/09/15 17:47:39 socCharge: 17
[cache ] TRACE 2020/09/15 17:47:39 chargeEstimate: 4h56m17.198971525s
```
und bleibt bei 4h56 stehen

schenlap, 2020-09-15:
sorry, ich sehe gerade ich war da noch auf master...., ich teste gleich nochmal

andig, 2020-09-15:
Mhhm, da fehlen alle neuen Log Statement- es sieht aus als würde der "Estimator" bei Dir gar nicht aufgerufen werden. Da Du auch sonst ein paar komische Statements hast:

    energy total: 1.000000, energy evse: 0.670000, delta :0.010000

müsstest Du mal selber debuggen warum das nicht passiert.

andig, 2020-09-15:
zu langsam...

schenlap, 2020-09-15:
```
[audi  ] TRACE 2020/09/15 17:50:16 https://msg.volkswagen.de/fs-car/bs/batterycharge/v1/Audi/DE/vehicles/WAUZXXXXXXX/charger                 
{"charger":{"settings":{"maxChargeCurrent":{"content":32,"timestamp":"2019-04-13T16:16:27Z"}},"status":{"chargingStatusData":{"chargingMode":{"cont
ent":"AC","timestamp":"2020-09-15T15:33:48Z"},"chargingReason":{"content":"immediate","timestamp":"2020-09-15T15:33:48Z"},"externalPowerSupplyState
":{"content":"available","timestamp":"2020-09-15T15:33:48Z"},"energyFlow":{"content":"on","timestamp":"2020-09-15T15:33:48Z"},"chargingState":{"con
tent":"charging","timestamp":"2020-09-15T15:33:48Z"}},"cruisingRangeStatusData":{"engineTypeFirstEngine":{"content":"typeIsElectric","timestamp":"2
020-09-15T15:46:41Z"},"primaryEngineRange":{"content":4,"timestamp":"2020-09-15T15:46:41Z"}},"ledStatusData":{"ledColor":{"content":"green","timest
amp":"2020-09-15T15:33:48Z"},"ledState":{"content":"blink","timestamp":"2020-09-15T15:33:48Z"}},"batteryStatusData":{"stateOfCharge":{"content":20,
"timestamp":"2020-09-15T15:46:41Z"},"remainingChargingTime":{"content":100,"timestamp":"2020-09-15T15:46:41Z"},"remainingChargingTimeTargetSOC":{"$
ontent":"unsupported","timestamp":"2020-09-15T15:46:41Z"}},"plugStatusData":{"plugState":{"content":"connected","timestamp":"2020-09-15T15:33:48Z"}
,"lockState":{"content":"locked","timestamp":"2020-09-15T15:33:48Z"}}}}}                                                                           
[lp-1  ] FATAL 2020/09/15 17:50:16 RemainingChargeDuration: timeRemaining=1h36m24.449836412s                                                       
[lp-1  ] FATAL 2020/09/15 17:50:16 RemainingChargeDuration: timeRemaining for soc=0s                                                               
[cache ] TRACE 2020/09/15 17:50:16 chargeEstimate: 0s   
```

schenlap, 2020-09-15:
das war jetzt noch bei 95% soc

schenlap, 2020-09-15:
[image]

schenlap, 2020-09-15:
bei 100% target soc wird Restzeit angezeigt (aber eine viel zu kleine) und trace sieht so aus:
```
lp-1  ] TRACE 2020/09/15 17:53:11 soc estimated: 20.79% (vehicle: 20.00%)                                                                         
[lp-1  ] DEBUG 2020/09/15 17:53:11 vehicle soc: 20%                                                                                                
[lp-1  ] FATAL 2020/09/15 17:53:11 RemainingChargeDuration: timeRemaining=1h36m24.449836412s                                                       
[cache ] TRACE 2020/09/15 17:53:11 chargedEnergy: 69.99999999999996                                                                                
[lp-1  ] FATAL 2020/09/15 17:53:11 RemainingChargeDuration: timeRemaining for soc=1h36m24.449836412s                                               
[cache ] TRACE 2020/09/15 17:53:11 chargeDuration: 3m0s                                                                                            
[cache ] TRACE 2020/09/15 17:53:11 socCharge: 20                                                                                                   
[cache ] TRACE 2020/09/15 17:53:11 chargeEstimate: 1h36m24.449836412s   
```

schenlap, 2020-09-15:
`{"content":100,"timestamp":"2020-09-15T15:46:41Z"}` das ist interessant, das ist ein uralter Wert, jetzt ist es 17:55. Dazwischen habe ich den gesamten Akku leergefahren.... Werte sind also unbrauchbar. In einer halben Stunde paßt der Wert dann vielleicht wieder, mal sehen

schenlap, 2020-09-15:
Restladezeit bleibt bei 1h36 stehen
```
lp-1  ] DEBUG 2020/09/15 17:57:41 charge power: 1300W
[cache ] TRACE 2020/09/15 17:57:41 chargePower: 1300
[cache ] TRACE 2020/09/15 17:57:41 chargeCurrent: 6
[wifi  ] TRACE 2020/09/15 17:57:41 http://10.0.0.208/getParameters
{"type":"parameters","list":[{"vehicleState":3,"evseState":true,"maxCurrent":16,"actualCurrent":6,"actualPower":1.3,"duration":1443398,"alwaysActive":true,"lastActionUser":"vehicle","lastActionUID":"vehicle","energy":0.97,"mileage":5.4,"meterReading":0.97,"currentP1":5.73,"currentP2":0,"currentP3":0}]}
[lp-1  ] TRACE 2020/09/15 17:57:41 soc estimated: 21.80% (vehicle: 20.00%)                                                                        
[lp-1  ] DEBUG 2020/09/15 17:57:41 vehicle soc: 21%
[lp-1  ] FATAL 2020/09/15 17:57:41 RemainingChargeDuration: timeRemaining=1h36m24.449836412s                                                      
[cache ] TRACE 2020/09/15 17:57:41 chargedEnergy: 159.99999999999991
[cache ] TRACE 2020/09/15 17:57:41 chargeDuration: 7m30s
[cache ] TRACE 2020/09/15 17:57:41 socCharge: 21
[lp-1  ] FATAL 2020/09/15 17:57:41 RemainingChargeDuration: timeRemaining for soc=1h36m24.449836412s                                              
[cache ] TRACE 2020/09/15 17:57:41 chargeEstimate: 1h36m24.449836412s
```

schenlap, 2020-09-15:
timestamp ist wohl UTC, dann paßt der Timestamp schon.

schenlap, 2020-09-15:
ich glaube ein Fehler ist mal hier:
`res := timeRemaining * time.Duration(percentRemaining/(100-s.socCharge))` Es scheint hier dass timeDuration einen int Wert nimmt, ist also immer = 0 wenn targetSoc < 100 ist. Ich hab's aber nicht geschafft zu lösen.

andig, 2020-09-15:
> Restladezeit bleibt bei 1h36 stehen

Das log ist aussagelos. Was sagt das API zu dem Zeitpunkt? Hast Du API Caching ausgemacht (cache Parameter)?

> Es scheint hier dass timeDuration einen int Wert nimmt, ist also immer = 0 wenn targetSoc < 100 ist. Ich hab's aber nicht geschafft zu lösen.

In der Formel ist kein TargetSoC drin? Mach gerne 2 Gleichungen draus und logge die Werte, dann siehst Du was los ist.

premultiply, 2020-09-15:
Nicht schlecht.
Warum überhaupt TimeDuration bei einem Faktor?

schenlap, 2020-09-15:
sonst kommt wohl `mismatched types float64 and time.Duration`
`res := timeRemaining * time.Duration(percentRemaining/(100-s.socCharge) * 1000) / time.Duration(1000)` wäre eine Lösung. Schön ist was anderes...

```
[lp-1  ] TRACE 2020/09/15 18:29:29 soc estimated: 30.34% (vehicle: 30.00%)
[lp-1  ] DEBUG 2020/09/15 18:29:29 vehicle soc: 30% # -> habe ich hinzugefügt
[lp-1  ] FATAL 2020/09/15 18:29:29 targetSoc=95  # -> habe ich hinzugefügt
[cache ] TRACE 2020/09/15 18:29:29 chargedEnergy: 30.00000000000003
[cache ] TRACE 2020/09/15 18:29:29 chargeDuration: 1m20s
[cache ] TRACE 2020/09/15 18:29:29 socCharge: 30
[lp-1  ] FATAL 2020/09/15 18:29:29 socCharge=30.3375
[lp-1  ] FATAL 2020/09/15 18:29:29 RemainingChargeDuration: timeRemaining=3h33m6.712644516s
[lp-1  ] FATAL 2020/09/15 18:29:29 RemainingChargeDuration: timeRemaining for soc=3h17m46.06933411s
[cache ] TRACE 2020/09/15 18:29:29 chargeEstimate: 3h17m46.06933411s
```

schenlap, 2020-09-15:
>In der Formel ist kein TargetSoC drin? Mach gerne 2 Gleichungen draus und logge die Werte, dann siehst Du was los ist.

doch, im percentRemaining

andig, 2020-09-15:
Versuch mal bitte:

    res := time.Duration(float64(timeRemaining) * (percentRemaining/(100-s.socCharge)))

ich hatte das eigentlich überlegt und die Hoffnung dass es keine Rundungsprobleme gäbe...

schenlap, 2020-09-15:
@andig das Rundungsproblem ist mit #351 behoben. Offen ist dennoch:
1) Die Zeit zählt trotzdem nicht herunter. Innerhalb der Cache-Time bleibt der Wert gleich
2) Bei sich änderder Ladeleistung wird der Wert nicht geändert

andig, 2020-09-15:
> Die Zeit zählt trotzdem nicht herunter. Innerhalb der Cache-Time bleibt der Wert gleich

Bekannte Einschränkung, ja. Steht auf der Liste.

> Bei sich änderder Ladeleistung wird der Wert nicht geändert

Genau- weil hier das Auto die Hoheit hat. Deshalb die offene Frage:

> ich frage mich allerdings langsam ob uns diese grusseligen APIs überhaupt irgendeinen Vorteil bringen?

Netter Versuch- aber macht das irgendwas besser?

andig, 2020-09-15:
Todo: über Nutzung des API entscheiden

premultiply, 2020-09-15:
Jetzt weiss ich auch warum die Restzeit bei mir runter zählt und bei euch nicht: Ich hab die cachetime auf 1 Minute stehen 😇👍

premultiply, 2020-09-15:
Sollte man statt der Restdauer nicht einfach den Endzeitpunkt (und im gleichen Stil auch den Startzeitpunkt) berechnen und die Restzeit ggf. von der UI dann selbst berechnen lassen? Das fände ich wesentlich eleganter und würde auch wohl das Cachingthema umschiffen.

premultiply, 2020-09-15:
Pro API-Nutzung denn: Bei konstanter Ladung ist die Fahrzeugschätzung WESENTLICH genauer als die EVCC-interne Schätzung und wenn es dringend ist braucht man diese Info möglichst exakt. Bei stark schwankender Leistung (PV-Überschuss mit vielen Hausverbrauchern) versagen beide Methoden gleichermaßen und man könnte die Anzeige ebenso gut komplett ausblenden.

andig, 2020-09-15:
Zeitpunkt ist ne gute Idee. Zusätzlich: macht Restzeit überhaupt Sinn oder wollen wir nicht sehen wanns fertig ist???

andig, 2020-09-15:
Damit sollte es gehen- kurzer Test wäre super.

Was meint Ihr- Endezeitpunkt oder Restdauer?

premultiply, 2020-09-15:
Eigentlich Endzeitpunkt aber mit einer Uhrzeit (+Datum) wirst du wohl UI-Probleme bekommen wenn es länger als 24 Stunden dauern soll.
Daher vielleicht Restdauer anzeigen und das Enddatum als Mouse-Over (Alt-Text).
Wichtig ist das die Lokalzeit des EVCC-Systems auch per API übertragen wird und damit gerechnet wird, da die Client-Uhr falsch sein kann.

andig, 2020-09-15:
Übertragen wird nur Duration, Zeit macht der Client selber. Das spart jede Menge Ärger...

premultiply, 2020-09-15:
Noch besser. ;)
Dann bleibt das API gleich.

PR #351 (2020-09-15): Fix remaining charge duration rounding issue
Fix #348
PR #352 (2020-09-16): Refactor ChargeRemainder into ChargeFinishTimer
Fix #348
closed 2020-09-16 by andig