evcc-io/evcc #363: Error 500 with Renault SOC

ttjaden, 2020-09-28:
**Describe the bug**
Getting an error every few seconds in the Web UI
[image]

**EVCC details:**

Show output of `evcc -v`

> evcc version 0.28 (7eb85ad)

Show evcc log output with `--log trace`

> Sep 28 11:19:20 oeko-t evcc[913]: [core  ] DEBUG 2020/09/28 11:19:20 ----
> Sep 28 11:19:20 oeko-t evcc[913]: [core  ] DEBUG 2020/09/28 11:19:20 pv power: -1549W
> Sep 28 11:19:20 oeko-t evcc[913]: [core  ] DEBUG 2020/09/28 11:19:20 grid power: 4049W
> Sep 28 11:19:20 oeko-t evcc[913]: [core  ] DEBUG 2020/09/28 11:19:20 site power: 4049W
> Sep 28 11:19:20 oeko-t evcc[913]: [lp-1  ] DEBUG 2020/09/28 11:19:20 charge power: 4920W
> Sep 28 11:19:20 oeko-t evcc[913]: [lp-1  ] WARN 2020/09/28 11:19:20 updating soc failed: unexpected response 500: {"type":"TECHNICAL","messages":[{"code":"err.tech.500","message":"{\"errors\":[{\"status\":\"Internal Server Error\",\"code\":\"500\",\"title\":\"Unexpected error\"}]}"
> Sep 28 11:19:20 oeko-t evcc[913]: [lp-1  ] DEBUG 2020/09/28 11:19:20 vehicle soc: 23%
> Sep 28 11:19:20 oeko-t evcc[913]: [lp-1  ] WARN 2020/09/28 11:19:20 updating remaining time failed: estimate not available
> Sep 28 11:19:20 oeko-t evcc[913]: [lp-1  ] DEBUG 2020/09/28 11:19:20 charger status: C
> Sep 28 11:19:20 oeko-t evcc[913]: [lp-1  ] DEBUG 2020/09/28 11:19:20 detected phases: 3p [8 8 8]A
> Sep 28 11:19:20 oeko-t evcc[913]: [lp-1  ] DEBUG 2020/09/28 11:19:20 max charge current: 3A = 9A + -6A (4049W @ 3p)
> Sep 28 11:19:20 oeko-t evcc[913]: [lp-1  ] DEBUG 2020/09/28 11:19:20 target charge current: 9A
> Sep 28 11:19:22 oeko-t influxd[1069]: [httpd] ::1 - - [28/Sep/2020:11:19:21 +0200] "POST /api/v2/write?bucket=evcc&org=&precision=ns HTTP/1.1" 204 0 "-" "influxdb-client-go/2.0.1  (linux; amd64)" b14b94dd-016b-11eb-a1f1-00163ccedce1 899861
> Sep 28 11:19:30 oeko-t evcc[913]: [core  ] DEBUG 2020/09/28 11:19:30 ----
> Sep 28 11:19:30 oeko-t evcc[913]: [core  ] DEBUG 2020/09/28 11:19:30 pv power: -1552W
> Sep 28 11:19:30 oeko-t evcc[913]: [core  ] DEBUG 2020/09/28 11:19:30 grid power: 4040W
> Sep 28 11:19:30 oeko-t evcc[913]: [core  ] DEBUG 2020/09/28 11:19:30 site power: 4040W
> Sep 28 11:19:31 oeko-t evcc[913]: [lp-1  ] DEBUG 2020/09/28 11:19:31 charge power: 4920W
> Sep 28 11:19:31 oeko-t evcc[913]: [lp-1  ] WARN 2020/09/28 11:19:31 updating soc failed: unexpected response 500: {"type":"TECHNICAL","messages":[{"code":"err.tech.500","message":"{\"errors\":[{\"status\":\"Internal Server Error\",\"code\":\"500\",\"title\":\"Unexpected error\"}]}"
> Sep 28 11:19:31 oeko-t evcc[913]: [lp-1  ] DEBUG 2020/09/28 11:19:31 vehicle soc: 23%
> Sep 28 11:19:31 oeko-t evcc[913]: [lp-1  ] WARN 2020/09/28 11:19:31 updating remaining time failed: estimate not available
> Sep 28 11:19:31 oeko-t evcc[913]: [lp-1  ] DEBUG 2020/09/28 11:19:31 charger status: C
> Sep 28 11:19:31 oeko-t evcc[913]: [lp-1  ] DEBUG 2020/09/28 11:19:31 detected phases: 3p [8.1 8 8.1]A
> Sep 28 11:19:31 oeko-t evcc[913]: [lp-1  ] DEBUG 2020/09/28 11:19:31 max charge current: 3A = 9A + -6A (4040W @ 3p)
> Sep 28 11:19:31 oeko-t evcc[913]: [lp-1  ] DEBUG 2020/09/28 11:19:31 target charge current: 9A

premultiply, 2020-09-28:
Das hab ich auch schon mal gehabt. Da fehlte dann irgendwie die VIN in den URLs. Sprich sie war ein leerer String.
Nach einem EVCC-Neustart ging es dann sofort wieder.

Vermutlich muss dann die VIN-Ermittlung noch einen zusätzlichen Check haben und diese ggf. wiederholt werden.

ttjaden, 2020-09-28:
> Das hab ich auch schon mal gehabt. Da fehlte dann irgendwie die VIN in den URLs. Sprich sie war ein leerer String.
> Nach einem EVCC-Neustart ging es dann sofort wieder.
> 
> Vermutlich muss dann die VIN-Ermittlung noch einen zusätzlichen Check haben und diese ggf. wiederholt werden.

Habe den Services mal neugestartet. Danach kommt die lange Fehlermeldung nicht mehr, jedoch noch die folgende:
[image]

andig, 2020-09-28:
Also- das sind mehrere Themen. Lasst  uns mal  sortieren:

> Das hab ich auch schon mal gehabt. Da fehlte dann irgendwie die VIN in den URLs. Sprich sie war ein leerer String.
Nach einem EVCC-Neustart ging es dann sofort wieder.

Standardfrage: ist die VIN in der config mit angegeben? Falls nein: was steht im Logfile (Level Warnung) bzgl.  ermittelter VIN? Der Fehler dürfte nur auftauchen falls keine ermittelt werden kann-  dann müsste EVCC aber schon beim Start abbrechen. Falls nein wäre  das ein Bug. Für Diagnose bitte `--log trace` und vollständigen Output im Fehlerfall per Mail an mich (cpuidle@gmx.de)

> Getting an error every few seconds in the Web UI

Kann es sein,  dass Du kein `cache`  bei Auto angegeben hast?  Solltes Du unbedingt um  das  Renault API nicht zu überlasten. Mit Cache wird  allerdings- falls es einen Fehler  gab- dieser für die Cachdauer erneut angezeigt. Evtl. sollten wir das ändern?

> Danach kommt die lange Fehlermeldung nicht mehr, jedoch noch die folgende:

Das Renault API liefert nur *manchmal* Ladezeitprognosen. Wenn  keine  da ist  geben wir diese Fehlermeldung aus damit der Anwender sich dessen bewusst ist. Tatsächlich ist das nicht so schön da wir ohnehin Fallback auf die interne Prognose eingebaut  haben. Mein Vorschlag wäre diese spezifische Fehlermeldung zu unterdrücken.

ttjaden, 2020-09-28:
> Standardfrage: ist die VIN in der config mit angegeben?

--> ja

> Kann es sein, dass Du kein `cache` bei Auto angegeben hast? Solltes Du unbedingt um das Renault API nicht zu überlasten. Mit Cache wird allerdings- falls es einen Fehler gab- dieser für die Cachdauer erneut angezeigt. Evtl. sollten wir das ändern?
> 
--> doch, cache ist bei 5min

```
- name: renault
  type: renault
  title: Zoe
  capacity: 52 # kWh
  user: *******@gmail.com # user
  password: ********** # password
  region: de_DE # gigya region
  vin: VF1*************
  cache: 5m
```
 

> Das Renault API liefert nur _manchmal_ Ladezeitprognosen. Wenn keine da ist geben wir diese Fehlermeldung aus damit der Anwender sich dessen bewusst ist. Tatsächlich ist das nicht so schön da wir [...]

Ja, das zu unterdrücken ist sicherlich für "einfache Nutzer" wie mich, hilfreich :+1:

premultiply, 2020-09-28:
Ich glaube noch nicht dass das mehrere Themen sind. Bei mir war das exakt der gleiche Fall wie hier. Die oben genannte Fehlermeldung hält sich endlos ran.

Im Log findet man dazu diese Einträge:
```
[renault] TRACE 2020/09/25 16:19:27 https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/297e6f7b-f6ca-4839-b7e1-7409e0ab38a5/kamereon/kca/car-adapter/v2/cars//battery-status?country=DE
{"type":"TECHNICAL","messages":[{"code":"err.tech.wired.unknown-error","message":"An unknown error happened"}],"errors":[{"errorCode":"err.tech.wired.unknown-error","errorMessage":"An unknown error happened"}],"error_reference":"TECHNICAL"}
```

Wie man an dem **//** hinter **cars** sieht fehlt da was in der URL.

andig, 2020-09-28:
@premultiply vin bei Dir angegeben oder autodetect?

premultiply, 2020-09-28:
Äh, sorry. Autodetect.

andig, 2020-09-28:
> Äh, sorry. Autodetect.

Also doch anderes Thema ;) Das ist in https://github.com/andig/evcc/commit/d73c374577856a44850a93888cc5b969708cc162 Behoben.

andig, 2020-09-28:
> Ja, das zu unterdrücken ist sicherlich für "einfache Nutzer" wie mich, hilfreich 👍

Auch erledigt. Ich glaube hier können wir zumachen?

andig, 2020-09-28:
Bzw. ein Punkt bleibt offen. Wenn es einen echten technischen Fehler gibt- z.B. Renault API „geht nicht“ (wie wohl häufiger) dann würde auch jetzt die Warnung wiederholt angezeigt. Wenn das nicht gewünscht ist- wie sähe dann eine „richtige“ Lösung aus?

premultiply, 2020-09-28:
Wenn es ein "echter Fehler" ist dann spricht meiner Meinung nach nichts gegen die angezeigte Meldung.

ttjaden, 2020-09-28:
> Bzw. ein Punkt bleibt offen. Wenn es einen echten technischen Fehler gibt- z.B. Renault API „geht nicht“ (wie wohl häufiger) dann würde auch jetzt die Warnung wiederholt angezeigt. Wenn das nicht gewünscht ist- wie sähe dann eine „richtige“ Lösung aus?

- Den 500 Fehler, der bei mir nach dem Neustart weg war, sollte natürlich bleiben. 

- Die "Warning: updating remaining time failed: estimate not available" kann meine Meinung nach gerne unterdrückt werden, bzw ggf auf einen kleinen "aufploppenden" Info-Button neben "Restzeit". So hat man Beides: a) Sichtbar die Info, dass die Renault-API keine Daten zur verbleibenden Zeit sendet und b) keine irritierenden Pop-ups oben rechts.

andig, 2020-09-28:
Dem sollte jetzt so sein. Falls nein bitte nochmal melden.

closed 2020-09-28 by andig