evcc-io/evcc #364: Lademodus "Extern"

premultiply, 2020-09-28:
Ich schlage vor einen zusätzlichen Lademodus mit dem Arbeitstitel "Extern" oder "Automatik" einzubauen.

Dieser soll eine einheitliche interne Schnittstelle für alle externen Steuerungsmöglichkeiten für Energiemanager wie z. B. prognosebasierte Systeme wie den SMA Sunny Home Manager, über eigene Hausautomationen bis hin zum einfachen Schaltkontakt eines Zweitarifzählers bieten.

Die konkreten Möglichkeiten bzw. die Implementierung richtet sich dann nach dem jeweils angebundenen System.

Während bei einem Zähler-Schaltkontakt die Möglichkeiten quasi auf einen automatisch gestarten und beendeten Sofortladen-Modus beschränkt sein dürften kann bei einem Energiemanager zunächst der Energiebedarf eines verbundenen Fahrzeugs gemeldet und dann die Ladung durch diesen freigegeben oder unterbrochen sowie die zulässige Leistungsaufnahme konkret angesteuert werden. So kann z. B. die Priorität der Fahrzeugladung Gegenüber anderen Geräten und Speichern im Haus selbst dort konfiguriert werden und z. B. die Wetterprognose für die Sonneneinstrahlung einbezogen werden.

Mit einer Anbindung von Awattar oder ähnlichen dynamischen Tarifen darüber könnte ähnlich wie beim Zweitarifzähler nur zu günstigen Tarifzeiten geladen werden.

Die anderen Lademodi werden natürlich weiterhin benötigt und überschreiben dann in jedem Fall die externen Vorgaben.

premultiply, 2020-09-28:
#150 #225 #358

andig, 2020-09-28:
Ich bin davon überzeugt, dass wir eine Möglichkeit zur externen Steuerung brauchen- das hatte ich in #358 gesehen. Was wäre hier zusätzlich zu #358? Ich habe bisher allerdings keine gute Idee, wie wir mit externer Steuerung im Kontext der Modi umgehen wollen. Eine Diskussion auf Slack dazu wäre gut.

andig, 2020-12-16:
Mangels konkreter Vorstellung was hier benötigt wird mache ich mal zu.

closed 2020-12-16 by andig

premultiply, 2020-12-18:
Extern -> Volle Kontrolle An/Aus+Ladeleistung durch SHM/Portal
PV -> nur PV-Überschuss
minPV -> Mindestladeleistung + PV-Überschuss
Sofort -> Sofortladung
Stop -> Aus

premultiply, 2020-12-18:
Gewünscht wird ein zusätzlicher Lademodus in dem EVCC die Kontrolle bzgl. An/Aus und der Höhe der momentan erlaubten Ladeleistung dem SHM überlassen wird.