evcc-io/evcc #366: evcc kann nicht gestartet werden, wenn eins der Meter offline (aus) ist

mpucherSmIT, 2020-09-30:
**Describe the bug**
Im Abschnitt Meters der yaml wird die CALC-Funktion für mehrere Modbus-Meter verwendet. Wenn nun eines der Meter (z.B. ein Wechselrichter) nachts aus ist, dann kann evcc nicht neu gestartet werden.

Fehlermeldung:
... FATAL ... cannot create default meter: not a SunSpec device

**Expected behavior**
evcc sollte sich (neu) starten lassen und das offline Meter ignorieren und bei Verfügbarkeit wieder berücksichtigen

andig, 2020-09-30:
OK- das Problem an der Stelle ist, dass wir gar nicht sicher sein können, ob da überhaupt ein Gerät lauscht. Alle Geräte sollten zumindest zum Start von EVCC mal vorhanden sein (im Moment "by design"), das flächig  abzustellen wäre eine  größere  Änderung.

Falls EVCC läuft wird das Log nachts aber nicht  mit  größeren Mengen an Fehlern überschwemmt wenn die WR nicht  da  sind, richtig?

andig, 2020-10-07:
Ich mache hier mal zu. Im Moment keine Lösung geplant- per Design sollten alle Geräte zum Start ansprechbar sein um Konfigurationsfehler auszuschließen.

closed 2020-10-07 by andig