evcc-io/evcc #378: Option to set a "always load to % SoC" regardless of time or PV-Energy

Optic00, 2020-10-08:
**Is your feature request related to a problem? Please describe.**
Since i would like to see EVCC as "set and forget" in day to day using, i would be really happy about having a minimum SoC Load option. I would be really frustrated if i plugged in the car after work with 10% SoC left and the next morning it would still be at 10% SoC because i didn't manually loaded up to 50%, or i did that and the next morning (weekend) the sun would be up but i forgot to change EVCC to load PV into the car since it was still set at 50% and "sofort"

**Describe the solution you'd like**
Option to set a specific minium SoC that will always load as soon as the car is plugged in. Maybe make it 0% by default for people who have another car in case of emergency or live in a city or can live with the risk of having an EV with low battery.

**Describe alternatives you've considered**
Use an external Script to call the API and load to a minium SoC.

**Additional context**
-

andig, 2020-10-08:
OK- es soll also *immer* auf konfigurierbare z.B. 10% geladen werden. D.h.- wenn das Setting immer aktiviert ist- und z.B. das Auto abends dran kommt wollt ihr gerne xkWh einkaufen obwohl ihr genau wisst, dass ihr erst am nächsten Tag Mittags wieder los müsst?

Korrekt?

premultiply, 2020-10-08:
Wenn der Lademodus nicht "Stop" ist.

Optic00, 2020-10-08:
genau. Wobei ich das Beispiel eher auf 40 oder 50% min SoC führen würde, einfach damit man im Notfall/spontan mit dem Auto noch wegkommt und eben nicht mit nur 10% im Akku nur noch 25km weit kommt. Die 10% sollten das Negativbeispiel sein was dann nicht so toll wäre.

andig, 2020-10-08:
Wählen kannst Du per Konfiguration selber. Nur bedeuten eben 50% Minimum im Zweifel auch 50% Netzbezug statt PV-Nutzung. Für mich wäre das Nichts, aber die Einsatzfälle sind sicher unterschiedlich.

userjp123, 2020-10-08:
@Optic00 : Verstehe ich das richtig: EVCC läuft bei dir und du steckst an, **ohne** dass du die Oberfläche anschaust ?

Wäre das nicht diese Einstellung ? (immer wenn abgesteckt, geht er auf sofort mit zB. 30%)

  _onDisconnect: # set defaults when vehicle disconnects
    mode: now 
    targetSoC: 30_

andig, 2020-10-08:
@userjp123 schicke Idee! Du hast recht, genau damit ist das lösbar- allerdings würde danach die Ladung stoppen und nicht wieder auf PV Mode für den Rest springen. Insofern hätte dieser Request/PR schon einen eigenen Nutzen.

userjp123, 2020-10-08:
Ich bekomme meine ZOE erst in 3 Wochen, habe also insofern noch keine praktische Erfahrung. 
Aber ich teste im Trockenmodus schon mal so einige Szenarien durch. (evcc läuft und ich steuere das Verhalten über Textdateien)
Und ich überlege jeden Tag, welche evcc-Einstellung ich denn wählen würde. :-)
Ich komme zu dem Schluss, dass ein Automatismus mir nicht helfen würde. Es gibt soviele Randbedingungen (SOC,Wetter, Wochenende etc.), dass wohl ich immer manuell einstellen werde. 

zum Request:
Zurück auf **Nur PV** hätte bei einem Tag wie heute nix genützt (PV 10kWp, Ost/West, keine Batterie). Die 6A Minimum (1380W) sind heute nicht erreicht worden. Und das wird im Winter wohl noch oft der Fall sein. Wenn man dann an so einem Tag nichts einspeisen will, dann müsste man schon auf **Min+PV** gehen.
Also sollte die Rückfall-Option für den Rest konfigurierbar sein.

Optic00, 2020-10-08:
@userjp123: ein Wintermodus quasi? :-)
@andig: mein usecase ist das der eGolf unser einziges Auto ist und wir recht ländlich wohnen. Müsste ich früh irgendwo hin und hätte 10% SoC wäre das sehr ärgerlich und würde den nächsten CCS aufsuchen. Wenn es dann mal bis 50% über Netzstrom lädt und sonst viel mit Sonne, dann ist das ein für mich ein guter Kompromiss. 

Danke nochmal für die schnelle Umsetzung, ich teste es sobald wie möglich!

PR #379 (2020-10-18): Add soc: min setting to always force-charge to this value
Fix #378

Breaking change: `targetSoC` setting has moved to settings below `soc`:

```
loadpoints:
- name: Main
  soc:
    target: ...
    min: ...
```
closed 2020-10-18 by andig