evcc-io/evcc #382: config für SMA Wechselrichter - SMA Home Manager 2.0

martinez81, 2020-10-23:
Hi!
Ich habe den evcc jetzt zum laufen gebracht. Ohne Docker... das war aber nicht mein Problem.

Ich bin mit dem Wiki und meiner SMA Anlage nicht ganz klar gekommen.
Jetzt funktioniert es. Das PV Laden konnte ich mangels Sonne noch nicht testen.
Hier mein evcc.yaml, vielleicht kann jemand nochmal drüber sehen ob das noch optimiert werden kann.
Ich würde, wenn das soweit okay ist, das Wiki beim Thema SMA noch ein bisschen füllen.

Viele Grüße
Martin

`uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

# log settings
# log: error
# levels:
#   core: debug
#   lp-1: debug
#   lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
- name: sma-grid
  type: sma
  serial: xxxxxxxxxx # insert your own serialnumber

- name: sma-pv
  type: modbus
# value: power
  model: sunspec
  uri: 192.168.0.xxx:502 # IP address, configured port (default is 502)
  id: 126 # Configured Modbus Device ID (SMA default ID 3 = 126)

# - name: battery
# type: ...
# - name: charge
#  type: ...

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/andig/evcc-config#chargers
chargers:
- name: go
  type: go-e # Go-e charger
  uri: http://192.168.0.xxx

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/andig/evcc-config#vehicles
# The client keys can be obtained here: https://pastebin.com/pS7Z6yyP
vehicles:
- name: tesla
  type: tesla
  title: Model 3
  clientid: xxx
  clientsecret: xxx
  user: xxx@xxx.com # your email address used on Tesla website
  password: xxx # your password used on Tesla website password
  vin: xxx # the VIN of you car (can be obtained from Tesla website)
  cache: 5m

# site describes the EVU connection, PV and home battery

site:
  title: Home # display name for UI
  meters:
    grid: sma-grid # grid meter
    pv: sma-pv # pv meter
#   battery: battery # battery meter
#   prioritySoC: 60 # give home battery priority up to this soc (0 to disable)

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
- title: GO-eCharger # display name for UI
  charger: go # charger
#  meters:
#    charge: go # charge meter
  vehicle: tesla
  mode: pv
  soc:
    min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 100 # always charge to 100%
    alwaysUpdate: false # set true to update vehicle soc even when disconnected
    estimate: false # set true to interpolate between api updates
    levels: # target soc levels for UI
    - 30
    - 50
    - 80
    - 100
  onDisconnect: # set defaults when vehicle disconnects
    mode: pv # switch back to pv mode
    targetSoC: 100 # charge to 100%
  phases: 3 # ev phases (default 3)
  sensitivity: 1 # current raise/lower step size (default 10A)
  enable: # pv mode enable behavior
    delay: 20s # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 20s # threshold must be exceeded for this long
    threshold: 400 # maximum import power (W)
  guardduration: 10s # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)

# mqtt message broker
# mqtt:
#  broker: http://192.168.0.xxx:1883
#  topic: evcc # root topic for publishing, set empty to disable
  # user:
  # password:

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>

`

andig, 2020-10-23:
@martinez81 warum probierst Du es nicht einfach aus bzw. was fehlt Dir noch? M.E. Sollten die Beispielkonfigurationen aus evcc-config für SMA ausreihen. Was hat Dir da gefehlt?

martinez81, 2020-10-23:
Ich habe es natürlich ausprobiert. Es funktioniert ja jetzt auch. 👍🏼

Nur war es für mich nicht so leicht einzustellen. Es erfordert schon ein paar "Sachen".
Ich selbst bin in Linux und Programmiersprachen ein Anfänger. Habe aber schon mehrere Jahre Erfahrung mit Raspberrys und vielen OpenSource Projekten.
Bei mir zuhause laufen Loxberry, octoprint, ioBroker, OpenWB, Teslamate, jede Menge ESP8266 (Tasmota & ESPeasy), Arduinos, SAE und ein selbst erstelltes Loxone System. Ich bin also nicht ganz unwissend 🙃
Aber trotzdem hat mich diese evcc.yaml ein wenig geärgert. Nach 3 Abende habe ich sie jetzt besser verstanden.
Das Problem sitzt mit Sicherheit vor diesem Bildschirm. Kein Frage. Ich will nur sagen, dass es für mich leider nicht so leicht war.

**Meine Probleme und wo mir openWB noch besser gefällt:**
_(Das soll hier keine negative Kritik sein, vielmehr möchte ich den Einstieg für Anfänger fördern)_
Ich finde die evcc.yami leider unübersichtlich. Das liegt sicher an meiner mangelnden Erfahrung mit YAML
Richtig cool wäre hier eine kleine Webseite, die eine fertige Datei bereit stellt.

Erst eben ist mir ein weiteres cooles Plugin aufgefallen:
_EVCC can integrate itself with Home Energy Management Systems. At this time, the SMA Home Manager (SHM) is the only supported system. To enable add_

Wie ist das zu verstehen? Steuert jetzt mein Home Manager 2.0 die Wallbox direkt? Ich habe sie mal in das Sunnyportal eingebunden....
**Wie läuft jetzt die Steuerung?**

**Warum ich meine PV und Grid Werte erst nicht hinbekommen habe:**
Hier [https://github.com/andig/evcc-config#meters](url) fehlt der SMA Wechselrichter.
Es gibt nur den SMA Home Manager 2.0 und ein SMA Sonny Island.
Irgendwann bin ich dann auf diese Seite [https://github.com/andig/evcc/wiki/4.-PV-inverter](url) gestoßen.
Da habe ich mir aus dem Eintrag **Two SMA PV inverter using Modbus/TCP Sunspec connection** einen Eintrag für meinen SMA Tripower gebastelt. Bei mir funktioniert aber den Punkt **value: Power** nicht. Den musste ich auskommentieren.
Wenn ich den in der Config lasse kommt dieser Fehler:
`[main  ] FATAL 2020/10/23 19:35:59 cannot create modbus meter: 1 error(s) decoding:

* '' has invalid keys: value`

Dieser Eintrag [https://github.com/andig/evcc/wiki/3.2.-SMA-Home-Manager-2.0---SMA-Energy-Meter](url) hat mich total irritiert.
Als 2. Punkt wird hier beschrieben, dass man über
`- name: sma-pv
  type: sma
  serial: 1234567890 # Serial number of the device, if this is defined uri is not needed!
  uri: 192.168.1.4 # IP address of the device, if this is defined serial is not needed!`
das PV Meter setzen kann. Das geht aber nicht. Er bricht dann immer ab, oder schreibt die Grid-Werte bei PV rein.

Das Plugin für unseren Nissan Leaf ZE1 BJ2020 geht nicht. Das ist aber leider überall so. Carwings wird bei den BJ2020 nicht mehr unterstütz. In der Nissan Connect App (eine neue) muss man sich jetzt per E-Mail Adresse anmelden. Über Carwings ging das über einen Benutzernamen. Das hat also nur zweitrangig mit evcc zu tun. Nur wäre es bei diesen PKW echt wichtig, da er kein SOC Limit von der Fahrzeug Seite kennt. Er lädt die Kiste immer auf 100% voll.

**Was mir super und besser als bei openWB gefällt.**

Die GUI! Super schlicht und schön.
Eine Kleinigkeit: Es wäre toll, wenn man einen Darkscreen setzten könnte.

Viele Grüße
Martin

andig, 2020-10-24:
Moin Martin,

erstmal: vielen Dank für Dein Feedback! Es hilft immer zu verstehen welche Probleme Anwender haben. Ich kann Dir allerdings nicht in allen Punkten folgen. Damit es übersichtlich bleibt wäre es normalerweise auch hilfreich, die Themen einzeln per Ticket zu machen damit es nciht so durcheinander geht.

> (Das soll hier keine negative Kritik sein, vielmehr möchte ich den Einstieg für Anfänger fördern)

Hab ich so verstanden, genau!

> Wie ist das zu verstehen? Steuert jetzt mein Home Manager 2.0 die Wallbox direkt? Ich habe sie mal in das Sunnyportal eingebunden....

Aktuell meldet EVCC seinen Bedarf an den SHM damit dieser planen kann, lässt sich aber nciht durch den SHM kontrollieren Dafür siehe https://github.com/andig/evcc/pull/365

> Warum ich meine PV und Grid Werte erst nicht hinbekommen habe:
Hier https://github.com/andig/evcc-config#meters fehlt der SMA Wechselrichter. Es gibt nur den SMA Home Manager 2.0 und ein SMA Sonny Island.

Ah, ok! Der Sunny Island sollte analog funktionieren, für PV brauchst Du eben nur keine Energie als Messwert- die ist nur beim Netzzähler relevant. Hast Du mal das ausprobiert:

```yaml
- type: modbus
  model: sunny-island
  uri: 192.168.1.4:502
  id: 126
  power: Power # default values, optionally override
```

> Irgendwann bin ich dann auf diese Seite https://github.com/andig/evcc/wiki/4.-PV-inverter gestoßen. Da habe ich mir aus dem Eintrag Two SMA PV inverter using Modbus/TCP Sunspec connection einen Eintrag für meinen SMA Tripower gebastelt. Bei mir funktioniert aber den Punkt value: Power nicht. Den musste ich auskommentieren.
> Wenn ich den in der Config lasse kommt dieser Fehler: ...
> `[main ] FATAL 2020/10/23 19:35:59 cannot create modbus meter: 1 error(s) decoding:
> '' has invalid keys: value`

Vmtl Schreibfehler im yaml- wie sieht denn Deine Config dann aus? Und warum `value`? Du musst doch den Messwert für Leistung angeben, also `power: Power`, was Du auch genauso weglassen kannst da Default. Woher hast Du das falsche `value`?

> Dieser Eintrag https://github.com/andig/evcc/wiki/3.2.-SMA-Home-Manager-2.0---SMA-Energy-Meter hat mich total irritiert. Als 2. Punkt wird hier beschrieben, dass man über ... type: sma serial/uri: 192.168.1.4 # IP address of the device, if this is defined serial is not needed! das PV Meter setzen kann. Das geht aber nicht. Er bricht dann immer ab, oder schreibt die Grid-Werte bei PV rein.

Das muss ich prüfen. Was heisst "geht nicht"? Bitte konkret? /cc @therealkerni

> Das Plugin für unseren Nissan Leaf ZE1 BJ2020 geht nicht. 

Kennst Du irgendein Skript das das kann? Dann könnte ich abschreiben ;)

> Die GUI! Super schlicht und schön.

Danke :)

> Eine Kleinigkeit: Es wäre toll, wenn man einen Darkscreen setzten könnte.

Momentan nicht geplant, aber Sponsoren (siehe https://github.com/sponsors/andig) bekommen Vorzugsbehandlung ;)

andig, 2020-10-25:
> Das Plugin für unseren Nissan Leaf ZE1 BJ2020 geht nicht. Das ist aber leider überall so.

Meld Dich mal bitte unter cpuidle@gmx.de. Denke das kriegen wir auch hin.

martinez81, 2020-10-25:
> Ah, ok! Der Sunny Island sollte analog funktionieren, für PV brauchst Du eben nur keine Energie als Messwert- die ist nur beim Netzzähler relevant. Hast Du mal das ausprobiert:
>
> ```yaml
> - type: [...]

Ich komme erst jetzt zum Antworten und testen...

Ja, mit den sunny-island Parameter klappt es. Diese lasse ich jetzt auch aktiv.

martinez81, 2020-10-25:
> > Das Plugin für unseren Nissan Leaf ZE1 BJ2020 geht nicht. Das ist aber leider überall so.
> 
> Meld Dich mal bitte unter [cpuidle@gmx.de](mailto:cpuidle@gmx.de). Denke das kriegen wir auch hin.

Okay, ich schreibe dir gleich eine E-Mail.

andig, 2020-10-25:
@martinez81 könntest Du ncohmal schauen wie Du das hässliche "panic" produziert hattest? Das würde ich gerne schöner machen.

martinez81, 2020-10-26:
> @martinez81 könntest Du ncohmal schauen wie Du das hässliche "panic" produziert hattest? Das würde ich gerne schöner machen.

Welche panic meinst du? 😁

andig, 2020-10-26:
Ehrm- ich dachte da wäre eine in Deinem Post gewesen. Jetzt sehe ich sie auch nicht mehr. Oder war das bei GE?

martinez81, 2020-10-26:
Meinst du das?

[image]

andig, 2020-10-26:
Ja genau- welche Config hat das provoziert?

TheRealKerni, 2020-10-29:
Hi, ich verstehe das Problem mit dem SMA EM oder HM nicht ganz.

- Gib entweder die Seriennummer ODER die IP an, nicht beides. Präferiert die Seriennummer weil auch SMA das so benutzt.
- Um welches Gerät genau handelt es sich?
- Wie ist das Gerät angeschlossen? Am Grid um den ganzen Netzbezug/Überschuß zu messen oder nur am PV?
- Hast du in der `site:` Konfiguration dann auch unter `meters:` in `grid:` den Meter angegeben oder unter `pv:`?

martinez81, 2020-10-29:
Es läuft jetzt auch bei mir.
Die Panik hatte ich nur mit den Docker auf mein Raspberry. Ich denke ich habe den config (evcc.yaml) hier nicht richtig bekannt gegeben.
Dazu hätte ich eine Frage: 
Wenn ich den evcc per Docker laufen lasse, also: **docker run -v $(pwd)/evcc.yaml:/etc/evcc.yaml -p 7070:7070 andig/evcc**
Wie und wo muss ich dann die evcc.yaml editieren? Das erschließt sich mir nicht. Liegt sie im /etc/ also: **sudo nano /etc/evcc.yaml** Oder liegt sie auch im Docker image? Wenn ja, wie kann sie editieren?

martinez81, 2020-10-29:
> Hi, ich verstehe das Problem mit dem SMA EM oder HM nicht ganz.
>
> * Gib entweder die Seriennummer ODER die IP an, nicht beides. Präferiert die Seriennummer weil auch SMA das so benutzt.
> * Um welches [...]

Meine Config sieht jetzt so aus:
`meters:
- name: sma-grid
  type: sma
  serial: 30061xxxxx

- name: sma-pv
  type: modbus
  model: sunny-island
  uri: 192.168.0.165:502
  id: 126
  power: Power # default values, optionally override
`

ich war bei meiner ursprünglichen Problem / Frage davon ausgegangen das der Home Manager 2.0 den Grid und PV überträgt. Ich habe dann dummerweise einmal beim GRID und einmal bei PV Meter die Seriennummer des Home Manager eingetragen.
Ich hätte wohl beim GRID Meter die vom Home Manager und beim PV Meter die vom Wechselrichter eintragen müssen. Stimmt das?

TheRealKerni, 2020-10-29:
Das Gerät stellt nur die Daten zur Verfügung welche es selbst misst. An der Stelle kann es die PV Erzeugung nicht messen, nur den Überschuss.

> Ich hätte wohl beim GRID Meter die vom Home Manager und beim PV Meter die vom Wechselrichter eintragen müssen. Stimmt das?

Korrekt.

premultiply, 2020-10-29:
Deine Konfig sollte dann so aussehen:

```
- name: grid
  type: sma
  serial: 30061xxxxx

- name: pv
  type: modbus
  model: sunspec
  uri: 192.168.0.165:502
  id: 126
  power: Power
```

andig, 2020-10-29:
power: Power ist default und kann entfallen, siehe https://github.com/andig/evcc/blob/master/meter/modbus.go#L38. Das müssen alle Zähler für evcc können. Energy und SoC sind optional und müssten konfiguriert werden, siehe README.

andig, 2020-10-29:
Brauchen wir füe SMA neue/ bessere Beispiele in evcc-config? Z.b. beim Sunny Island einschl. Soc?

premultiply, 2020-10-29:
Ja, ich hab dafür schon einen PR erstellt.

Ok, dann kann Power explizit noch weg.
Was genau macht nochmal Energy? Ist das der Zählerstand?

andig, 2020-10-30:
Genau. Brauchst Du nur beim Ladezähler, hier unnütz.

martinez81, 2020-11-01:
Ich habe jetzt noch mal probiert die PV Leitung über die Seriennummer des Wechselrichtern auszulesen.
Geht NICHT bei mir.

**- name: sma-pv
  type: sma
  serial: 30065xxxxx**

So werden keine Werte vom Wechselrichter (Tripower) ausgelesen.

Es geht nur so:

**- name: sma-pv
  type: modbus
  model: sunny-island
  uri: 192.168.0.165:502
  id: 126**

Ich denke neue Beispiele würden schon helfen.

premultiply, 2020-11-01:
Sunny Island ist ein Batteriewechselrichter...

Der Gerätetyp sma ist nur für den SMA Sunny Home Manger 2.0 oder ein SMA Energy Meter gedacht. Bei einem WR kann da nix kommen. Das geht bei SMA wie Modbus mit dem Typ sunspec.

martinez81, 2020-11-01:
> Sunny Island ist ein Batteriewechselrichter...
> 
> Das Gerätemodell sma ist nur für den SMA Sunny Home Manger 2.0 oder ein SMA Energy Meter gedacht. Bei einem PV-WR kann da nix kommen. Das geht bei SMA-WR via Modbus mit dem Modell sunspec.

geht aber auch mit model: sunny-island

premultiply, 2020-11-01:
sunspec wäre korrekt.

martinez81, 2020-11-01:
> sunspec wäre korrekt.

okay, ich habe das getestet. Geht 👍🏼

martinez81, 2020-11-01:
Ich habe jetzt diese config auch mal hier eingetragen:

https://github.com/andig/evcc/wiki/3.2.-SMA-Home-Manager-2.0---SMA-Energy-Meter

andig, 2020-11-03:
@martinez81 ich würde gerne vermeiden ein schnell veraltendes Wiki zu haben. Deine Konfiguration wird ja schon vollständig hier in https://github.com/andig/evcc-config abgebildet- der Wikieintrag ist m.E. unnötig?

andig, 2020-11-09:
Ich denke hier können wir zu machen ;)

closed 2020-11-09 by andig

electric-m, 2023-01-22:
@andig @martinez81 Ich möchte hier, auch wenn es schn über 2 Jahre her ist, trotzdem teilen, dass ich auch sehr mit der Integration des PV-Wechselrichters gekämpft habe.

Ich habe einen Sunny Tripower STP15000 TL-30.
Dieser hat kein Webinterface, das sich mit Passwort abfragen lässt.
Alle VErsuche mit dem SMA-Template schlugen fehl.

Am Ende hat dann die Config von @martinez81 funktioniert.
Die Doku habe ich natürlich dazu durchgesehen, bin aber gerade in Bezug auf Modbus oft auf tote Links gestoßen.

andig, 2023-01-22:
> Am Ende hat dann die Config von @martinez81 funktioniert.

@electric-m Du meinst diese Config?

```
- name: sma-pv
type: modbus
model: sunny-island
uri: 192.168.0.165:502
id: 126
```

@premultiply kommt mir *SEHR* komisch vor. Was für ein `model` soll das sein? 

> Die Doku habe ich natürlich dazu durchgesehen, bin aber gerade in Bezug auf Modbus oft auf tote Links gestoßen.

@electric-m könntest Du das bitte konkret machen? Welche Links auf welcher Seite? Ohne die Info werden wir sie nicht fixen können :(

premultiply, 2023-01-22:
Der Thread ist uralt und völlig veraltet.
model: sunny-island gibt es nicht. Das ist sunspec.

electric-m, 2023-01-22:
@andig @premultiply Dass der Thread veraltet ist, war mir klar. Ich wollte es nur an dieser Stelle ergänzen, da es mir nicht richtig erschien, eine neue Dsicussion/oder ein Issue zu öffnen.

Ich habe zunächst über evcc configure --advanced versucht, den Wechselrichter wie in der Doku beschrieben, so hinzuzufügen:

```
meters:
  - name: my_pv
    type: template
    template: sma-inverter
    usage: pv
    host: 192.0.2.2 # IP-Adresse oder Hostname
    password: # Passwort für Benutzergruppe Benutzer # Optional
```

Das war jedoch nicht erfolgreich, vermutlich wegen fehlendem Passwort:

```
cannot create meter 'pv2': cannot create meter 'template': cannot create meter 'sma': login failed
```

Ein Passwort habe ich jedoch nicht, da der Wechselrichter wie gesagt kein Webinterface hat.
Es kann ja nicht das Anlagenpasswort aus dem Sunny Explorer  gemeint sein?

Soweit OK, als Nächstes habe ich dann in der Doku die Möglichkeit gefunden, ein custom meter anzulegen.

Auf der Seite zu den Plugins sehe ich nur Demo-Codes für das isolierte Plugin, jedoch nicht, wie man sie konkret durch Einrücken in die Config einbezieht:
https://docs.evcc.io/docs/reference/plugins

nach dem mir die notation nicht ganz klar war, wie ich das Plugin in einem meter nutzen kann, habe ich es so versucht:

```
meters:  
- type: custom
  name: pvtestmeter
    source: sma
    uri: 192.168.4.51 # alternative to serial
    # serial: 123456 # alternative to uri
    value: ActivePowerPlus # ID of value to read
    # password: "0000" # optional (default: 0000)
    # interface: eth0 # optional
    # scale: 1 # optional scale factor for value
```

Funktioniert hat es dann letzendlich so:

```
meters:
- name: grid1
  type: sma
  serial: 1234567890 # insert your own serialnumber

- name: pv2
  type: modbus
# value: power
  model: sunspec
  uri: 192.168.2.xxx:502 # IP address, configured port (default is 502)
  id: 126 # Configured Modbus Device ID (SMA default ID 3 = 126)
```

Die toten Links finde ich gerade nicht mehr, gehe aber gleich nochmal auf die Suche.

Hier lief jedoch die Zeile mit "source" in einen Fehler.
Ich habe auch testweise die URI auskommentiert und die Serial verwendet.

premultiply, 2023-01-22:
Probier mal das Anlagenpasswort für die Speedwire-Verbindung wenn du kein anderes Passwort hast.
Ansonsten einfach einen normalen SunSpec-Inverter laut Doku hinzufügen.
SunSpec-Modbus-ID ist bei SMA per Default 126.

andig, 2023-01-22:
Sollen wir das für Sma vllt nochmal als Extratemplate mit der richtigen ID verpacken? 126 ist ja schon Herrschaftswissen?

electric-m, 2023-01-22:
> Sollen wir das für Sma vllt nochmal als Extratemplate mit der richtigen ID verpacken? 126 ist ja schon Herrschaftswissen?

Außer User-Sicht zu befürworten. Die ID 3 ist versierten Nutzern sicher bekannt, aber auf die 126 kommt man nicht so wirklich, da man sie sonst quasi nie benötigt.

In diesem Zusammenhang könnte man vielleicht in der Doku des SMA-Templates noch 1 bis 2 Sätze hinzufügen, die erklären, welche Art von passwort benötigt wird:

WebConnect Passwort? (Weboberfläche des WR)
Speedwire Passwort? (Sunny Exlporer etc.)

Hier müsste man mal noch sehen, ob da das Installateuerspasswort benötigt wird, oder eine Nutzerpasswort ausreicht.

Ich teste gleich mal für meinen Fall, ob es über Speedwire nun doch geht.
Sunspec funktioniert ja wie oben beschrieben. Ich würde, falls Speedwire funktioniert auch eher darauf gehen, um den Modbus zu entlasten, da schon einige Instanzen wie ioBroker etc dort abfragen.

electric-m, 2023-01-22:
@andig @premultiply Also, mit dem Nutzerpasswort, welches ich für den Login in SunnyExplorer nutzer, klappt der Zugriff auch mit dem Template für den SMA PV-Wechselrichter.