evcc-io/evcc #386: SIGSEGV in socestimator

schenlap, 2020-10-26:
```
[core  ] DEBUG 2020/10/26 09:16:57 pv power: 678W
[core  ] DEBUG 2020/10/26 09:16:57 grid power: -365W
[core  ] DEBUG 2020/10/26 09:16:57 site power: -365W
[lp-1  ] DEBUG 2020/10/26 09:16:57 charge power: 0W
[lp-1  ] DEBUG 2020/10/26 09:16:57 charger status: B
[lp-1  ] DEBUG 2020/10/26 09:16:57 charge start energy: 6kWh
[lp-1  ] INFO 2020/10/26 09:16:57 car connected
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x28 pc=0x381b08]

goroutine 47 [running]:
github.com/andig/evcc/core/wrapper.(*SocEstimator).Reset(0x0)
        /home/foobar/github.com/andig/evcc/core/wrapper/socestimator.go:44 +0x18
github.com/andig/evcc/core.(*LoadPoint).evVehicleConnectHandler(0x18d2160)
        /home/foobar/github.com/andig/evcc/core/loadpoint.go:291 +0x140
reflect.Value.call(0x976638, 0x1b0c888, 0x13, 0xb05759, 0x4, 0x15e699c, 0x0, 0x0, 0x0, 0xee5a98, ...)                                                                                       
        /usr/lib/go-1.14/src/reflect/value.go:460 +0x638
reflect.Value.Call(0x976638, 0x1b0c888, 0x13, 0x15e699c, 0x0, 0x0, 0x0, 0x0, 0x0)
        /usr/lib/go-1.14/src/reflect/value.go:321 +0x78
github.com/asaskevich/EventBus.(*EventBus).doPublish(0x189dd60, 0x1b66f60, 0xb09c92, 0x7, 0x0, 0x0, 0x0)                                                                                    
        /home/foobar/pkg/mod/github.com/asaskevich/!event!bus@v0.0.0-20200428142821-4fc0642a29f3/event_bus.go:160 +0x74                                                                   
github.com/asaskevich/EventBus.(*EventBus).Publish(0x189dd60, 0xb09c92, 0x7, 0x0, 0x0, 0x0)
        /home/foobar/pkg/mod/github.com/asaskevich/!event!bus@v0.0.0-20200428142821-4fc0642a29f3/event_bus.go:144 +0x218                                                                  
github.com/andig/evcc/core.(*LoadPoint).updateChargerStatus(0x18d2160, 0xb0748c, 0x5)
        /home/foobar/github.com/andig/evcc/core/loadpoint.go:398 +0x328
github.com/andig/evcc/core.(*LoadPoint).Update(0x18d2160, 0x0, 0xc076d000)
        /home/foobar/github.com/andig/evcc/core/loadpoint.go:642 +0x1d4
github.com/andig/evcc/core.(*Site).update(0x18aeb80, 0xed5128, 0x18d2160)
        /home/foobar/github.com/andig/evcc/core/site.go:324 +0x8c
github.com/andig/evcc/core.(*Site).Run(0x18aeb80, 0x540be400, 0x2)
        /home/foobar/github.com/andig/evcc/core/site.go:376 +0x18c
created by github.com/andig/evcc/cmd.run
        /home/foobar/github.com/andig/evcc/cmd/root.go:185 +0x578
```

kam beim Abschließen des Autos. Ich verwende eine ältere Version von evcc (bad8276aae685be47bfbd3a59d747c946f3ad637). Falls das Problem bekannt und behoben bitte den Report schließen. Die beschriebe Stelle schaut aber noch gleich aus bei mir zur aktuellen master.

In der verwendeten Config habe ich kein Auto angegeben da derzeit die Audi API wieder spinnt, vielleicht liegt es daran.

andig, 2020-10-26:
Danke- ist das aktuelle Version? Dachte das hätte ich behoben?

schenlap, 2020-10-26:
Nein, ist eine ältere Version. In den Commit-Messages habe ich aber nichts gefunden was auf einen Fix hindeutet. Wenn es schon behoben ist muss ich updaten..

closed 2020-10-26 by schenlap

andig, 2020-10-26:
Hatte ich auf kleinem Dienstweg behoben: https://github.com/andig/evcc/commit/9cebe21dd233da2324857e86a093821f10280708