evcc-io/evcc #387: Support Nissan EVs

andig, 2020-10-26:
Need login credentials for testing, please contact me at cpuidle@gmx.de

martinez81, 2020-11-18:
Hi!
Mit einen Kumpel habe ich es jetzt hinbekommen bei meinen Leaf den SOC auszulesen.
Es funktioniert mit [https://gitlab.com/tobiaswkjeldsen/dartnissanconnect](url)

Das ganze funktioniert im groben so (Linux oder OSX):

---------------------
% brew tap dart-lang/dart
% brew install dart git
% git clone https://gitlab.com/dutchdevil83/dartnissanconnect.git
% cd dartnissanconnect
% pub get
---------------------

Dann in der Datei example/nissanconnect_example.dart unter username: und passwort: die Zugangsdaten vom Auto eintragen - Abschließend das Programm starten:

---------------------
% dart example/nissanconnect_example.dart
---------------------

Wir haben dann einen neue Datei nissanconnect_soc.dart angelegt.
Hier der Inhalt:

---------------------

import 'package:dartnissanconnect/dartnissanconnect.dart';
import 'package:intl/intl.dart';

main() {
  NissanConnectSession session = new NissanConnectSession(debug: false);

  session
      .login(username: "user", password: "passwort")
      .then((vehicle) {

    			print("Modell:     ${vehicle.modelName}");
    			vehicle.requestBatteryStatus().then((battery) {
            print("Ladezustand der Batterie: ${battery.batteryPercentage}");
    			});
  		});
}

---------------------

Starten mit:

---------------------
% dart example/nissanconnect_soc.dart
---------------------

Die Ausgabe sieht dann so aus:
Modell:     Leaf
Ladezustand der Batterie: 45.0

Was kannst du daraus machen?
Kannst du das so irgendwie Einpflegen?

Gruß
Martin

PS: Datei hängt an.
[nissanconnect_soc.dart.zip](https://github.com/andig/evcc/files/5562736/nissanconnect_soc.dart.zip)

martinez81, 2020-11-18:
Hi!
Mit einen Kumpel habe ich es jetzt hinbekommen bei meinen Leaf den SOC auszulesen.
Es funktioniert mit [https://gitlab.com/tobiaswkjeldsen/dartnissanconnect](url)

Das ganze funktioniert im groben so (Linux oder OSX):

---------------------
% brew tap dart-lang/dart
% brew install dart git
% git clone https://gitlab.com/dutchdevil83/dartnissanconnect.git
% cd dartnissanconnect
% pub get
---------------------

Dann in der Datei example/nissanconnect_example.dart unter username: und passwort: die Zugangsdaten vom Auto eintragen - Abschließend das Programm starten:

---------------------
% dart example/nissanconnect_example.dart
---------------------

Wir haben dann einen neue Datei nissanconnect_soc.dart angelegt.
Hier der Inhalt:

---------------------

import 'package:dartnissanconnect/dartnissanconnect.dart';
import 'package:intl/intl.dart';

main() {
  NissanConnectSession session = new NissanConnectSession(debug: false);

  session
      .login(username: "user", password: "passwort")
      .then((vehicle) {

    			print("Modell:     ${vehicle.modelName}");
    			vehicle.requestBatteryStatus().then((battery) {
            print("Ladezustand der Batterie: ${battery.batteryPercentage}");
    			});
  		});
}

---------------------

Starten mit:

---------------------
% dart example/nissanconnect_soc.dart
---------------------

Die Ausgabe sieht dann so aus:
Modell:     Leaf
Ladezustand der Batterie: 45.0

Was kannst du daraus machen?
Kannst du das so irgendwie Einpflegen?

Gruß
Martin

PS: Datei hängt an.
[nissanconnect_soc.dart.zip](https://github.com/andig/evcc/files/5562739/nissanconnect_soc.dart.zip)

martinez81, 2020-11-18:
entschuldige bitte das Format. Keine Ahnung warum das so geworden ist.

andig, 2020-11-19:
> entschuldige bitte das Format. Keine Ahnung warum das so geworden ist.

Du hast Dein Posting nicht formatiert, kannst Du aber noch ändern: https://guides.github.com/features/mastering-markdown/

Nachdem Du den SoC jetzt auslesen kannst, kannst Du das Skript auch in EVCC integrieren. Das Beispiel dafür findest Du hier: https://github.com/andig/evcc-config#generisch-script-1

martinez81, 2020-11-19:
> Du hast Dein Posting nicht formatiert, kannst Du aber noch ändern: https://guides.github.com/features/mastering-markdown/
>
> Nachdem Du den SoC jetzt auslesen kannst, kannst Du das Skript auch in EVCC [...]

Das ich diesen dart Aufruf ein ein Script packen kann leuchtet mir noch nicht ganz ein. Ich bespreche das mit meinem Kumpel.

Nehmen wir mal an, ich kann das über ein Script hinbekommen.
Bleibt das Problem doch für andere User bestehen. Alle die einen Leaf nach Mitte 2019 haben, kommen nicht mehr über Carwings auf ihr Auto. Deshalb dachte ich du übernimmst das direkt in dein Programm?!

andig, 2020-11-19:
> Deshalb dachte ich du übernimmst das direkt in dein Programm?!

Ja- aber wie gesagt wenn ich Zeit habe :O

martinez81, 2020-11-19:
> 
> 
> > Deshalb dachte ich du übernimmst das direkt in dein Programm?!
> 
> Ja- aber wie gesagt wenn ich Zeit habe :O. Im Moment ist zuviel los, kommt aber. Wollte Dich nur nicht warten lassen ;)

Okay! Dann versuchen wir es mal.

martinez81, 2020-11-20:
Moin!
Es funktioniert. Wir mussten die Lib von Dartnissanconnect leicht ändern. Mit der normalen Lib wird (z.B.) **84.0%** ausgegeben.
EVCC möchte wohl nur Ganzzahlen als Input. Mit geänderter Lib wird jetzt nur **84** übertragen.
[image]

andig, 2020-11-20:
Perfekt!

martinez81, 2020-11-21:
Wenn jemand einen Nissan Leaf BJ nach Mitte 2019 hat und gern jetzt schon seinen Akkustand in EVCC haben möchte. Kann er das bis zur neuen Version von EVCC so machen:

Den Inhalt dieser ZIP Datei auf den Raspberry kopieren (wo EVCC läuft).
_Z.B: nach /home/pi/dartnissanconnect/_
[soc.zip](https://github.com/andig/evcc/files/5577386/soc.zip)

Die evcc.yaml Datei von EVCC editieren:

Bei _vehicles_ muss folgendes hin:

`- name: leaf
  type: default
  title: Leaf # display name for UI
  capacity: 40 # kWh
  charge:
    type: script # use script plugin
    cmd: /bin/sh -c "/home/pi/dartnissanconnect/soc --user=USER --pass=PASSWORD"
    timeout: 10s # kill script after 10 seconds
  cache: 5m # cache duration
`
USER und PASSWORD entsprechend ändern!!

evcc.yaml speichern und EVCC Neustarten.

ACHTUNG:
Bei den loadpoints müsst ihr natürlich noch den Leaf bekannt geben, also:

`  vehicles:
  - leaf
`

andig, 2020-11-21:
https://guides.github.com/features/mastering-markdown/ ;)

martinez81, 2020-11-21:
> https://guides.github.com/features/mastering-markdown/ ;)

Danke! Irgendwie stehe ich auf Kriegsfuß mir der Formatierung hier 🤨

andig, 2020-11-21:
@martinez81 könntest Du mal mit tracen, welche HTTP Requests die Dart Bibliothek macht? Ich komme noch nicht ganz durch den FLow und kann grad kein Dart installieren :(

martinez81, 2020-11-21:
Hi Andig,
ich werde versuche dir zu helfen!

Bzgl. Dart installieren, das geht ganz einfach:

$ wget https://storage.googleapis.com/dart-archive/channels/stable/release/2.6.0/sdk/dartsdk-linux-arm-release.zip
$ unzip dartsdk-linux-arm-release.zip

Dann in  $HOME/.profile den Pfad hinzufügen
$ nano .profile

hinzufügen in .profile:

if [ -d "$HOME/dart-sdk/bin" ] ; then
    PATH="$PATH:$HOME/dart-sdk/bin"
fi

% git clone https://gitlab.com/dutchdevil83/dartnissanconnect.git
% cd dartnissanconnect
% pub get

so sollte dartnissanconnect laufen....

andig, 2020-11-21:
Voll nett, aber geht unter OSX nicht und brew funktioniert nicht wg. zu neuer Version. Ein Trace durch einen Proxy würde helfen ;)

martinez81, 2020-11-21:
So habe ich es unter OSX zum laufen gebracht:
% brew tap dart-lang/dart
% brew install dart git
% git clone https://gitlab.com/dutchdevil83/dartnissanconnect.git
% cd dartnissanconnect
% pub get

Ich mache dir aber auch den Trace. Ich muss mich da nur einlesen. Habe ich noch nie gemacht...

andig, 2020-11-22:
@martinez81 Du kannst gerne schonmal probieren ob das Auto mit dem PR für Dich funktioniert.

martinez81, 2020-11-22:
> @martinez81 Du kannst gerne schonmal probieren ob das Auto mit dem PR für Dich funktioniert.

Meinst du mit PR PreRelease?
Wo finde ich das?

andig, 2020-11-22:
> Wo finde ich das?

Gar nicht. Die PRs können nur- bei Bedarf- lokal gebaut werden. Jetzt ist es im master Branch und im Release dabei.

PR #465 (2020-11-22): Add Nissan api
Fix #387. 

Missing feature is the token refresh, see https://github.com/Tobiaswk/dartnissanconnect/issues/1
closed 2020-11-22 by andig