evcc-io/evcc #391: Visualisierung vom minSoC Laden

Optic00, 2020-10-28:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Man sieht zwar das ein MinSoC geladen wird, in der GUI wird aber nicht angezeigt wo dies Ziel liegt. Außerdem wird bei der Restzeit nur angezeigt bis wann das Auto das Ladeziel erreicht nicht, nicht die MinSoC.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Vielleicht könnte man das als transparentes overlay machen? also das was noch fehlt, mit 50% transparenz und einem blinken? Die Restzeit wäre super, wenn die sich nur auf die MinSoC beziehen würde. Vielleicht auch irgendwie blinkend, gelb Batterie Symbol (wie bei Leistung)

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Den aktuellen Lademodus ggf. textlich noch Darstellen, oder die Restzeit für MinSoC separat anzeigen.

**Additional context**
Add any other context or screenshots about the feature request here.

Screenshot anbei vom IST Zustand und von einem kleinen Mockup (zugegeben nur mal so als Anregung)

[image]
[image]

premultiply, 2020-10-28:
Statt hier MinSoc zu "missbrauchen" könntest du einfach ein passendes Ladeziel (z. B. 40%) definieren und mit einem ebenfalls passenden Lademodus (z. B. Sofort) laden.

Optic00, 2020-10-29:
@premultiply wieso meinst du das hier was "missbraucht" wird? MinSoC ist ein Modus der automatisch startet wenn die Ladung unter einem bestimmten % Wert liegt und dann bis dahin auflädt, egal ob Sonne oder nicht (z.B. spät Abends). Das ganze besser zu visiualisieren wäre mein Anliegen. Wenn ich eine Ladeziel definiere und die Ladung per "Sofort" starte, dann wäre das MinSoC ja völlig überflüssig.

premultiply, 2020-10-29:
MinSoc ist halt kein Lademodus sondern nur ein Feature.
Die Restzeit bezieht sich immer auf das Ladeziel im gerade aktiven Lademodus.

andig, 2020-10-29:
Ich sehe hier zwei Anforderungen:
1. Restzeit bis MinSoc anzeigen
1. Zielwert des MinSoc anzeigen

Optionen:
1. Umbau der aktuellen Restzeit. Erscheint sinnvoll da je nach Modus danach ohnehin nicht weiter geladen würde. Visualisierung welche Zeit gemeint ist. Da jetzt mehrere Werte "visualisiert" werden müssten- Leistung ebenso- könnte ich mir das gemeinsam mit 2 vorstellen
2. und 1.: virtuell aktiven Lademodus ("Now") und Ladeziel ("MinSoC%") in den vorhanden Einstellungen z.B. gelb highlighten wie vorgeschlagen. Kein "neuer" Lademodus. Damit entfällt das zusätzliche Icon bei Leistung oder Restzeit.

Macht Sinn? Dann probiere ich mal was.

andig, 2020-10-29:
Mhhm, schön ist anders:

[image]

Optic00, 2020-10-29:
Danke für das Feedback,
ich finde das mit dem durchgängig gelb hinterlegten deutlich lesbarer als das untere. Man könnte es auch schraffiert so hinterlegen.
@andig Könnte man den Ladebalken so wie bei meinem Mockup noch mit dem Rest 50% transparent anzeigen lassen, was hälst du davon? Wäre für die anderen Modi eig. auch ganz chick? Aber natürlich nur wenn in dem Moment auch geladen wird..

premultiply, 2020-10-29:
Wenn überhaupt macht meines Erachtens nach da nur eine Visualisierung im Ladebalken Sinn.
Die die gelbe Hervorhebung hier ist erstens nicht hübsch und tzwetens im Fall des Ladeziels auch hier nur zufällig mit einem Ladeziel übereinstimmend.
Wenn dann würde ich vorne Modus und Ladeziel gelb färben.

andig, 2020-10-29:
> Wenn dann würde ich vorne Modus und Ladeziel gelb färben.

Du meinst Modus in gelb? Welche Aussage hätte das? In Realität wird ja gem. "Now", also mit voller Leistung, geladen?

Die Anpassung der Restladezeit mach aber in jedem Fall Sinn?

premultiply, 2020-10-29:
Aussage wäre nur da passiert irgendwas im Hintergrund abseits des gewählten Lademodus und des Ladeziels.
Welche Aussage hat die Restladezeit für MinSoc? Wenn man ein Ladeziel haben will und unbedingt die Restladezeit bis dahin kennen will wählt man eben einen entsprechendes Ladeziel und geht da nicht über MinSoc. Schneller geht es ja auch nicht wenn man die Restladezeit kennt.

Optic00, 2020-10-29:
@premultiply Wenn die Ladung nur aufgrund von MinSOC grad stattfindet, ist die Restladezeit die angezeigt wird halt einfach falsch imho. Es ist ja nur eine Frage der Visualisierung, wieso sollte man es nicht anzeigen, dafür ist die Anzeige doch da? Das es dadurch nicht schneller geht dürfte jedem klar sein.

premultiply, 2020-10-29:
Und wie sollte das aussehen wenn die Ladung anschließend zum Ladeziel fortgesetzt wird?
Das ist ja nur ein gewisser Bereich innerhalb eines normalen Ladevorgangs.
Ich sehe keinen praktischen Nutzen und nur größere Verwirrung durch komplexe Logik.

Optic00, 2020-10-29:
wenn die Ladung zum Ladeziel aufgrund von "Sofort" oder "PV" stattfindet, ist die MinSoC Ladung nicht aktiv, von daher sehe ich da kein Problem oder Verwirrung. MinSoC wurde bei mir z.B. gestern Abend ausgelöst als ich das Auto um 21 Uhr angeschlossen hatte. Hätte es in dem Moment schöner empfunden das nochmal eben zu sehen was eingestellt war in der config und statt der 100% Restladezeit die der aktuell stattfindenden Ladung.

andig, 2020-10-29:
> wenn die Ladung zum Ladeziel aufgrund von "Sofort" oder "PV" stattfindet, ist die MinSoC Ladung nicht aktiv, von daher sehe ich da kein Problem oder Verwirrung

Ich teile die Einschätzung. Restladedauer ist soger grob falsch wenn als Modus z.b. PV Min eingestellt ist und nachts die Sonne nicht scheint. Dennoch kann ich den Bedarf nachvollziehen, wissen zu wollen wann die Minimalreichweite für den Weg zur Arbeit erreicht sein wird.

andig, 2020-10-29:
Alternativer Ansatz: Modus/Ziel komplett ausblenden und stattdessen „Notladen“ mit allen Parametern anzeigen, dann kann ich aber für später nix mehr einstellen.

Optic00, 2020-10-29:
@andig ich fand es wie in deinem ersten Mockup gut. Alles auszublenden halte ich nicht für Zielführend, grad wenn man dann z.B. für den nächsten Tag noch etwas umstellen möchte etc.

andig, 2020-10-29:
Variante 3:

[image]

Variante 4:

[image]

Variante 5:

[image]

Wenn wir eine davon machen müssten wir konsequenterweise aber auch frage warum wir das bei TargetSoC nicht machen sollten?

Optic00, 2020-10-29:
Variante 5 finde ich gut

Caibuk, 2020-10-29:
Mir gefällt Variante 4 sehr gut. Dann könnte man das Grau von Var. 5 für TargetSoC nehmen (falls größer als minSoC).

andig, 2020-10-29:
> Mir gefällt Variante 4 sehr gut. Dann könnte man das Grau von Var. 5 für TargetSoC nehmen (falls größer als minSoC).

Persönlich mag ich die 4 auch lieber. Konsequenterweise müsste beim TargetSoC dann aber der gedimmte Balken in blau gehalten sein.

PR #393 (2020-11-03): Improve minSoC UX
Fix #391
closed 2020-11-03 by andig