evcc-io/evcc #404: min+PV funktioniert bei mir nicht

blawusel, 2020-11-04:
seit gestern habe ich nun meinen Smart EQ und wollt mit EVCC und einer Walbe Eco 2.0 PV Überschuss laden. 
Leider wird nur immer mit 3,68 also Full Power geladen , was ist falsch ?

Nov 04 13:06:24 Giga evcc[27090]: [core  ] DEBUG 2020/11/04 13:06:24 ----
Nov 04 13:06:24 Giga evcc[27090]: [core  ] DEBUG 2020/11/04 13:06:24 pv power: 2961W
Nov 04 13:06:24 Giga evcc[27090]: [core  ] DEBUG 2020/11/04 13:06:24 grid power: 1W
Nov 04 13:06:24 Giga evcc[27090]: [core  ] DEBUG 2020/11/04 13:06:24 battery power: -1025W
Nov 04 13:06:24 Giga evcc[27090]: [core  ] DEBUG 2020/11/04 13:06:24 site power: -1024W
Nov 04 13:06:24 Giga evcc[27090]: [lp-1  ] DEBUG 2020/11/04 13:06:24 charge power: 3680W
Nov 04 13:06:24 Giga evcc[27090]: [lp-1  ] DEBUG 2020/11/04 13:06:24 charger status: C
Nov 04 13:06:24 Giga evcc[27090]: [lp-1  ] DEBUG 2020/11/04 13:06:24 max charge current: 16A = 16A + 4A (-1024W @ 1p)
Nov 04 13:06:24 Giga evcc[27090]: [lp-1  ] DEBUG 2020/11/04 13:06:24 pv timer reset
Nov 04 13:06:24 Giga evcc[27090]: [lp-1  ] DEBUG 2020/11/04 13:06:24 target charge current: 16A
===
root@Giga:~# cat /etc/evcc.yaml
uri: 0.0.0.0:7075 # uri for ui
interval: 10s # control cycle interval

log: error
levels:
  core: debug
  lp-1: debug
  lp-2: debug

influx:
  url: http://localhost:8086
  database: evcc
  # user:
  # password:

# meters
meters:
    #- name: grid
    #  type: default
    #  power: # power reading
    #    type: mqtt # use mqtt
    #    topic: mbmd/sdm1-1/Power # mqtt topic
    #    timeout: 10s # don't use older values
#- name: pv
#  type: default
#  power:
#    type: modbus
#    uri: 192.168.1.74:502
#    id: 1
#    model: sunspec
#    value: Power

- name: sonnenPV
  type: default
  power:
    type: http
    uri: http://192.168.1.75:8080/api/v1/status
    jq: .Production_W
- name: sonnenBat
  type: default
  power:
    type: http
    uri: http://192.168.1.75:8080/api/v1/status
    jq: .Pac_total_W
    scale: -1
- name: sonnenGrid
  type: default
  power:
    type: http
    uri: http://192.168.1.75:8080/api/v1/status
    jq: .GridFeedIn_W
    scale: -1

chargers:
- name: wallbe
  type: wallbe # Wallbe charger
  uri: 192.168.1.8:502 # ModBus address
# legacy: true # enable for older Wallbes with Phoenix EV-CC-AC1-M3-CBC-RCM controller

vehicles:
- name: default
  type: default
  title: Default
  capacity: 18 # kWh
  charge:
    type: script # use script
    cmd: /bin/sh -c "echo 50" # actual command
    timeout: 3s # kill script after 3 seconds
  cache: 5m

site:
  title: Home # display name for UI
  meters:
    grid: sonnenGrid # grid meter
    pv: sonnenPV # pv meter
    battery: sonnenBat # battery meter

loadpoints:
- title: Garage # display name for UI
  charger: wallbe # charger
  #meters:
      #charge: charge # charge meter
      #vehicle: audi
  mode: now
  soc:
    alwaysUpdate: false # set true to update vehicle soc even when disconnected
    levels: # target soc levels for UI
    - 30
    - 50
    - 80
    - 100
  onDisconnect:
#    mode: pv # switch back to pv mode
    mode: off # switch back to pv mode
    targetSoC: 100 # charge fully
  phases: 1 # ev phases (default 3)
  sensitivity: 1 # current raise/lower step size (default 10A)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 10m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)

Stimmt da was in der Konfig nicht ? 
Kann mir bitte jemand helfen

goebelmeier, 2020-11-04:
Puh, könntest du bitte mal einzelne [Gravis](https://de.wikipedia.org/wiki/Gravis_(Typografie)) um dein Logfile und deine Config machen, dann greift das GitHub Markdown nicht und man kann es sicher besser lesen

goebelmeier, 2020-11-04:
Ausserdem würde ich erwarten, dass die PV power negativ und nicht positiv ist. Was für einen Wechselrichter hast du denn? Bei meinem E3DC PV Meter muss ich noch als config Option `scale: -1` mitgeben. Und dann sieht der funktionierende DEBUG Output so aus:
`evcc_1        | [core  ] DEBUG 2020/11/04 15:01:17 pv power: -2375W
evcc_1        | [core  ] DEBUG 2020/11/04 15:01:17 grid power: 102W
evcc_1        | [core  ] DEBUG 2020/11/04 15:01:17 battery power: -333W
evcc_1        | [core  ] DEBUG 2020/11/04 15:01:17 battery soc: 94%
evcc_1        | [core  ] DEBUG 2020/11/04 15:01:17 site power: -231W
evcc_1        | [lp-1  ] DEBUG 2020/11/04 15:01:17 charge power: 1573W
evcc_1        | [lp-1  ] DEBUG 2020/11/04 15:01:17 charger status: C
evcc_1        | [lp-1  ] DEBUG 2020/11/04 15:01:17 vehicle soc: 75%
evcc_1        | [lp-1  ] DEBUG 2020/11/04 15:01:17 detected phases: 1p [8.022693634033203 0 0]A
evcc_1        | [lp-1  ] DEBUG 2020/11/04 15:01:17 max charge current: 8A = 7A + 1A (-231W @ 1p)
evcc_1        | [lp-1  ] DEBUG 2020/11/04 15:01:17 pv timer reset
evcc_1        | [lp-1  ] DEBUG 2020/11/04 15:01:17 target charge current: 8A
evcc_1        | [lp-1  ] DEBUG 2020/11/04 15:01:17 set charge current: 8A`

blawusel, 2020-11-04:
sorry , habe mich auch gewunderte warum mit copy and paste so eine komische Anzeige kommt.

Bei Erzeugung  steht im Moment (GUI Anzeige) 0,3 W und im Log steht  .... pv power: 223W 
der Wert sollte also negativ sein ? 
Als Wechselrichter habe ich eine Sonnenbatterie die mit html abgefragt wird :

- name: sonnenPV
  type: default
  power:
    type: http
    uri: http://192.168.1.75:8080/api/v1/status
    jq: .Production_W
- name: sonnenBat
  type: default
  power:
    type: http
    uri: http://192.168.1.75:8080/api/v1/status
    jq: .Pac_total_W
    scale: -1
- name: sonnenGrid
  type: default
  power:
    type: http
    uri: http://192.168.1.75:8080/api/v1/status
    jq: .GridFeedIn_W
    scale: -1
....
site:
  title: Home # display name for UI
  meters:
    grid: sonnenGrid # grid meter
    pv: sonnenPV # pv meter
    battery: sonnenBat # battery meter

Also sollte ich bei sonnenPV noch den scale:-1 einarbeiten ?

andig, 2020-11-04:
Wer keine Gravis hat darf auch mittels Tab einrücken. Details siehe https://guides.github.com/features/mastering-markdown/.

andig, 2020-11-04:
> Also sollte ich bei sonnenPV noch den scale:-1 einarbeiten ?

Warum probierst Du es nicht einfach aus?

blawusel, 2020-11-04:
so jetzt steht im log : pv power: -14W
und im GUI auch 14 W 
jetzt wird es dunkel , das Auto ist vollgeladen , ich probiere es dann mal demnächst noch mal

blawusel, 2020-11-05:
hmm .. leider lädt er immer noch mit full power:

root@Giga:~# systemctl status evcc
● evcc.service - evcc
   Loaded: loaded (/etc/systemd/system/evcc.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2020-11-04 16:24:09 CET; 21h ago
 Main PID: 13869 (evcc)
    Tasks: 6 (limit: 4915)
   CGroup: /system.slice/evcc.service
           └─13869 /usr/local/bin/evcc --log error

Nov 05 13:30:10 Giga evcc[13869]: [core  ] DEBUG 2020/11/05 13:30:10 ----
Nov 05 13:30:10 Giga evcc[13869]: [core  ] DEBUG 2020/11/05 13:30:10 pv power: -2540W
Nov 05 13:30:10 Giga evcc[13869]: [core  ] DEBUG 2020/11/05 13:30:10 grid power: 56W
Nov 05 13:30:10 Giga evcc[13869]: [core  ] DEBUG 2020/11/05 13:30:10 battery power: -1442W
Nov 05 13:30:10 Giga evcc[13869]: [core  ] DEBUG 2020/11/05 13:30:10 site power: -1386W
Nov 05 13:30:10 Giga evcc[13869]: [lp-1  ] DEBUG 2020/11/05 13:30:10 charge power: 3680W
Nov 05 13:30:10 Giga evcc[13869]: [lp-1  ] DEBUG 2020/11/05 13:30:10 charger status: C
Nov 05 13:30:10 Giga evcc[13869]: [lp-1  ] DEBUG 2020/11/05 13:30:10 max charge current: 16A = 16A + 6A (-1386W @ 1p)
Nov 05 13:30:10 Giga evcc[13869]: [lp-1  ] DEBUG 2020/11/05 13:30:10 pv timer reset
Nov 05 13:30:10 Giga evcc[13869]: [lp-1  ] DEBUG 2020/11/05 13:30:10 target charge current: 16A

Version ist die V 0.29 .

goebelmeier, 2020-11-05:
`site power: -1386W` darf nur negativ sein wenn du einspeist. Ich denke dass ist aktuell bei dir aber Netzbezug. Bedeutet, dass du bei
```
- name: sonnenGrid
  type: default
  power:
    type: http
    uri: http://192.168.1.75:8080/api/v1/status
    jq: .GridFeedIn_W
    scale: -1
```
den scale Parameter entfernen musst. Und auf 0.30 kannst du auch updaten :-)

blawusel, 2020-11-05:
leider auch nicht
systemctl status evcc
● evcc.service - evcc
Loaded: loaded (/etc/systemd/system/evcc.service; enabled; vendor preset: enabled)
Active: active (running) since Thu 2020-11-05 14:02:07 CET; 3min 0s ago
Main PID: 22162 (evcc)
Tasks: 5 (limit: 4915)
CGroup: /system.slice/evcc.service
└─22162 /usr/local/bin/evcc --log error

Nov 05 14:05:07 Giga evcc[22162]: [core ] DEBUG 2020/11/05 14:05:07 ----
Nov 05 14:05:07 Giga evcc[22162]: [core ] DEBUG 2020/11/05 14:05:07 pv power: -2550W
Nov 05 14:05:07 Giga evcc[22162]: [core ] DEBUG 2020/11/05 14:05:07 grid power: 18W
Nov 05 14:05:07 Giga evcc[22162]: [core ] DEBUG 2020/11/05 14:05:07 battery power: -1254W
Nov 05 14:05:07 Giga evcc[22162]: [core ] DEBUG 2020/11/05 14:05:07 site power: -1236W
Nov 05 14:05:07 Giga evcc[22162]: [lp-1 ] DEBUG 2020/11/05 14:05:07 charge power: 3680W
Nov 05 14:05:07 Giga evcc[22162]: [lp-1 ] DEBUG 2020/11/05 14:05:07 charger status: C
Nov 05 14:05:07 Giga evcc[22162]: [lp-1 ] DEBUG 2020/11/05 14:05:07 max charge current: 16A = 16A + 5A (-1236W @ 1p)
Nov 05 14:05:07 Giga evcc[22162]: [lp-1 ] DEBUG 2020/11/05 14:05:07 pv timer reset
Nov 05 14:05:07 Giga evcc[22162]: [lp-1 ] DEBUG 2020/11/05 14:05:07 target charge current: 16A
 

??
dann mach ich mal noch schnell ein update ...

blawusel, 2020-11-05:
schade jetzt ist er wieder voll , kann also nicht weiter testen ...

blawusel, 2020-11-05:
vielleicht liegt es ja an den Werten die die Batterie von Sonnen liefert ?
Ich probiere morgen (wenn die Sonne scheint) einfach mal nur den Wechselrichter (SolarEdge) aus .

goebelmeier, 2020-11-05:
Also irgendwas stimmt an deinen Vorzeichen nicht. 

- `pv power` muss negativ wenn die PV-Anlage produziert
- `grid power` muss negativ sein, wenn du Überschuss in's Netz einspeist und positiv wenn du aus dem Netz beziehst
- `battery power` muss negativ sein, wenn der Hausakku geladen wird und positiv wenn aus dem Hausakku ausgespeist wird

Bei dir sieht es also aktuell so aus, dass du 2550W produzierst, 18W aus dem Netz beziehst und mit 1254W in den Akku lädst. Meine Vermutung ist also, dass auch der Akku scale falsch rum ist.

blawusel, 2020-11-05:
systemctl status evcc
● evcc.service - evcc
   Loaded: loaded (/etc/systemd/system/evcc.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-11-05 15:12:26 CET; 41s ago
 Main PID: 28260 (evcc)
    Tasks: 5 (limit: 4915)
   CGroup: /system.slice/evcc.service
           └─28260 /usr/local/bin/evcc --log error

Nov 05 15:12:56 Giga evcc[28260]: [core  ] DEBUG 2020/11/05 15:12:56 site power: -46W
Nov 05 15:12:56 Giga evcc[28260]: [lp-1  ] DEBUG 2020/11/05 15:12:56 charge power: 0W
Nov 05 15:12:56 Giga evcc[28260]: [lp-1  ] DEBUG 2020/11/05 15:12:56 charger status: A
Nov 05 15:13:06 Giga evcc[28260]: [core  ] DEBUG 2020/11/05 15:13:06 ----
Nov 05 15:13:06 Giga evcc[28260]: [core  ] DEBUG 2020/11/05 15:13:06 pv power: -352W
Nov 05 15:13:06 Giga evcc[28260]: [core  ] DEBUG 2020/11/05 15:13:06 grid power: 36W
Nov 05 15:13:06 Giga evcc[28260]: [core  ] DEBUG 2020/11/05 15:13:06 battery power: -91W
Nov 05 15:13:06 Giga evcc[28260]: [core  ] DEBUG 2020/11/05 15:13:06 site power: -55W
Nov 05 15:13:06 Giga evcc[28260]: [lp-1  ] DEBUG 2020/11/05 15:13:06 charge power: 0W
Nov 05 15:13:06 Giga evcc[28260]: [lp-1  ] DEBUG 2020/11/05 15:13:06 charger status: A
[image]

also müsste es so richtig sein ?

goebelmeier, 2020-11-05:
Nein, deine Batterie wird entladen und EVCC sieht den Wert negativ (`battery power: -91W`), da ist der Fehler. Wenn da ein scale in deiner Config in der Battery-Section steht, muss es weg. Wenn keins da ist, muss da eins hin :-)

blawusel, 2020-11-05:
ok geändert , jetzt sieht es so aus 
Nov 05 16:33:03 Giga evcc[2396]: [core  ] DEBUG 2020/11/05 16:33:03 site power: 359W
Nov 05 16:33:03 Giga evcc[2396]: [lp-1  ] DEBUG 2020/11/05 16:33:03 charge power: 0W
Nov 05 16:33:03 Giga evcc[2396]: [lp-1  ] DEBUG 2020/11/05 16:33:03 charger status: A
Nov 05 16:33:13 Giga evcc[2396]: [core  ] DEBUG 2020/11/05 16:33:13 ----
Nov 05 16:33:13 Giga evcc[2396]: [core  ] DEBUG 2020/11/05 16:33:13 pv power: -0W
Nov 05 16:33:13 Giga evcc[2396]: [core  ] DEBUG 2020/11/05 16:33:13 grid power: 10W
Nov 05 16:33:13 Giga evcc[2396]: [core  ] DEBUG 2020/11/05 16:33:13 battery power: 338W
Nov 05 16:33:13 Giga evcc[2396]: [core  ] DEBUG 2020/11/05 16:33:13 site power: 348W
Nov 05 16:33:13 Giga evcc[2396]: [lp-1  ] DEBUG 2020/11/05 16:33:13 charge power: 0W
Nov 05 16:33:13 Giga evcc[2396]: [lp-1  ] DEBUG 2020/11/05 16:33:13 charger status: A
[image]

das mit dem Scale ist nicht so einfach :-)

So nun muss ich mal wieder auf die Sonne warten und ein bisschen Smart fahren, um die Batterie ein wenig leer zu bekommen,
Ist der Smart EQ eigentlich schon als Vehicles integriert, also Abfrage vom Ladezustand usw.

blawusel, 2020-11-05:
und noch mal Danke für die schnelle Reaktion !!!!!

blawusel, 2020-11-07:
ich verstehe es nicht , ich habe die Batterie jetzt mal weggelassen , aber trotzdem regelt da nix .. 
jetzt wird konstant mit 6A geladen egal ob ich auf min+PV oder nur PV schalte,

Nov 07 14:35:31 Giga evcc[6452]: [lp-1  ] DEBUG 2020/11/07 14:35:31 max charge current: 5A = 6A + -1A (38W @ 1p)
Nov 07 14:35:31 Giga evcc[6452]: [lp-1  ] DEBUG 2020/11/07 14:35:31 target charge current: 6A
Nov 07 14:35:41 Giga evcc[6452]: [core  ] DEBUG 2020/11/07 14:35:41 ----
Nov 07 14:35:41 Giga evcc[6452]: [core  ] DEBUG 2020/11/07 14:35:41 pv power: -1776W
Nov 07 14:35:41 Giga evcc[6452]: [core  ] DEBUG 2020/11/07 14:35:41 grid power: 2W
Nov 07 14:35:41 Giga evcc[6452]: [core  ] DEBUG 2020/11/07 14:35:41 site power: 2W
Nov 07 14:35:41 Giga evcc[6452]: [lp-1  ] DEBUG 2020/11/07 14:35:41 charge power: 1380W
Nov 07 14:35:41 Giga evcc[6452]: [lp-1  ] DEBUG 2020/11/07 14:35:41 charger status: C
Nov 07 14:35:41 Giga evcc[6452]: [lp-1  ] DEBUG 2020/11/07 14:35:41 max charge current: 5A = 6A + -1A (2W @ 1p)
Nov 07 14:35:41 Giga evcc[6452]: [lp-1  ] DEBUG 2020/11/07 14:35:41 target charge current: 6A

Hier müsste doch mit 7A geladen werden ?
Ich möchte einfach nur sofort laden (das funktioniert) oder mit mindestens 6A und alles was dann noch aus der PV Anlage kommt ???

blawusel, 2020-11-07:
[image]

bei den Spitzen habe ich auf sofort laden umgeschaltet..

andig, 2020-11-07:
@blawusel ich glaube Du schaust fälschlicherweise auf

    pv power: -1776W

dann hast Du aber Deinen Hausverbrauch vergessen. Denn

    grid power: 38W

also etwas Bezug. Deshalb 

    max charge current: 5A = 6A + -1A (38W @ 1p)

Dank Min+PV bleibt der Charger mit 6A aber aktiviert. 

Alles wie erwartet. Ich denke wir können das Ticket mal zumachen?

blawusel, 2020-11-07:
sorry ich will es einfach nur verstehen ...
Wie berechnet sich denn max charge current ? 
in den Beispiel 1776W / 230 V = 7,72 A 
jetzt werden zusätzlich 38W (Grid Power) verbraucht,  also wird noch 1A abgezogen  währen dann 6,72 A
bei max charge current steht aber 5A wie kommen die zustande ?

Das Problem  bei meiner Konstellation  ist , das die Sonnenbatterie fast immer zwischen 20W bis 80W aus dem Netzt zieht sobald irgendwelche Belastung auftritt .
[image]

andig, 2020-11-07:
> Wie berechnet sich denn max charge current ?

Aktueller Steom +/- äquivalent am Netzzähler. Nichmal: die PV Erzeugung hat damit nur indirekt zu tun!

blawusel, 2020-11-08:
also ist es egal wie viel erzeugt wird  (1776W), weil 38W bezogen  werden,  wird -1A von min (6A) abgezogen , also 5 A und weil das zu wenig ist wird mit 6A geladen ?

Ich habe gerade noch mal probiert , sobald ich die Ladung des Autos starte springt der Grid Power auf Bezug und damit wird dann nur noch mit 6A geladen.
Außerdem wird ein sehr hoher Verbrauch angezeigt , ich lade das Auto mit 1,4 KW und habe sonst im Moment nur ca. 300 W Verbrauch. Scheint also auch ein Fehler in der SonnenBatterie zu sein.

[image]

Schade damit ist das mit einer Batterie und dem Grid einer Sonnenbatterie (Sonnen GmbH) nicht machbar.
Oder hat noch jemand ne Idee ?

premultiply, 2020-11-08:
Ich verstehe dein Problem nicht.
EVCC regelt auf den Überschuss/Bezug am Hausübergabepunkt. Weder die PV-Erzeugung noch der eigentliche Ladestrom hat eine direkte Relevanz.
Ich gehe davon aus dein Fahrzeug lädt mit 3*6A*230V = 4140VA. Da passen die ca. 3700W genau zum Ladegerät des Fahrzeugs.

blawusel, 2020-11-08:
nein ich habe nur 1 Phase,  L2 und L3 sind nicht angeklemmt und mein  Smart kann auch nur eine Phase

andig, 2020-11-09:
> Ich verstehe dein Problem nicht.

Geht mir genauso. Ich mache hier mangels erkennbarem Problem mal zu. Wenns ein konkretes Problem gibt bitte mit Logfile und Fehlerbeschreibung neues Ticket aufmachen.

closed 2020-11-09 by andig