evcc-io/evcc #405: Unterstützung von Umgebungsvariablen zur Konfiguration

StefanSchoof, 2020-11-06:
**Is your feature request related to a problem? Please describe.**
Ich möchte gerne Passworter und andere Konfiguration soweit wie möglich trennen. (Berufskrankheit ;) ) Das ermöglicht mir meine Konfiguration einfach bei github hochzuladen (https://github.com/StefanSchoof/evcc-config) und somit kann ich einfach auf einen alten Config stand und habe schon ein Backup. (Nach Linus Torvalds: "Real men don’t use backups, they post their stuff on a public ftp server and let the rest of the world make copies." )

**Describe the solution you'd like**
Aktivieren Umgebungsvariablen in viper https://github.com/spf13/viper#working-with-environment-variables, so das man das Passwörter per Umgebungsvariable an evcc übergeben kann.

**Describe alternatives you've considered**
Platzhalter für Passwörter in der config, was zurück drehen der config und aktualisieren von gut schwieriger macht.

**Additional context**
-

andig, 2020-11-06:
@StefanSchoof AutomaticEnv ist bereits gesetzt. Bis auf den fehlenden Prefix sollte das also gehen. Mir ist allerdings nicht klar, wie das innerhalb von Listen wie z.B. den vehicles funktionieren soll?

StefanSchoof, 2020-11-06:
Aus https://github.com/spf13/viper#accessing-nested-keys. (`GetInt("host.ports.1")`) hatte ich gehofft, dass man `VEHICLES_0_PASSWORD` das Passwort setzen kann.  Aber leider klappt, dass nicht.

andig, 2020-11-06:
SO oder viper Ticket aufmachen?

StefanSchoof, 2020-11-06:
Ich probiere und Google noch ein bisschen und mache das dann.

StefanSchoof, 2020-11-09:
Das ist ein Problem in viper: https://github.com/spf13/viper/issues/935. Insgesamt scheint es mit dem Unmarshal und env vars nicht so gut zu gehen: https://github.com/spf13/viper/issues/761

Denke daher, das ist nichts was man einfach hinbekommt :(

closed 2020-11-09 by StefanSchoof

andig, 2020-11-09:
Schade :(