evcc-io/evcc #412: vendor-specific vehicles interfaces for Mercedes

Brovning, 2020-11-09:
https://developer.mercedes-benz.com/products/electric_vehicle_status/specification
"Electric Vehicle Status API 2.0 OAS3
Tryout URL: https://api.mercedes-benz.com/vehicledata_tryout/v2
Production URL: https://api.mercedes-benz.com/vehicledata/v2

This API allows to determine the electrical status and remaining distance of connected Mercedes-Benz vehicles remotely. Data is categorized into resources and containers for the polling interface. They are defined as follows:

Resource: A resource is a single information about a vehicle. It's a measurement that consists of a name (also referred to as resource ID), a value, and an associated timestamp. See below for a list and description of all resources that this API provides.
Container: A container is a set of resources that are defined to group data for a certain use case.
The interface is a ISO 20078-compliant REST endpoint to query the latest data for a vehicle. If the vehicle did not send an update for a resource within 12 hours, the response will be empty.

Resources
This is an alphabetical list of all resources that the API provides.

Name	Description	Unit/Range
soc	Displayed state of charge for the HV battery	0..100 %
rangeelectric	Electric range	0..2046 km"

Link to API-Specification: https://developer.mercedes-benz.com/doc/products/electric_vehicle_status/swagger/swagger_external.yaml

Authentication Doc: https://developer.mercedes-benz.com/content-page/oauth-documentation

andig, 2020-11-09:
@Brovning vmtl. kein Problem wenn ich einen Testzugang bekommen könnte. Magst Du Dich mal unter cpuidle@gmx.de melden?

Brovning, 2020-11-09:
Mail gesendet.

Weitere Info:

"1. Get the Available Resources That Can Be Read Out
You can use the resources resource of the Electric Vehicle Status API to get the resources that are available for readout. The response contains the available resources for the corresponding car. You can specify the car using the vehicleId path parameter in the request.
`curl -X GET "https://api.mercedes-benz.com/vehicledata/v2/vehicles/<insert_your_vehicle_id_here>/resources" -H "accept: application/json;charset=utf-8" -H "authorization: Bearer <insert_the_access_token_here>"`

Response:

> [
>     {
>         "name": "soc",
>         "version": "1.0",
>         "href": "/vehicles/EXVETESTVIN000001/resources/soc"
>     },
>     {
>         "name": "rangeelectric",
>         "version": "1.0",
>         "href": "/vehicles/EXVETESTVIN000001/resources/rangeelectric"
>     }
> ]

2. Get the value of a specific resource
You can use the name of the specific resource to get the value of a concrete resource. The response contains the value of the specific resource and the corresponding readout timestamp.
`curl -X GET "https://api.mercedes-benz.com/vehicledata/v2/vehicles/<insert_your_vehicle_id_here>/resources/soc" -H "accept: application/json;charset=utf-8" -H "authorization: Bearer <insert_the_access_token_here>"`

> {
>     "soc": {
>         "value": "50",
>         "timestamp": 1541750489000
>     }
> }

3. Get all values of the Electric Vehicle Status API
You can use the containers resource of the Electric Vehicle Status API to get the values of all resources that are available for readout. The response contains the available resource values and the corresponding readout timestamp for the corresponding car.

`curl -X GET "https://api.mercedes-benz.com/vehicledata/v2/vehicles/<insert_your_vehicle_id_here>/containers/electricvehicle" -H "accept: application/json;charset=utf-8" -H "authorization: Bearer <insert_the_access_token_here>"`

> [
>     {
>         "soc": {
>             "value": "35",
>             "timestamp": 1541749824000
>         }
>     },
>     {
>         "rangeelectric": {
>             "value": "1021",
>             "timestamp": 1541749824000
>         }
>     }
> ]

andig, 2020-11-09:
Zwei Probleme:

- Oauth Flow mit Redirect. Das ist schwierig da die Redirect URI immer nur lokal gültig ist, ich sie also nicht beim Provider registrieren kann. Konzeptionelle Lücke bzw. m.E. falscher OAuth Flow.
- Client ID und Secret. Kann ich natürlich nicht ins Repo einchecken da sie sonst jeder hätte. Das liesse sich zur Not umgehen, ist aber mit viel Aufwand verbunden.

andig, 2020-11-09:
Der Authorization Flow landet auf einem PingID Endpoint der sich v.a. durch Nachladung von JS auszeichnet. Für EVCC ist das leider nicht nutzbar. Mit schwerem Herzen wontfix :(

closed 2020-11-17 by andig