evcc-io/evcc #415: Change url path

userjp123, 2020-11-10:
Hallo,

ich weiß nicht, ob ich der einzige bin, der evcc auf einem Server laufen lässt, auf dem noch andere Apps laufen.

Ich habe zB auch nextcloud dort laufen und bekomme damit Probleme, wenn ich evcc dort as-it-is laufen lasse.
Ursache:
evcc nutzt im Pfad: /api , /ws  und evt. noch /dist ?

Ich benötige aber für die Umleitung in der VirtualHost von Apache eine eindeutige Aussage darüber, dass die URL zu evcc gehört.
Nextcloud arbeitet auch mit /api und /ws.
Ich ändere deshalb im Code immer folgendes ab:

/api -> /wbapi
/ws  -> /wbws
--------------------
evcc/server/http.go

	api := router.PathPrefix("/wbapi").Subrouter()
	
	router.HandleFunc("/wbws", SocketHandler(hub))
	
----------------------
evcc/assets/js/app.js

	axios.defaults.baseURL = loc.protocol + "//" + loc.hostname + (loc.port ? ":" + loc.port : "") + "wbapi";

	const uri = protocol + "//" + loc.hostname + (loc.port ? ":" + loc.port : "") + "wbws";

**FRAGE**:
Wäre es möglich, dass das von Haus aus geändert wird ?  (Präfix wb , aber kann ja auch was anderes sein)
Ziel ist, dass alle URLs zu/von evcc eindeutig sind.

Vielen Dank & Viele Grüße

andig, 2020-11-10:
Wenn Du Apache nutzt kannst Du einfach mod_proxy dazwischen schalten. Es gibt keinen Grund, deshalb evcc auf Port 80 laufen zu lassen. Das würde ohnehin nicht funktionieren wenn z.B. auch SSL notwendig wäre.

M.E. sollten wir dafür nichts ändern!

userjp123, 2020-11-10:
Läuft mit mod_proxy und SSL und es funktioniert ja.

userjp123, 2020-11-10:
Meine URL zu evcc: https://_domain_/wb1
Das wird dann in der Virtualhost  auf http://127.0.0.1:7070 gemappt

andig, 2020-11-10:
Verstehe ich nicht. Wo ist dann Dein *konkretes* Problem?

userjp123, 2020-11-10:
Ich müsste /api und /ws zu evcc mappen und dass darf ich nicht, da Nextcloud das auch benutzt. 
Ich muss an der URL erkennen können, dass die an evcc geleitet werden muss.

Meine Virtualhost:

                RewriteCond %{HTTP:Upgrade} =websocket [NC]
                RewriteRule /wb1/(.*)           ws://127.0.0.1:7070/$1 [P]
                RewriteRule /wbws           ws://127.0.0.1:7070/wbws [P]

                RewriteCond %{HTTP:Upgrade} !=websocket [NC]
                RewriteRule /wb1/(.*)           http://127.0.0.1:7070/$1 [P]
                RewriteRule /wbapi/(.*)           http://127.0.0.1:7070/wbapi/$1 [P]

userjp123, 2020-11-10:
...zumindest habe ich nur diese Möglichkeit gefunden. 
Evtl. findet sich ja ein Spezi mit einer anderen Lösung.

andig, 2020-11-10:
Du möchtest also, dass EVCC statt absoluter URLs relative nutzt? Für die Websockets sollte das schon der Fall sein, für's UI ist das hier drin: https://github.com/andig/evcc/pull/414/files. Sobald die relativ sind sollten Deine Probleme gelöst sein?

userjp123, 2020-11-10:
Müsste ich dann testen, wenn das raus ist. 
Wichtig ist, dass bei jeder URL eine eindeutige Kennung vorhanden ist.
Diese Seiten hier in github haben immer **andig** im Pfad: https://github.com/andig/......

(Man kann sich zB in Firefox die URLs anschauen: F12)
Auf evcc übertragen ist der **Jetzt**-Zustand:
https://github.com/api
https://github.com/ws
etc.

andig, 2020-11-10:
> Wichtig ist, dass bei jeder URL eine eindeutige Kennung vorhanden ist.

Nein, das ist völlig irrelevant. Du mappst _Deine_ unqiue URL auf den EVCC Pfad auf anderem Port in der Apache Conf. Dafür muss auf EVCC-Seite nichts unique sein.

Ich mache hier mal zu, bitte angegebenen PR beobachten.

closed 2020-11-10 by andig

userjp123, 2020-11-11:
Jetzt gerade gesehen: 
Fehler wurde mit V0.31 behoben: **loc.pathname** hatte bisher gefehlt.

axios.defaults.baseURL = loc.protocol + "//" + loc.hostname + (loc.port ? ":" + loc.port : "") + **loc.pathname** + "api";

userjp123, 2020-11-11:
Danke :-)