evcc-io/evcc #420: Ladestrom wird nicht geregelt

blawusel, 2020-11-10:
Hallo,
erst einmal vielen Dank für dieses Klasse Projekt und den Support den man hier bekommt.
Mit eurer Hilfe habe ich EVCC zum Laufen bekommen und hoffe ihr könnt mir auch bei dem Problem weiterhelfen. 
Die Wallbe Eco 2.0s ist einphasig angeschlossen und ich möchte damit meinen Smart EQ auch einphasig laden , der Smart hat keinen Schnellader und kann somit auch nur einphasig mit max 16A mit der Eco2.0s geladen werden.
Mein Problem ist , die Wallbe oder EVCC regelt den Ladestrom nicht , es wird immer mit 16A geladen , obwohl nur mit 6A geladen werden sollte.
[image]
Ich habe auch mal ein log mitlaufen lassen.
[evcc.log](https://github.com/andig/evcc/files/5518044/evcc.log)
Ab der  Zeit von ca  15:04:30 Uhr wird geladen.
Auf der Wallbe selbst ist nichts zu erkennen.
[image]

Im Log sieht man folgendes 

[http  ] TRACE 2020/11/10 15:05:01 GET http://192.168.1.75:8080/api/v1/status
200 OK
{"Apparent_output":232,"BackupBuffer":"0","BatteryCharging":false,"BatteryDischarging":false,"Consumption_Avg":2114,**"Consumption_W":3905,**"Fac":50.01300048828125,"FlowConsumptionBattery":false,"FlowConsumptionGrid":true,"FlowConsumptionProduction":false,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":false,"GridFeedIn_W":-3910,"IsSystemInstalled":1,"OperatingMode":"2","Pac_total_W":-5,"Production_W":0,"RSOC":6,"RemainingCapacity_W":357,"Sac1":77,"Sac2":78,"Sac3":77,"SystemStatus":"OnGrid","Timestamp":"2020-11-10 15:05:01","USOC":0,"Uac":233,"Ubat":48}
[core  ] DEBUG 2020/11/10 15:05:01 grid power: 3910W
[core  ] DEBUG 2020/11/10 15:05:01 site power: 3910W
[lp-1  ] DEBUG 2020/11/10 15:05:01 charge power: 1380W
[cache ] TRACE 2020/11/10 15:05:01 gridPower: 3910
[cache ] TRACE 2020/11/10 15:05:01 mode: minpv
[cache ] TRACE 2020/11/10 15:05:01 chargePower: 1380
[cache ] TRACE 2020/11/10 15:05:01 chargeCurrent: 6
[influx] TRACE 2020/11/10 15:05:01 write gridPower=3910 (map[])
[influx] TRACE 2020/11/10 15:05:01 write chargePower=1380 (map[loadpoint:Garage])
[wallbe] TRACE 2020/11/10 15:05:01 read charge time (102): 00 1C 00 00
[cache ] TRACE 2020/11/10 15:05:01 chargedEnergy: 7.64230571795
[cache ] TRACE 2020/11/10 15:05:01 chargeDuration: 28s
[influx] TRACE 2020/11/10 15:05:01 write chargedEnergy=7.64230571795 (map[loadpoint:Garage])
[wallbe] TRACE 2020/11/10 15:05:01 read status (100): 00 43
[lp-1  ] DEBUG 2020/11/10 15:05:01 charger status: C
[cache ] TRACE 2020/11/10 15:05:01 connected: true
[cache ] TRACE 2020/11/10 15:05:01 charging: true
[wallbe] TRACE 2020/11/10 15:05:01 read charge enable (400): 01
[lp-1  ] DEBUG 2020/11/10 15:05:01 max charge current: 0A = 6A + -17A (3910W @ 1p)
[lp-1  ] DEBUG 2020/11/10 15:05:01 **target charge current: 6A**
[core  ] DEBUG 2020/11/10 15:05:11 ----
[http  ] TRACE 2020/11/10 15:05:11 GET http://192.168.1.75:8080/api/v1/status

Die Version von EVCC ist 0.30.
Es währe sehr schön wenn ihr weiterhelfen könntet .

andig, 2020-11-10:
Du hast 3910W Netzbezug. Bei pv+min wird daher auf minimalen Strom geschaltet. Funktioniert wie erwartet:

> [lp-1 ] DEBUG 2020/11/10 15:05:01 target charge current: 6A

Warum sagst Dum dass mit 16A geladen wird? Auf Grund des Bezugs?

Falls ja: ist Deine Wallbe älteren Datums? Dann schau mal bitte in die Doku unter https://github.com/andig/evcc#charger und setze `legacy: true`.

blawusel, 2020-11-10:
Ja,  auf Grund des hohen Bezuges , mein "Grundrauschen" liegt so um die 200 bis 300 W . siehe die Zeit so gegen 15:03:00. 
Auch auf meinem verbauter Stromzähler (extra Zähler für die WAllbox) wird ein Strom von 3.6 kW angezeigt. 
Die Wallbe ist im August original von Wallbe gekauft , vielleicht ist es ja auch dieser Parameter.
Kann da was kaputt gehen wenn ich legacy: true setzte ? 
mbmd habe ich nicht (Ubuntu 18.04.5 LTS)

andig, 2020-11-10:
Versuchs einfach, kaputt geht nix da der Strom begrenzt ist.

Caibuk, 2020-11-10:
> root@Giga:/tmp# mbmd read -a 192.168.1.8:502 -d 255 -t holding -e int 300 1
> 16

Register 300 gibt Dir den aktuell vom Controller/von der Wallbe freigegebenen Strom (16 A -> genauso, wie es gerade auch ist).
Probier mal Register 528 aus. Da bekommst Du die "Remote Charge Current Limitation". Die müsste in Deinem Fall 60 sein.

Das hatte ich alles mit meiner (alten) Wallbe eco auch durch.
legacy: true sollte Dein Freund sein.

Caibuk, 2020-11-10:
Und mach bitte den Haken bei "Auto Refresh" raus. Solange das Browserfenster damit geöffnet ist, wird immer wieder der bei "Set Charging Current" angegebene Wert gesetzt (auch wenn evcc im Hintergrund etwas anderes vorgibt). Das hat mir letztes Wochenende ein paar graue Haare beschert.

blawusel, 2020-11-10:
root@Giga:~# mbmd read -a 192.168.1.8:502 -d 255 -t holding -e int 528 1
6
leider steht da 6 
Ich habe jetzt mal legacy: true gesetzt.
Leider ist der Smart aber gerade unterwegs , werde mit loggen sobald er wieder hier ist.

blawusel, 2020-11-10:
Klasse , es geht :-)
[image]

{"Apparent_output":232,"BackupBuffer":"0","BatteryCharging":false,"BatteryDischarging":false,"Consumption_Avg":1738,**"Consumption_W":1769,**"Fac":50.030002593994141,"FlowConsumptionBattery":false,"FlowConsumptionGrid":true,"FlowConsumptionProduction":false,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":false,"GridFeedIn_W":-1774,"IsSystemInstalled":1,"OperatingMode":"2","Pac_total_W":-5,"Production_W":0,"RSOC":6,"RemainingCapacity_W":356,"Sac1":77,"Sac2":78,"Sac3":77,"SystemStatus":"OnGrid","Timestamp":"2020-11-10 19:58:16","USOC":0,"Uac":233,"Ubat":47}
[site  ] DEBUG 2020/11/10 19:58:16 battery power: -5W
[cache ] TRACE 2020/11/10 19:58:16 batteryPower: -5
[influx] TRACE 2020/11/10 19:58:16 write batteryPower=-5 (map[])
[site  ] DEBUG 2020/11/10 19:58:16 site power: 1769W
[lp-1  ] DEBUG 2020/11/10 19:58:16 charge power: 1380W
[cache ] TRACE 2020/11/10 19:58:16 mode: minpv
[cache ] TRACE 2020/11/10 19:58:16 chargePower: 1380
[influx] TRACE 2020/11/10 19:58:16 write chargePower=1380 (map[loadpoint:Garage])
[cache ] TRACE 2020/11/10 19:58:16 chargeCurrent: 6
[wallbe] TRACE 2020/11/10 19:58:16 read charge time (102): 00 D8 00 00
[cache ] TRACE 2020/11/10 19:58:16 chargedEnergy: 80.47888771886666
[cache ] TRACE 2020/11/10 19:58:16 chargeDuration: 3m36s
[influx] TRACE 2020/11/10 19:58:16 write chargedEnergy=80.47888771886666 (map[loadpoint:Garage])
[wallbe] TRACE 2020/11/10 19:58:16 read status (100): 00 43
[lp-1  ] DEBUG 2020/11/10 19:58:16 charger status: C
[cache ] TRACE 2020/11/10 19:58:16 connected: true
[cache ] TRACE 2020/11/10 19:58:16 charging: true
[wallbe] TRACE 2020/11/10 19:58:16 read charge enable (400): 01
[lp-1  ] DEBUG 2020/11/10 19:58:16 max charge current: 0A = 6A + -8A (1769W @ 1p)
[lp-1  ] DEBUG 2020/11/10 19:58:16 **target charge current: 6A**
[site  ] DEBUG 2020/11/10 19:58:26 ----
[http  ] TRACE 2020/11/10 19:58:26 GET http://192.168.1.75:8080/api/v1/status

Danke an alle !!!

andig, 2020-11-10:
Scheint als würde Wallbe 2 Controller verkaufen und es wäre kein Unterschied zwischen alt und neu...

closed 2020-11-10 by andig