evcc-io/evcc #422: Keba Box (BMW Wallbox) Log Error

mode2k, 2020-11-11:
**Describe the bug**
Wallbox is working fine.
But every 10s there is this log after starting or stopping the charge:
ERROR 2020/11/11 21:53:23 charge controller error: ena could not enable: TCH-OK :done
When restarting evcc the error does not ocour till a chage starts and stops.

**To Reproduce**
Steps to reproduce the behavior:
Configure Charger like this:
chargers:
- name: keba
  type: keba # Wallbe charger
  uri: 192.168.6.250:7090

- Start evcc
- Start charging
- Stop Chaging

**Expected behavior**
No Error in Log

**EVCC details:**
Show output of `evcc -v`
evcc version 0.31 (feb564d)

Show evcc log output with `--log trace`
[lp-1  ] DEBUG 2020/11/11 22:53:01 charger status: C
[cache ] TRACE 2020/11/11 22:53:01 connected: true
[cache ] TRACE 2020/11/11 22:53:01 charging: true
[keba  ] TRACE 2020/11/11 22:53:01 recv from 192.168.6.250:7090 {
"ID": "2",
"State": 3,
"Error1": 0,
"Error2": 0,
"Plug": 7,
"AuthON": 0,
"Authreq": 0,
"Enable sys": 1,
"Enable user": 1,
"Max curr": 10000,
"Max curr %": 166,
"Curr HW": 16000,
"Curr user": 6000,
"Curr FS": 0,
"Tmo FS": 0,
"Curr timer": 0,
"Tmo CT": 0,
"Setenergy": 0,
"Output": 20,
"Input": 0,
"Serial": "XXXXXXXX",
"Sec": 7878743
}
[keba  ] TRACE 2020/11/11 22:53:01 recv from 192.168.6.250:7090 {
"ID": "3",
"U1": 233,
"U2": 232,
"U3": 232,
"I1": 10209,
"I2": 10162,
"I3": 10187,
"P": 7092318,
"PF": 997,
"E pres": 3401,
"E total": 12121212,
"Serial": "XXXXXXXX",
"Sec": 7878743
}
[lp-1  ] DEBUG 2020/11/11 22:53:01 detected phases: 3p [10.2 10.2 10.2]A
[cache ] TRACE 2020/11/11 22:53:01 chargeCurrents: [10.209 10.162 10.187]
[cache ] TRACE 2020/11/11 22:53:01 activePhases: 3
[keba  ] TRACE 2020/11/11 22:53:01 recv from 192.168.6.250:7090 TCH-OK :done
[keba  ] TRACE 2020/11/11 22:53:01 recv from 192.168.6.250:7090 {
"ID": "2",
"State": 3,
"Error1": 0,
"Error2": 0,
"Plug": 7,
"AuthON": 0,
"Authreq": 0,
"Enable sys": 1,
"Enable user": 0,
"Max curr": 10000,
"Max curr %": 166,
"Curr HW": 16000,
"Curr user": 6000,
"Curr FS": 0,
"Tmo FS": 0,
"Curr timer": 0,
"Tmo CT": 0,
"Setenergy": 0,
"Output": 20,
"Input": 0,
"Serial": "XXXXXXXX",
"Sec": 7878743
}
[lp-1  ] ERROR 2020/11/11 22:53:01 charge controller error: ena could not enable: TCH-OK :done
[keba  ] TRACE 2020/11/11 22:53:01 recv from 192.168.6.250:7090 {"Enable sys": 0}
[keba  ] TRACE 2020/11/11 22:53:01 listener: recv blocked
[keba  ] TRACE 2020/11/11 22:53:01 recv from 192.168.6.250:7090 {"State": 5}
[keba  ] TRACE 2020/11/11 22:53:01 listener: recv blocked
[keba  ] TRACE 2020/11/11 22:53:01 recv from 192.168.6.250:7090 {"Max curr": 0}
[keba  ] TRACE 2020/11/11 22:53:01 listener: recv blocked
[mqtt  ] TRACE 2020/11/11 22:53:01 recv PV/Solaredge/akt_leistung: '0'
[mqtt  ] TRACE 2020/11/11 22:53:02 recv PV/Smartmeter/akt_leistung: '6544.7'
[mqtt  ] TRACE 2020/11/11 22:53:03 recv PV/Smartmeter/akt_leistung: '779.5'
[mqtt  ] TRACE 2020/11/11 22:53:05 recv PV/Smartmeter/akt_leistung: '773.6'
[mqtt  ] TRACE 2020/11/11 22:53:06 recv PV/Solaredge/akt_leistung: '0'
[mqtt  ] TRACE 2020/11/11 22:53:07 recv PV/Smartmeter/akt_leistung: '734.7'
[mqtt  ] TRACE 2020/11/11 22:53:08 recv PV/Smartmeter/akt_leistung: '742.2'
[mqtt  ] TRACE 2020/11/11 22:53:10 recv PV/Smartmeter/akt_leistung: '745.6'
[site  ] DEBUG 2020/11/11 22:53:11 ----
[site  ] DEBUG 2020/11/11 22:53:11 pv power: 0W
[site  ] DEBUG 2020/11/11 22:53:11 grid power: 746W
[site  ] DEBUG 2020/11/11 22:53:11 site power: 746W
[cache ] TRACE 2020/11/11 22:53:11 pvPower: 0
[cache ] TRACE 2020/11/11 22:53:11 gridPower: 745.6
[cache ] TRACE 2020/11/11 22:53:11 mode: off
[keba  ] TRACE 2020/11/11 22:53:11 recv from 192.168.6.250:7090 {
"ID": "3",
"U1": 0,
"U2": 0,
"U3": 0,
"I1": 0,
"I2": 0,
"I3": 0,
"P": 0,
"PF": 0,
"E pres": 3408,
"E total": 12121212,
"Serial": "XXXXXXXX",
"Sec": 7878753
}
[lp-1  ] DEBUG 2020/11/11 22:53:11 charge power: 0W
[cache ] TRACE 2020/11/11 22:53:11 chargePower: 0
[cache ] TRACE 2020/11/11 22:53:11 chargeCurrent: 6
[keba  ] TRACE 2020/11/11 22:53:11 recv from 192.168.6.250:7090 {
"ID": "3",
"U1": 0,
"U2": 0,
"U3": 0,
"I1": 0,
"I2": 0,
"I3": 0,
"P": 0,
"PF": 0,
"E pres": 3408,
"E total": 12121212,
"Serial": "XXXXXXXX",
"Sec": 7878753
}
[cache ] TRACE 2020/11/11 22:53:11 chargedEnergy: 340.8
[cache ] TRACE 2020/11/11 22:53:11 chargeDuration: 50s
[keba  ] TRACE 2020/11/11 22:53:11 recv from 192.168.6.250:7090 {
"ID": "2",
"State": 5,
"Error1": 0,
"Error2": 0,
"Plug": 7,
"AuthON": 0,
"Authreq": 0,
"Enable sys": 1,
"Enable user": 1,
"Max curr": 15000,
"Max curr %": 250,
"Curr HW": 16000,
"Curr user": 15000,
"Curr FS": 0,
"Tmo FS": 0,
"Curr timer": 0,
"Tmo CT": 0,
"Setenergy": 0,
"Output": 20,
"Input": 0,
"Serial": "XXXXXXXX",
"Sec": 7878753
}
[lp-1  ] DEBUG 2020/11/11 22:53:11 charger status: B
[lp-1  ] INFO 2020/11/11 22:53:11 stop charging <-
[cache ] TRACE 2020/11/11 22:53:11 chargeCurrent: 6
[cache ] TRACE 2020/11/11 22:53:11 connected: true
[cache ] TRACE 2020/11/11 22:53:11 charging: false
[keba  ] TRACE 2020/11/11 22:53:11 recv from 192.168.6.250:7090 {
"ID": "2",
"State": 5,
"Error1": 0,
"Error2": 0,
"Plug": 7,
"AuthON": 0,
"Authreq": 0,
"Enable sys": 1,
"Enable user": 1,
"Max curr": 15000,
"Max curr %": 250,
"Curr HW": 16000,
"Curr user": 15000,
"Curr FS": 0,
"Tmo FS": 0,
"Curr timer": 0,
"Tmo CT": 0,
"Setenergy": 0,
"Output": 20,
"Input": 0,
"Serial": "XXXXXXXX",
"Sec": 7878753
}
[keba  ] TRACE 2020/11/11 22:53:11 recv from 192.168.6.250:7090 {
"ID": "3",
"U1": 0,
"U2": 0,
"U3": 0,
"I1": 0,
"I2": 0,
"I3": 0,
"P": 0,
"PF": 0,
"E pres": 3408,
"E total": 12121212,
"Serial": "XXXXXXXX",
"Sec": 7878753
}
[cache ] TRACE 2020/11/11 22:53:11 chargeCurrents: [0 0 0]
[keba  ] TRACE 2020/11/11 22:53:11 recv from 192.168.6.250:7090 TCH-OK :done
[keba  ] TRACE 2020/11/11 22:53:11 recv from 192.168.6.250:7090 {
"ID": "2",
"State": 5,
"Error1": 0,
"Error2": 0,
"Plug": 7,
"AuthON": 0,
"Authreq": 0,
"Enable sys": 1,
"Enable user": 0,
"Max curr": 15000,
"Max curr %": 250,
"Curr HW": 16000,
"Curr user": 15000,
"Curr FS": 0,
"Tmo FS": 0,
"Curr timer": 0,
"Tmo CT": 0,
"Setenergy": 0,
"Output": 20,
"Input": 0,
"Serial": "XXXXXXXX",
"Sec": 7878753
}
[lp-1  ] ERROR 2020/11/11 22:53:11 charge controller error: ena could not enable: TCH-OK :done

**If using Docker:**
No Docker

andig, 2020-11-12:
Es wäre super wenn Du das Issue formatieren könntest (github markdown). Bitte auch alle relevanten Informationen angeben, insbesondere Konfiguration. 

Bzgl. Fehlermeldung: wäre ggf möglich mir zum analysieren Remotezugang zu geben? Bitte unter cpuidle@gmx.de melden. PV-geführte Ladung, start/stop funktioniert sonst?

mark-sch, 2020-11-12:
Hi, konnte den (ähnlichen) Fehler auch reproduzieren. Der Ladebeginn, z.B. Min+PV klappt einwandfrei. Wird danach die Stop Funktion gewählt tritt der Fehler "charge controller error:ena" auf. Danach folgt "listener: recv blocked". Im Trace ist es diese Stelle:

keba  ] TRACE 2020/11/12 14:14:56 recv from 192.168.5.45:7090 {
"ID": "2",
"State": 3,
"Error1": 0,
"Error2": 0,
"Plug": 7,
"AuthON": 1,
"Authreq": 0,
"Enable sys": 1,
"Enable user": 1,
"Max curr": 6000,
"Max curr %": 100,
"Curr HW": 16000,
"Curr user": 6000,
"Curr FS": 0,
"Tmo FS": 0,
"Curr timer": 0,
"Tmo CT": 0,
"Setenergy": 0,
"Output": 20,
"Input": 0,
"Serial": "197xxxxx",
"Sec": 83xxx
}
[keba  ] TRACE 2020/11/12 14:14:56 recv from 192.168.5.45:7090 {
"ID": "3",
"U1": 235,
"U2": 235,
"U3": 239,
"I1": 5782,
"I2": 5811,
"I3": 0,
"P": 2673593,
"PF": 978,
"E pres": 234,
"E total": 187297,
"Serial": "197xxxxx",
"Sec": 83xxx
}
[lp-1  ] DEBUG 2020/11/12 14:14:56 detected phases: 2p [5.78 5.81 0]A
[cache ] TRACE 2020/11/12 14:14:56 chargeCurrents: [5.782 5.811 0]
[influx] TRACE 2020/11/12 14:14:56 write chargeCurrents=[5.782 5.811 0] (map[loadpoint:Ladepunkt])
[cache ] TRACE 2020/11/12 14:14:56 activePhases: 2
[keba  ] TRACE 2020/11/12 14:14:56 recv from 192.168.5.45:7090 TCH-OK :done
[keba  ] TRACE 2020/11/12 14:14:56 recv from 192.168.5.45:7090 {
"ID": "2",
"State": 3,
"Error1": 0,
"Error2": 0,
"Plug": 7,
"AuthON": 1,
"Authreq": 0,
"Enable sys": 1,
"Enable user": 0,
"Max curr": 6000,
"Max curr %": 100,
"Curr HW": 16000,
"Curr user": 6000,
"Curr FS": 0,
"Tmo FS": 0,
"Curr timer": 0,
"Tmo CT": 0,
"Setenergy": 0,
"Output": 20,
"Input": 0,
"Serial": "197xxxxx",
"Sec": 83xxx
}
[lp-1  ] ERROR 2020/11/12 14:14:56 charge controller error: ena could not enable: TCH-OK :done
[keba  ] TRACE 2020/11/12 14:14:56 recv from 192.168.5.45:7090 {"Enable sys": 0}
[keba  ] TRACE 2020/11/12 14:14:56 listener: recv blocked
[keba  ] TRACE 2020/11/12 14:14:56 recv from 192.168.5.45:7090 {"State": 5}
[keba  ] TRACE 2020/11/12 14:14:56 listener: recv blocked
[keba  ] TRACE 2020/11/12 14:14:56 recv from 192.168.5.45:7090 {"Max curr": 0}
[keba  ] TRACE 2020/11/12 14:14:56 listener: recv blocked
[keba  ] TRACE 2020/11/12 14:14:57 recv from 192.168.5.45:7090 {"E pres": 237}
[keba  ] TRACE 2020/11/12 14:14:57 listener: recv blocked
[site  ] DEBUG 2020/11/12 14:14:57 ----
[site  ] DEBUG 2020/11/12 14:14:57 pv power: 279W
[site  ] DEBUG 2020/11/12 14:14:57 consumption power: 2959W
[site  ] DEBUG 2020/11/12 14:14:57 grid power: -16W
[cache ] TRACE 2020/11/12 14:14:57 pvPower: 279
[cache ] TRACE 2020/11/12 14:14:57 consumptionPower: 2959
[site  ] DEBUG 2020/11/12 14:14:57 battery power: 2781W
[cache ] TRACE 2020/11/12 14:14:57 gridPower: -16
[cache ] TRACE 2020/11/12 14:14:57 batteryPower: 2781
[site  ] DEBUG 2020/11/12 14:14:57 battery soc: 62%
[site  ] DEBUG 2020/11/12 14:14:57 site power: 2765W
[cache ] TRACE 2020/11/12 14:14:57 batterySoC: 62
[influx] TRACE 2020/11/12 14:14:57 write batterySoC=62 (map[])

andig, 2020-11-12:
Ich sehe nicht wo der Ladepunkt hier das enable=false auslöst? Wäre Remotezugang möglich?

andig, 2020-11-12:
Ich denke es liegt hier dran https://github.com/andig/evcc/blob/master/charger/keba.go#L198. Nach setzen von enabled=false wird das Ergebnis nochmal nachgelesen da UDP unzuverlässig ist. Nur `EnableUser` ist der UDP setzbar, bei Euch ist aber `EnableSys` auch true. Daher der Fehler.

Die Änderung könntet ihr eben lokal ausprobieren falls ihr compilieren könnt? Mir ist allerdings nicht ganz transparent, wie die Box dann mit dem `EnableSys` umgeht bzw. inwiefern sich das auf den Betriebszustand auswirkt.

mark-sch, 2020-11-12:
Das drüclen von "stop" führt tatsächlich auch nicht zum stoppen der Wallbox - EnableSys spiegelt das wohl wieder, obwohl EnableUser korrekt auf 0 gesetzt wurde. Die Funktion enable wird im Code auch korrekt mit false aufgerufen, habe das gerade mit einer log message getestet. Sehe in der Keba Dokumentation: https://www.keba.com/file/downloads/e-mobility/KeContact_P20_P30_UDP_ProgrGuide_en.pdf

ATTENTION: Some electric vehicles (EVs) do not yet meet the standard requirements and an "ena 0" can lead to an error in the charging station.

Kann es damit zu tun haben bzw. ist das Udp STOP command eine Alternative?

andig, 2020-11-12:
Mit dem Fahrzeug hat das m.E. Nichts zu tun. https://github.com/andig/evcc#keba-preparation habt ihr beachtet? Gibt es sonst etwas das die Box auf „enabled“ zwingen könnte, z.B. gibts evtl. einen DIP Schalter für dauer-enable?

mark-sch, 2020-11-12:
#428

closed 2020-11-13 by andig

mode2k, 2020-11-14:
Problem nach neustart der Wallbox nicht mehr reproduzierbar. Wahrscheinlich hatte die Box sich verschluckt.
PS: Formatierung der Logs als COde hatte ich versucht. Dann sind leider immer die Zeilenumbrüche verloren gegangen. Auch <space> <space> oder \ am Ende haben nicht geholfen.

PR #429 (2020-11-13): Refactor KEBA error checking
Fix #422. Replaces #428.