evcc-io/evcc #423: Flexibility in charge min / max current per vehicle

brunohaller, 2020-11-12:
Let me state before, that I am currently evaluating whether evcc and a new charger could be a solution for me and so I do not have practical experience yet. So please apologize if some of the features exist already and I just didn't find it.

**Problem statement**
Different vehicles have can different charging characteristics. For example the Renault Zoe needs at least 10A and can charge up to 32A on one phase (when ignoring topics like "Schieflast"). While many other cars support the standard 6A - 16A.

So when one wants to charge multiple such vehicles on the same loadpoint, it is not sufficient to have a fixed min and max current, as either the Zoe would not load correctly or the other car would also start from 10A only.

**Describe the solution you'd like**
Ideally there should be a min/max-current configuration on the vehicle level.
evcc would auto-detect which vehicle is connected and adjust accordingly.

**Describe alternatives you've considered**
Other possibilites would be to have a manual switch in the application.

andig, 2020-11-12:
For time being this sounds like a very specific requirement. Will reevaluate at a later time.

brunohaller, 2020-11-12:
That's ok, it's not urgent either. But i would not say it is very specific, the Renault Zoe is a very popular car :-)

premultiply, 2020-11-12:
Current ZOE Versions (R) do charge fine in a range from 6 to 32A on a single phase. So there is no urgent need to change anything related to this.

brunohaller, 2020-11-12:
OK cool, I didn't know that. As our Zoe is the most recent one, probably that problem does not exist at all for me! Thank you

closed 2020-11-12 by andig