evcc-io/evcc #427: URL for Batterystatus for renault is wrong

lorbetzki, 2020-11-12:
Hey,

it has started with a feature request in a forum and ended in bad mood and bughunting ;) . 

The URL in line 300 is not complete, it's very important to put the countrycode at the end. Without this information no data (or error) will be received. 

--- renault.go.org      2020-11-12 17:10:04.591562427 +0100
+++ renault.go  2020-11-12 17:27:00.899546404 +0100
@@ -300,7 +300,7 @@

 // batteryAPI provides battery api response
 func (v *Renault) batteryAPI() (interface{}, error) {
-       uri := fmt.Sprintf("%s/commerce/v1/accounts/%s/kamereon/kca/car-adapter/v2/cars/%s/battery-status", v.kamereon.Target, v.accountID, v.vin)
+       uri := fmt.Sprintf("%s/commerce/v1/accounts/%s/kamereon/kca/car-adapter/v2/cars/%s/battery-status?country=DE", v.kamereon.Target, v.accountID, v.vin)
        res, err := v.kamereonRequest(uri)

        // repeat auth if error

Best Regards
Kristian aka lorbi

premultiply, 2020-11-12:
Mmmh. Works for me as expected...

andig, 2020-11-12:
Schau bitte an wie man einen Fehler beschreibt: beobachtetes Verhalten, erwartetes Verhalten. Was ist hier konrket im Ablauf falsch? Logfile? Konfiguration?

andig, 2020-11-12:
Bitte nicht falsch verstehen, aber das läuft genau so bei verschiedenen Usern.

cathiele, 2020-11-12:
Gibt es bei Renault regionspezifische Endpunkte? Bei Hyundai / Kia ist das ja Unterschiedlich je nach Region.

andig, 2020-11-12:
Ja, beim Token abholen, später nicht mehr.

lorbetzki, 2020-11-12:
Sehr interessant. 

@premultiply 
which model you are driving? 

Ich vemute sehr, das die Modelle unterschiedlich behandelt werden: Ich fahre einen Renault Zoe R135 (Phase2), meine VIN fängt mit VF1 an. 

rufe ich mit Curl den Link **ohne** Countrycode mit meiner Authentification und VIN auf erhalte ich folgendes Array:
(
    [type] => FUNCTIONAL
    [messages] => Array
        (
            [0] => Array
                (
                    [code] => err.func.wired.missingRequestParam
                    [message] => must be present
                    [parameters] => Array
                        (
                            [name] => country
                            [type] => Country
                        )
                    [propertyPath] => country
                )
        )
    [errors] => Array
        (
            [0] => Array
                (
                    [errorCode] => err.func.wired.missingRequestParam
                    [errorMessage] => must be present
                )
        )
    [error_reference] => FUNCTIONAL
)

rufe ich mit Countrycode auf bekomme ich folgendes zurück
(
    [data] => Array
        (
            [type] => Car
            [id] => VF1...
            [attributes] => Array
                (
                    [timestamp] => 2020-11-12T06:57:04Z
                    [batteryLevel] => 62
                    [batteryTemperature] => 20
                    [batteryAutonomy] => 179
                    [batteryCapacity] => 0
                    [batteryAvailableEnergy] => 32
                    [plugStatus] => 0
                    [chargingStatus] => 0
                    [chargingRemainingTime] => 100
                    [chargingInstantaneousPower] => 20,7
                )
        )
)
Ich sehe auch das Werte in der Renault.go nicht enthalten sind, wie z.b "batteryAutonomy" dies entspricht der Restreichweite. 

in beiden fällen sehe ich mit "evcc -l trace" nichts auffäliges. Wer nutzt denn einen Phase 1 und wer Phase 2? So zum gegenprüfen? Kann man die Ausgabe im Trace noch erweitern für die Vehicle?

andig, 2020-11-12:
@lorbetzki zeig uns bitte config und logfile und eine Fehlerbeschreibung.

lorbetzki, 2020-11-12:
Anbei die Config
[evcc.yaml.txt](https://github.com/andig/evcc/files/5533351/evcc.yaml.txt)
und ein Trace, Passwörter, Mail und IDs habe ich entfernt (hoffentlich)
[trace.txt](https://github.com/andig/evcc/files/5533397/trace.txt)
Fehlerbeschreibung: Es werden keine Batterie-Status Daten meiner Zoe R135 Phase 2 ausgelesen.

andig, 2020-11-12:
Ich kann in dem Log nix erkennen. Bitte mal:

Log: error
Levels:
  Renault: trace

Und

    Evcc vehicle -c ....

lorbetzki, 2020-11-13:
die letzte Zeile sieht seht interessant aus:
[trace.txt](https://github.com/andig/evcc/files/5539048/trace.txt)
Dort kommt der  link mit ?country=DE vor
[renault] TRACE 2020/11/13 20:48:54 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/8dd01fe2-0d77-49f1-9917-38c3f3b17496/kamereon/kca/car-adapter/v2/cars/VF1<DELETED>/battery-status?country=DE
Der Trace stammt aber von der unmodifizierten (habe eine extra VM erstellt) version. Es kommen aber **keine** Battery-Status Daten in MQTT an.
Mit der oben genannte Modifikation sieht das trace ähnlich aus, aber da kommen Daten in MQTT an.
[trace-modded-evcc.txt](https://github.com/andig/evcc/files/5539080/trace-modded-evcc.txt)
Wo liegt der Unterschied?

andig, 2020-11-13:
Die trace.txt glaube ich Dir nixht. Hier ist die url: https://github.com/andig/evcc/blob/master/vehicle/renault.go#L303. Kein Query Parameter. Solange ich keine Config, keine Fehlerbeschreibung und kein vollständiges Log mit commit Id bekomme mache ich das Issue hier zu.

premultiply, 2020-11-13:
Es gibt tatsächlich leichte Unterschiede in den Antworten zwischen ZOE Ph1 und ZOE Ph2.
Aber eigentlich nicht in dem relevanten Bereich. SoC ist definitiv bei Beiden gleich.
Die Referenzimplementierung ist pyze.

lorbetzki, 2020-11-13:
Hi,

https://github.com/jamesremuscat/pyze/issues/34

Zitat
> It seems that the country parameter must be specified for all URLs in the latter form.

Ich muss den eintragen sonst kommt bei meiner R135 keine Daten.

andig, 2020-11-14:
> It seems that the country parameter must be specified for all URLs in the latter form.

Mir fehlt immer noch das Logfile aus dem das erkenntlich wäre?

@premultiply was passiert bei Dir wenn Du den Parameter zusätzlich mit dran hängst?

lorbetzki, 2020-11-14:
**Fehlerbeschreibung**
Fahrzeugdaten meines Renault Zoe R135 Bj. 2020 werden nicht zum MQTT Broker published.
Prinzipiell scheinen aber die Daten anzukommen. Zumindet wird beim Laden des Fahrzeuges der aktuelle SOC angezeigt. 

**Reproduzierbar**
- füge in der evcc.yaml einen Mqtt Broker hinzu.
- starte evcc und beobachte am Broker was für Daten ankommen

als Broker dient die kommerzielle Automatisierungssoftware Symcon.

**Erwartetes Verhalten**
Erwarten würde ich, das Fahrzeugdaten meines Renault Zoe R135 Bj. 2020 zum MQTT Broker published werden. 
EVCC/loadpoints/1 wird nicht gefüllt

**Workaraound**
Nach Änderung der Zeile 300 in der Renault.go werden Fahrzeugdaten auch published.

Original
`  uri := fmt.Sprintf("%s/commerce/v1/accounts/%s/kamereon/kca/car-adapter/v2/cars/%s/battery-status", v.kamereon.Target, v.accountID, v.vin)`

Änderung
`  uri := fmt.Sprintf("%s/commerce/v1/accounts/%s/kamereon/kca/car-adapter/v2/cars/%s/battery-status?country=DE", v.kamereon.Target, v.accountID, v.vin)`

Verhalten nach Workaround
Die Fahrzeugdaten meines Renault Zoe R135 Bj. 2020 werden nun zum MQTT Broker published. 
EVCC/loadpoints/1 wird gefüllt mit Daten

**Sonstiges**
Ich kann MQTT nicht mitloggen. Gibt es dazu extra Parameter?
Mit der angefügten Config bekomme ich sowohl unter Ubuntu 20.04 LTS, Ubuntu 18.04 LTS und Raspbian Buster nach Eingabe von ./evcc vehicle -c ... als Ausgabe für den Battery-Status folgende Zeile.
`[renault] TRACE 2020/11/14 13:16:26 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/<ACCOUNTID>/kamereon/kca/car-adapter/v2/cars/VF1.../battery-status?country=DE`
EVCC wurde dabei immer von der aktuellen Github geladen! 

@Premultiply wie sieht bei Dir der trace aus wenn Du in der evcc.yaml 
Levels:
 Renault: trace 
anfügst und ./evcc vehicle -c <pfad zur Config anfügst>?

**Daten**
[evcc.yaml.txt](https://github.com/andig/evcc/files/5540776/evcc.yaml.txt)

@andig 
Ich bin kein Entwickler, ich weiss nicht wie ich an ein vollständiges Log kommen soll, bzw habe das was ich bekommen habe weitegeleitet. Ich weiss auch nicht was du mit einer commit id meinst. 

Da das Issue geschlossen ist, mein Problem aber durch die oben genannte Änderung gelöst ist, lasse ich das so für die Nachwelt stehen. Vielleicht kann jemand das Problem auch verifizieren.

andig, 2020-11-14:
Danke für die Beschreibung. Es wird jetztnklar, dass es nicht am abholen der Daten, sondern am Mqtt liegt. Du hast also an der falschen Stelle gesucht! Zum logging hatte ich oben geschrieben. Bitte anslog noch mqtt: trace setzen.

andig, 2020-11-16:
ping @lorbetzki möchtest du nochmal ein sauberes Log machen? Sonst würde ich hier zu machen.

premultiply, 2020-11-16:
Ich komme momentan auch nicht zum einfach mal so testen.

lorbetzki, 2020-11-18:
@andig 
Hi,

ich würde gerne weiterhelfen, aber trotz der Einstellung in evcc.yaml
`Log: error
Levels:
 mqtt: trace
`
werden nur exakt diese zwei Zeilen ausgegeben
`[mqtt  ] INFO 2020/11/18 20:19:41 connecting evcc-72133203 at mein.hostname.local:1883
[mqtt  ] DEBUG 2020/11/18 20:19:41 mein.hostname.local:1883 connected`
Es werden aber Daten im Broker verarbeitet. Pünktliche alle 10 sek. wird aktualisiert.

andig, 2020-11-20:
> Es werden aber Daten im Broker verarbeitet. Pünktliche alle 10 sek. wird aktualisiert.

Du kannst auf dem Broker auch mittels `mosquitto_sub -h <broker> -p 1883 -t "evcc/#"` lauschen.

andig, 2020-11-21:
Ich mache mal fürs erste zu da kein Fehler erkennbar.

closed 2020-11-21 by andig