evcc-io/evcc #430: Ladevorgang stoppt vorzeitig wenn keine VW We Connect Daten empfangen werden

mark-sch, 2020-11-13:
**Describe the bug**
Die Estimate SoC Funktion ist sehr praktisch, so werden auch zwischen den fünf minütigen API Updates treffsichere SoC Werte des Wagens geschätzt. Bei mir vielen gestern Mittag bei 90% SoC neue Updates über die VW API aus. Der Wagen wurde Abends bei ca. 50% SoC angesteckt, die API zeigte noch den alten Wert von 90% SoC und EVCC machte 30 Min. Soforladen, stoppte dann aber, als bei estimate 100% der Akku auf anscheinend voll geschätzt wurde. Die VW Api gibt weiterhin die 90% zurück (VW Server hängen oder Wagen hat aus irgendeinem Grund keine Funkverbindung). Somit Lade Deadlock, der Wagen ist ca. 60% voll, EVCC hat auf 100% geschätzt und das Sofortladen ist beendet.

**To Reproduce**
Steps to reproduce the behavior:
In der evcc.yaml unter den Loadpoints / soc den Parameter estimate: true konfigurieren (set true to interpolate between api updates) und warte bis die We Connect API ausfällt, oft Sonntags, aber auch sonst sehr häufig bei VW :-|

**Expected behavior**
Als Workaround habe ich estimate deaktiviert. Wahrscheinlich macht es Sinn, die estimate Schätzungen zeitlich zu begrenzen, z.B. auf vehicles / cache + 1 Min. und dann mit den VW API Werten zu überschreiben. Oder nach dieser Zeit den Ladezustand nicht weiter extrapolieren, d.h. Schätzungen max im Zeitraum von cache + 1Min. und dann im letzten Schätzwert verharren, bis neue API Updates kommen.

andig, 2020-11-13:
Der use cae von @premultiply war als Workaround für die ständig ausfallende Renault API gedacht, also nicht nur cache+1m.

Die Grundfrage ist Nutzung estimated SoC für TargetSoC (oder auch minSoC). Ein Nebenaspekt ist Abschaltung bei TargetSoC=100. Diese ist m.E. unsinnig da das Auto da ja von alleine abschaltet. Diesen Punkt würde ich deshalb in der Diskussion abtrennen.

premultiply, 2020-11-13:
Die SoC-Schätzung basiert darauf, dass mit jedem SoC-Update eine lineare Steigung zwischen zwei ganzzahligen SoC-Werten mit genügeng Abstand der Samples berechnet wird.
Je kleiner die gemeldete SoC-Änderung via API ist bzw. je kürzer das Zeitintervall zwischen den eingehenden Werten ist desto ungenauer wird diese Berechnung. Dagegen helfen zwei Maßnahmen:
Entweder SoC-Estimator in der Konfig abschalten da in diesem Fall auch eigentlich unnötig oder das Intervall zwischen den Abfragen bzw. den Cache-Timeout erhöhen. Ein-Prozent-Updates werden vom Estimator ignoriert. Daraus kann man keine sinnvolle Steigung errechnen. Im Zweifelsfall könnte es in diesem Fall sogar sein, dass die Steigung nie angepasst wurde und mit dem Initialwert aus der angegebenen Batteriekapazität weitergerechnet wurde.
Für weitere Optimierungen oder fundierte Vervesserungsvorschläge hinsichtlich des Algorithmus bin ich natürlich offen.

andig, 2020-11-13:
> Für weitere Optimierungen oder fundierte Vervesserungsvorschläge hinsichtlich des Algorithmus bin ich natürlich offen.

Ich denke hier geht es weniger um den Algorithmus sondern mehr um das Fallbackhandling wenn über längere Zeit keine Werte kommen:
- wie verhält sich minSoC ohne SoC?
- wie verhält sich targetSoC ohne SoC?
- wo und wann merken wir, dass diese Situation eingetreten ist?

andig, 2020-11-13:
PS.: noch schlimmer im Fall oben wenn das API verfügbar aber eingefroren ist. Es wäre sinnvoll, dafür eine Erkennung einzubauen, veilleicht mit Timeout?

mark-sch, 2020-11-13:
Genau, das Problem ist aufgetreten weil die VW API eingefroren ist und uralt Werte liefert - ist immer noch so, seit 24h. Wenn das Charging bei 100% nicht stoppt wäre das Deadlock Scenario schon mal durchbrochen, tip tip. Kann EVCC trotzdem erkennen wenn das Laden beendet ist und korrekte Werte an den MQTT Loadpoint / charging = true/false liefern? Sonst würden andere Seiteneffekte entstehen. Will mich in Kürze ran machen diesen MQTT Wert zum Steuern des Entladeverhaltens zu nutzen, in Bezug zum Hybridwechselrichter und dem Haus-Batteriespeicher. Im EVCC Sofort-Lademodus soll der Sunny5-Hybrid-WR die Entladeleistung auf die Grundlast begrenzen und nicht Vollgas Battery2Battery laden. Zumindest nicht unterhalb eines gewissen Haus SoC Levels. Wie ist das bei anderen Hybrid WR gelöst?

andig, 2020-11-13:
> Kann EVCC trotzdem erkennen wenn das Laden beendet ist und korrekte Werte an den MQTT Loadpoint / charging = true/false liefern?

Ja das geht, weil die Wallbox dann von C auf B geht. Kannst Du mal https://github.com/andig/evcc/pull/432 aufprobieren obs Seiteneffekte gibt? Ich kann den aber auch einfach reinschieben...

>  Im EVCC Sofort-Lademodus soll der Sunny5-Hybrid-WR die Entladeleistung auf die Grundlast begrenzen und nicht Vollgas Battery2Battery laden. Zumindest nicht unterhalb eines gewissen Haus SoC Levels.

Interessante Idee, lass uns das gerne interaktiv im Slack Kanal diskutieren (siehe Wiki für Teilnahme).

andig, 2020-11-16:
@premultiply wie wollen wir mit https://github.com/andig/evcc/issues/430#issuecomment-726663211 weiter machen?

premultiply, 2020-11-16:
Mmmh wenn keine neuen Daten (mit neuem Timestamp) kommen sollte die letzte (oder initiale) virtuelle Batteriekapazität eigentlich beibehalten werden und mit dem Chargemeter einfach weitergezählt werden.

andig, 2020-11-16:
Genau das ist ja hier das Problem: API hängt auf 90% aber Auto ist leer :(

premultiply, 2020-11-16:
Wie soll das gehen?
Wenn der API-SOC gleich bleibt erfolgt keine Anpassung der Neu-Schätzung.
Würde bedeuten der Batteriekapazitätswert aus der Konfig ist komplett falsch.

andig, 2020-11-16:
Wenn der SoC gleich bleibt trotz längerem Laden stimmt was nicht, das wäre eine Indikation?

premultiply, 2020-11-16:
Ich glaube hier sind einfach zwei Sachen unglücklich zusammen gefallen:
1. Die API ist während des Ladevorgangs ausgefallen
2. Das Abfrageintervall zu kurz mit der aktivierten SoC-Schätzung so dass hier wahrscheinlich zu Beginn mit 2%-Schritten mit erheblichem Fehler neu geschätzt wurde.

andig, 2020-11-16:
Gut. Nachdem #432 drin ist machen wir mal als wontfix zu. Wenn das häufiger vorkommen sollte müssten wir neu überlegen.

closed 2020-11-16 by andig