evcc-io/evcc #433: PV Mode berücksichtigt Enable Threshold nicht

schenlap, 2020-11-14:
**Describe the bug**
Ich möchte gerne das Laden starten sobald ich 500W einspeise (Grid power = -500). Tatsächlich wird aber erst bei 1380W Einspeisung die Ladung gestartet

**To Reproduce**
Steps to reproduce the behavior:
1. PV Mode
2. Enable Threshold auf -500 setzen
```
loadpoints:
- title: main # name for logging
  mode: pv
  charger:  configurable
  enable: # pv mode enable behavior
    delay: 5s # threshold must be exceeded for this long, 2m
    threshold: -500 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5s # threshold must be exceeded for this long, 2m
    threshold: 3000 # maximum import power (W # MIN = 1380-enable(600) = 780)
  sensitivity: 5 # current raise/lower steps size (default 1A)
  guardduration: 1s # switch charger contactor not more often than this (default 10m)
  maxcurrent: 16 # maximum charge current (default 16A)
  phases: 1 # ev phases (default 3)
```
3. Warten bis mehr als 500W eingespeist werden

**Expected behavior**
PV Ladung startet nach Delay

Das Probel scheint bei
https://github.com/andig/evcc/blob/master/core/loadpoint.go
zu liegen da hier abgebrochen wird.

```
	// in PV mode disable charger if car not charging and minCurrent not possible
	if mode == api.ModePV && lp.status != api.StatusC {
		lp.pvTimer = time.Time{}

		if targetCurrent < lp.MinCurrent {
			lp.log.DEBUG.Printf("target<MinCurrent -> return")
			return 0
		}

		lp.log.DEBUG.Printf("MinCurrent -> return")
		return lp.MinCurrent
	}
```

andig, 2020-11-20:
Magst Du mal diesen ganzen Block mal so ändern:

    targetCurrent = lp.MinCurrent

M.E. ist das eine nette Abkürzung, wird über den Block dahinter aber sowieso sichergestellt. Ganz sicher bin ich aber nicht...

schenlap, 2020-11-21:
geht auch nicht so gut, jetzt fängt er schon zu laden an obwohl noch keine Einspeisung stattfindet. Anbei noch meine config, vielleicht ist da ein Fehler:

schenlap, 2020-11-21:
und noch das diff:

andig, 2020-11-21:
Ich schaue später nochmal. Der Teil war schon immer kniffelig und hat nicht umsonst so viele Tests...

andig, 2020-11-21:
So könnte es gehen. Mir ist momentan nicht ganz klar warum ich diese Abkürzung eingebaut habe. Schau bitte mal obs da irgendwelche unerwarteten Seiteneffekte gibt.

PR #464 (2020-11-27): Honour thresholds in PV mode
Fix #433
closed 2020-11-27 by andig