evcc-io/evcc #436: Charger startet im PV Modus obwohl nicht genügend Produktion der PV vorhanden ist

mpucherSmIT, 2020-11-15:
**Describe the bug**
EVCC ist im Modus PV. Charger ist ein dreiphasiger phoenix em-cp. Die PV produzierte um 12:15 Uhr ca. 2000 W. Davon wurden 1300W direkt verbraucht und mit 700W in die Batterie geladen.

Der phoenix hat trotzdem den Ladevorgang gestartet. Ein wenig wundert mich der hohe Wert site power: 3758W in Zeile 2. Was bedeutet dieser Wert?

**Expected behavior**
Im PV Modus sollte der Ladevorgang erst bei ausreichend PV-Leistung gestartet werden.

**EVCC details:**
Show output of `evcc -v`

Show evcc log output with `--log trace`
[20201115 - Start Ladevorgang im PV Mode bei nicht ausreichender Produktion PV.log](https://github.com/andig/evcc/files/5542277/20201115.-.Start.Ladevorgang.im.PV.Mode.bei.nicht.ausreichender.Produktion.PV.log)

**If using Docker:**
evcc version 0.31 (d6a387a)

andig, 2020-11-15:
Das Log hilft nicht da die Vorgeschichte zum Ladestart um 12:15 fehlt :(. Sitepower ist die Summe auf Netz- und Batteriezähler. Die beiden bestimmen den Überschuss im Haushalt.

mpucherSmIT, 2020-11-15:
Kann das Log vervollständigen. Wie weit zurück brauchst du es?

mpucherSmIT, 2020-11-15:
Hab mal fünf Minuten vorher ergänzt. Reicht das?

[20201115 - Start Ladevorgang im PV Mode bei nicht ausreichender Produktion PV.log](https://github.com/andig/evcc/files/5542617/20201115.-.Start.Ladevorgang.im.PV.Mode.bei.nicht.ausreichender.Produktion.PV.log)

andig, 2020-11-15:
Bitte Config ergänzen

andig, 2020-11-15:
Also:

    [site  ] DEBUG 2020/11/15 12:10:15 pv power: 1885W
    [site  ] DEBUG 2020/11/15 12:10:15 grid power: -10W
    [site  ] DEBUG 2020/11/15 12:10:15 battery power: 1732W
    [site  ] DEBUG 2020/11/15 12:10:15 battery soc: 20%
    [site  ] DEBUG 2020/11/15 12:10:15 site power: 1722W

Site ist die Summe aus PV und Batterie. Du hast also 1,7kW Verbrauch. Charger hat Status A, also kein Auto. Ich würde auf Vorzeichenfehler bei der Batterie tippen, aber der Battery SoC sinkt, sie wird also definitiv entladen. Woher kommt der Riesenverbrauch?

/cc @TheRealKerni vielleicht hast Du eine Idee?
/cc @mclane

andig, 2020-11-15:
Dann:

    [emcp  ] TRACE 2020/11/15 12:15:19 read charge enable (400): 01
    [lp-1  ] DEBUG 2020/11/15 12:15:19 sync enabled state to disable

Dein Charger sollte eigentlich aus sein, ist er aber nicht, daher das "sync". Warum ist er nicht wie erwartet aus? Neustart erfolgt?

    [emcp  ] TRACE 2020/11/15 12:15:19 write charge enable (400) 0: 00 00

...ist erfolgreich.

    [lp-1  ] DEBUG 2020/11/15 12:15:19 max charge current: 0A = 6A + -6A (3758W @ 3p)
    [lp-1  ] DEBUG 2020/11/15 12:15:19 target charge current: 0A

Aufgrund des hohen Bezugs sollte Ladung deaktiviert werden. Die Rechnung stimmt.

Und kurz darauf ist der Charger wieder aktiv:

    [emcp  ] TRACE 2020/11/15 12:15:25 read charge enable (400): 01
    [lp-1  ] DEBUG 2020/11/15 12:15:25 sync enabled state to disable

Hier stimmt was nicht!

mpucherSmIT, 2020-11-15:
Um 12:10:15 Uhr wurde die Batterie mit 1732W entladen, da der Hausverbrauch bei ~ 3500 lag (Splülmaschine, Dörrautomat und Herd).

Da passt wohl wirklich was mit den Vorzeichen und Berchnungen nicht.

Hier die Konfig:

[evcc.dist.yaml.txt](https://github.com/andig/evcc/files/5542957/evcc.dist.yaml.txt)

mclane, 2020-11-15:
Ich hab in der config gesehen, dass beim meter "power" diverse modbus Abfragen zusammenaddiert werden. Das hatte ich auch mal. Wenn die Antworten nicht rechtzeitig eintreffen, stimmt die Überschussrechnung nicht, das könnte solche Effekte hervorrufen.

mpucherSmIT, 2020-11-15:
> Ich hab in der config gesehen, dass beim meter "power" diverse modbus Abfragen zusammenaddiert werden. Das hatte ich auch mal. Wenn die Antworten nicht rechtzeitig eintreffen, stimmt die Überschussrechnung nicht, das könnte solche Effekte hervorrufen.

Das Problem kenne ich aus grafana/influxdb. Aber in der EVCC UI werden die Werte von PV, Netzbezug/-einspeisung und Batterie eigentlich ganz korrekt angezeigt. Zumal die CALC Funktion nur beim PV-Meter eingesetzt wird.

andig, 2020-11-15:
> Da passt wohl wirklich was mit den Vorzeichen und Berchnungen nicht.

Doch, das stimmt alles wenn es den Verbrauch wirklich gab. Mir scheint eher Dein Charger will nciht stoppen :O

mclane, 2020-11-16:
Ist der DIP Schalter 10 in der Phoenix auf on? Wenn nicht, reagiert er nicht auf das Schreiben von Coil 400

mpucherSmIT, 2020-11-16:
@mclane schaut nicht so aus:

[image]

Der Dip 10 ist auch hardwareseitig auf off.

Aus dem Handbuch:
[image]

Das werde ich gleich mal versuchen! Wenn es das war, löst das vermutlich auch #437 

@andig falls es das war, soll ich das im Wiki als Vorbereitungsmaßnahme für die emcp dokumentieren?

mclane, 2020-11-16:
Du hast auch DIP7 auf on. Hast Du da einen Schater dran? DIP 7 würde ich ausschalten, damit er nur auf die Coil 400 reagiert.

mpucherSmIT, 2020-11-16:
Da hängt ein Schlüsselschalter dran, sollte keine Probleme machen.

PR #441 (2020-11-16): Warn when status needs be re-synced
Fix #436, #437 by issuing warning and updating README.
closed 2020-11-16 by andig