evcc-io/evcc #437: STOP hat keinen Effekt (Charger lädt weiter)

mpucherSmIT, 2020-11-15:
**Describe the bug**
Bei aktiver Ladung hat der Stop Button in der UI keinen Effekt. Charger lädt einfach weiter.

[20201115 - Stop.pdf](https://github.com/andig/evcc/files/5542390/20201115.-.Stop.pdf)

**Expected behavior**
Laden soll bei Klick auf Stop in der UI beendet werden.

Show evcc log output with --log trace
[20201115 - Stop Modus hat keinen Effekt.log](https://github.com/andig/evcc/files/5542392/20201115.-.Stop.Modus.hat.keinen.Effekt.log)

If using Docker:
evcc version 0.31 (d6a387a)

mpucherSmIT, 2020-11-15:
Vielleicht hängt das auch mit #436 zusammen? Mir ist in der phoenix Web-UI aufgefallen, dass man kein current mit 0A einstellen kann.

mclane, 2020-11-16:
Ja das ist dasselbe Problem wie #436. DIP Schalter 10 muss auf on stehen damit man über Modbus Register 400 ein- und ausschalten kann. evcc ist da unschuldig -> kein bug!

closed 2020-11-16 by andig