evcc-io/evcc #438: Fragen zu evcc.yaml: cannot create keba charger: listen udp :7090: bind: address already in use

m-a-r-c-o-g-h, 2020-11-15:
=====
uri: 192.168.1.12:7070 # uri for ui

chargers:
- name: keba1
  type: keba
  uri: 192.168.25.11:7090 # KEBA address
  
====

Hallo,
ich versuche gerade evcc ans Laufen zu bekommen, scheitere jedoch kläglich.
Läuft auf einem Raspberry Pi.

Beim Start von evcc kommt folgende Meldung:

cannot create keba charger: listen udp :7090: bind: address already in use

Habt ihr Tipps für mich?

Gruß, Marco

andig, 2020-11-15:
Sieht alles richtig aus. Was sagt `evcc charger -c <config>`?

/cc @mark-sch hast Du eine Idee?

m-a-r-c-o-g-h, 2020-11-15:
evcc charger -c evcc.yaml
liefert
cannot create keba charger: listen udp :7090: bind: address already in use

Habe ich die Frage richtig verstanden?

andig, 2020-11-15:
OK. Kann es sein, dass Du noch ein anderes Programm laufen hast? Ein zweites EVCC? Was sagt lsof?

m-a-r-c-o-g-h, 2020-11-15:
Aktive Internetverbindungen (Nur Server)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      1574/apache2        
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      1536/sshd           
tcp        0      0 0.0.0.0:1883            0.0.0.0:*               LISTEN      1515/mosquitto      
tcp        0      0 0.0.0.0:1884            0.0.0.0:*               LISTEN      1515/mosquitto      
tcp        0      0 0.0.0.0:9001            0.0.0.0:*               LISTEN      1515/mosquitto      
tcp6       0      0 :::22                   :::*                    LISTEN      1536/sshd           
tcp6       0      0 :::1883                 :::*                    LISTEN      1515/mosquitto      
udp        0      0 0.0.0.0:68              0.0.0.0:*                           1487/dhcpcd         
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           303/avahi-daemon: r 
udp        0      0 0.0.0.0:39662           0.0.0.0:*                           303/avahi-daemon: r 
udp        0      0 0.0.0.0:9522            0.0.0.0:*                           1554/python3        
udp6       0      0 :::5353                 :::*                                303/avahi-daemon: r 
udp6       0      0 :::48587                :::*                                303/avahi-daemon: r

andig, 2020-11-15:
Zeig mal bitte Deine komplette config.

premultiply, 2020-11-15:
Braucht evcc vielleicht nur mehr Rechte um den Port öffnen zu dürfen?

cathiele, 2020-11-15:
Oder schauen ob ausversehen an zwei Stellen die 7090 in der Konfiguration verwendet wird.

andig, 2020-11-15:
> Oder schauen ob ausversehen an zwei Stellen die 7090 in der Konfiguration verwendet wird.

Deshalb:

> Zeig mal bitte Deine komplette config.

Bin gespannt :)

m-a-r-c-o-g-h, 2020-11-16:
uri: 192.168.1.12:7070 # uri for ui
interval: 10s # control cycle interval

log: error
levels:
  core: debug
  lp-1: debug
  lp-2: debug

meters:
- name: grid
  type: default
  power:
    type: http # use http plugin
    auth:
      type: basic
      user: entfernt # Discovergy user name
      password: entfernt # password 
    uri: https://api.discovergy.com/public/v1/last_reading?meterId=entfernt # append meter id
    jq: .values.power
    scale: 0.001

chargers:
- name: keba1
  type: keba
  uri: 192.168.25.11 # KEBA address
  
vehicles:
- name: bmwi3
  type: bmw
  title: i3 # display name for UI
  capacity: 40 # kWh
  user: entfernt # user
  password: entfernt # password
  vin: WBMW... # optional
  cache: 5m # cache API response

site:
  title: Home # display name for UI
  meters:
    grid: grid # grid meter
    pv: grid # pv meter
    battery: battery # battery meter
  prioritySoC: 60 # give home battery priority up to this soc (0 to disable)

loadpoints:
- title: Garage # display name for UI
  charger: keba1 # charger
  meters:
   charge: grid # charge meter
  vehicle: bmwi3
  mode: pv
  soc:
    min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 100 # always charge to 100%
    alwaysUpdate: false # set true to update vehicle soc even when disconnected
    estimate: false # set true to interpolate between api updates
    levels: # target soc levels for UI
    - 30
    - 50
    - 80
    - 100
  onDisconnect: # set defaults when vehicle disconnects
    mode: pv # switch back to pv mode
    targetSoC: 100 # charge to 100%
  phases: 3 # ev phases (default 3)
  sensitivity: 1 # current raise/lower step size (default 10A)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 10m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)

mqtt:

influx:

messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
  services:

andig, 2020-11-16:
Bist Du sicher, dass Du diese Config verwendest? Oben hattest Du die URI des KEBA noch anders angegeben?

m-a-r-c-o-g-h, 2020-11-16:
Ja, das war der letzte Stand.
Es macht scheinbar keinen Unterschied ob mit Portangabe oder ohne (:7090).

Jetzt kam auch noch die Meldung: invalid meter: battery

andig, 2020-11-16:
Letzte Idee: meld Dich für Remote Zugang unter cpuidle@gmx.de

andig, 2020-11-20:
Mangels Feedback geschlossen.

closed 2020-11-20 by andig

die4M, 2024-05-17:
Hallo Zusammen. Ich habe das gleich Problem...evcc läuft daher habe ich mich nie darum gekümmert. da nun aber die Phasenumschaltung meiner Easee home nicht mehr geht, muss ich das nun angehen. Ich habe gelesen dass eventuell evcc mehrmals läuft. Woran kann man das sehen? (Ich bin ein Laie für Pis und solche Programmierungen!!!)

 Unter -tulpn wird folgendes ausgegeben:
[image]
Danke Euch :-)

andig, 2024-05-17:
@die4M hier ging es um Keba. Das Thema ist geschlossen.