evcc-io/evcc #439: Go-eCharger Update 40.0 - api amx hinzugefügt

martinez81, 2020-11-16:
Im Okt 2020 wurde ein Update für den Go-eCharger rausgegeben.
Hierbei wurde die API erweitert.
Es kann jetzt der Ladestrom gesetzt werden ohne das der Wert fest ins EEPROM gespeichert wird. Das ist eine gute Veränderung.
Wäre es möglich das in die nächste Version mit einzubinden? Evtl. über eine 2. Go-eCharger Config, so könnte man "alte" und "neue" Firmware benutzten.

[https://www.goingelectric.de/forum/viewtopic.php?p=1390428#p1390428](url)

Gru?

andig, 2020-11-16:
Muss das Update manuell eingespielt werden? Sollten wir auf jeden Fall benutzen!

martinez81, 2020-11-16:
Man kann es wohl auch über die App einspielen. Ich werde es heute Abend testen.

cathiele, 2020-11-16:
Ich habe es vor einiger Zeit über die App einspielen können. Die App war dabei direkt mit dem charger eigenen  WLAN verbunden

andig, 2020-11-16:
Bitte mal testen

martinez81, 2020-11-16:
> Bitte mal testen

Das Update hat problemlos funktioniert. Es hat keine 15s, inkl. Neustart, gedauert.

andig, 2020-11-16:
Und funktioniert der PR auch?

martinez81, 2020-11-16:
> Und funktioniert der PR auch?

Ich hänge gerade das ich den Befehl im Browser nicht zusammen bekomme 🙃
http://ip-GO-eCharger/set?amp=16
geht nicht sollte ja sowieso gehen.

Was muss ich eingeben?

martinez81, 2020-11-16:
Oh Mann, jetzt habe ich es:

http://ip-GO-eCharger/mqtt?payload=amx=16

So ging es! Also amx läuft.

andig, 2020-11-16:
Super. Wenn wir jetzt noch jemanden hätten der es ohne Upgrade testen könnte dann wüssten wir was schief gehen kann. Oder wir schieben es einfach rein???

martinez81, 2020-11-16:
Hmm, ich gehe mal davon aus, dass ein GO-eCharger mit älterer Firmware den amx Befehl einfach nicht annimmt.
Die Frage ist ob man einfach sagt, das EVCC nur mit GO-eCharger ab Firmware 40.0 funktioniert.
Das Update ist per App ja wirklich kein Problem.

cathiele, 2020-11-20:
ich habe die 40er firmware drauf und bekomme mit der version das hier:
`
[lp-1  ] INFO 2020/11/19 19:18:27 car connected
[lp-1  ] INFO 2020/11/19 20:04:30 set target soc: 50
[lp-1  ] ERROR 2020/11/20 10:16:37 charge controller error: amx update failed: 0
[lp-1  ] ERROR 2020/11/20 10:16:47 charge controller error: amx update failed: 0
[lp-1  ] ERROR 2020/11/20 10:16:57 charge controller error: amx update failed: 0
[lp-1  ] ERROR 2020/11/20 10:17:07 charge controller error: amx update failed: 0
[lp-1  ] ERROR 2020/11/20 10:17:17 charge controller error: amx update failed: 0
[lp-1  ] ERROR 2020/11/20 10:17:37 charge controller error: amx update failed: 0
[lp-1  ] ERROR 2020/11/20 10:17:47 charge controller error: amx update failed: 0
[lp-1  ] ERROR 2020/11/20 10:17:57 charge controller error: amx update failed: 0

`
bin wieder auf die vorherige version zurück gegangen.

andig, 2020-11-20:
@cathiele hättest Du mal ein trace log dafür?

@martinez81 funktioniert evcc damit bei Dir?

martinez81, 2020-11-20:
> 
> @martinez81 funktioniert evcc damit bei Dir?

Du hast doch EVCC noch garnicht hochgezogen?! 
Äh, habe ich die Version 0.32 verpasst? Ich habe Version 0.31 darauf.
GO-eCharger auf 40.0

andig, 2020-11-20:
Die Änderung ist aktuell nur im Master Branch

> So ging es! Also amx läuft.

Argh- ich dachte Du hättest den Branch getestet :(

martinez81, 2020-11-20:
> Die Änderung ist aktuell nur im Master Branch
> 
> > So ging es! Also amx läuft.
> 
> Argh- ich dachte Du hättest den Branch getestet :(

Tut mir leid, ich weiß nicht wo und was der Master Branch ist.
Wie komme ich da hin?

Ich habe nur den Befehl im Browser getestet:
http://ip-GO-eCharger/mqtt?payload=amx=16

andig, 2020-11-20:
Ich habs wieder ausgebaut. Für https://github.com/andig/evcc/pull/457 bräuchte ich ein Logfile. @cathiele weiß wie's geht ;)

cathiele, 2020-11-20:
> Ich habs wieder ausgebaut. Für #457 bräuchte ich ein Logfile. @cathiele weiß wie's geht ;)

Jepp, teste ich und liefere den trace.

andig, 2020-11-21:
Sollte jetzt funktionieren- vielen Dank an @cathiele

closed 2020-11-21 by andig