evcc-io/evcc #443: Frage: Automatische Phasenerkennung

userjp123, 2020-11-16:
Hallo,

bei mir klappt die automatische Phasenerkennung nicht. 
Da ich umschalte, ist mir das wichtig.

Ich gehe davon aus, dass das die Ursache ist:
Im config-File sind bei meinem charge-meter die Einträge für power und energy.

  power: Power # default value, optionally override
  energy: Sum # default value, optionally override
  ??????: ??????
  
Wie lauten die Einträge für die Ströme L1/2/3 ?

Ich verwende einen Zähler orno3p. Der kann ja auch die Einzelströme messen.

Vielen Dank.

andig, 2020-11-16:
Das geht mit `type: modbus` nicht, Du brauchst `type: default` analog https://github.com/andig/evcc-config#meter-generisch-mqtt und musst dann jeden Wert einzeln konfigurieren. currents als Liste.

userjp123, 2020-11-17:
OK, das muss ich mir dann mal anschauen.
Zähler orno3p hängt über USB/RS485-Konverter am PC.
Wäre das hier der Weg ? 

```
- type: default
  power:
    type: modbus # use ModBus plugin
    model: orno3p # ORNO OR-WE-516
    id: 1
    register: # manual register configuration
      address: # ???  
      type: holding # ???
      decode: float32s # ???
  energy:
    type: modbus # use ModBus plugin
    model: orno3p # ORNO OR-WE-516
    id: 1
    register: # manual register configuration
      address: # ???  
      type: holding # ???
      decode: float32s # ???
  currents:
    type: modbus # use ModBus plugin
    model: orno3p # ORNO OR-WE-516
    id: 1
    register: # manual register configuration
      address: # ???  
      type: holding # ???
      decode: float32s # ???
```

Wie konfiguriert man _currents als Liste_ ?
Danke schon mal.

andig, 2020-11-17:
Machs mal so:

```
- type: default
  power:
    type: modbus
    model: orno3p
    id: 1
    value: Power
  energy:
    type: modbus
    model: orno3p
    id: 1
    value: Sum
  currents:
  - type: modbus
    model: orno3p
    id: 1
    value: CurrentL1
  - type: modbus
    model: orno3p
    id: 1
    value: CurrentL2
  - type: modbus
    model: orno3p
    id: 1
    value: CurrentL3
```

Du kannst die Register auch manuell konfigurieren, das ist aber nur bei nicht unterstützten Zählern notwendig.

userjp123, 2020-11-17:
Noch nicht ganz:
Ich musste das device noch hinzunehmen. Klar, irgendwo müssen ja die Verbindungsdaten stehen.
```
- name: charge2
  type: default
  power:
    type: modbus
    device: /dev/ttyUSB1
    baudrate: 9600
    comset: "8E1"
    model: orno3p
    id: 1
    value: Power
  energy:
    type: modbus
    device: /dev/ttyUSB1
    baudrate: 9600
    comset: "8E1"
    model: orno3p
    id: 1
    value: Sum
  currents:
  - type: modbus
    device: /dev/ttyUSB1
    baudrate: 9600
    comset: "8E1"
    model: orno3p
    id: 1
    value: CurrentL1
  - type: modbus
    device: /dev/ttyUSB1
    baudrate: 9600
    comset: "8E1"
    model: orno3p
    id: 1
    value: CurrentL2
  - type: modbus
    device: /dev/ttyUSB1
    baudrate: 9600
    comset: "8E1"
    model: orno3p
    id: 1
    value: CurrentL3
```

Beim Start zeigt er schon mal _currents_ an:
```
[lp-1  ] INFO 2020/11/17 18:54:24   charge meter config:
[site  ] INFO 2020/11/17 18:54:24     power ✓
[site  ] INFO 2020/11/17 18:54:24     energy ✓
[site  ] INFO 2020/11/17 18:54:24     currents ✓
```

Aber:
1. Er zeigt 0 W Charge-Leistung in der UI an.
2. Er erkennt die Phasen nicht, hier waren 3 Phasen akiv:
```
[lp-1  ] DEBUG 2020/11/17 19:01:44 max charge current: 4A = 6A + -2A (361W @ 1p)
```

andig, 2020-11-17:
...und verwendest Du den Zähler auch?

userjp123, 2020-11-17:
Ja, der sitzt in der Verteilung. Ich habe den eingebaut um mal sagen zu können, wieviel kWh in mein Auto geflossen sind.
Und evcc nutzt den ja auch als Ladezähler. 
Du meinst wahrscheinlich, dass evcc auch ohne Zähler die Leistung berechnen kann. 
Aber die (im Code schon vorhandene) Phasenerkennung fände ich schon gut, da ich ja sonst jedesmal das config-File anpassen muss.

andig, 2020-11-17:
Zeig doch bitte mal ein Logfile, dann müssen wir nicht raten ;)

userjp123, 2020-11-17:
Kann jetzt nicht mehr mit Auto testen.

Habe das hier ausgeführt. 
Dort sollten bei **charge2** (das ist die obige config mit den strömen) auch die Werte wie bei **charge** angezeigt werden
1W (Wallbox-Standby-Verbrauch) und 27 kWh
```
sudo ./evcc meter
[main  ] INFO 2020/11/17 21:09:59 evcc 0.31 (34e03b3)
[main  ] INFO 2020/11/17 21:09:59 using config file xxxxx

charge2
[modbus] TRACE 2020/11/17 21:09:59 Measurement(0): 0.00
Power: 0W
[modbus] TRACE 2020/11/17 21:09:59 Measurement(0): 0.00
Energy: 0.0kWh
[modbus] TRACE 2020/11/17 21:09:59 Measurement(0): 0.00
[modbus] TRACE 2020/11/17 21:09:59 Measurement(0): 0.00
[modbus] TRACE 2020/11/17 21:09:59 Measurement(0): 0.00
Current L1..L3: 0.0A 0.0A 0.0A

charge
[modbus] TRACE 2020/11/17 21:09:59 modbus: send 01 03 00 1c 00 02 05 cd
[modbus] TRACE 2020/11/17 21:09:59 modbus: recv 01 03 04 3a 83 12 6f 4b 8f
[modbus] TRACE 2020/11/17 21:09:59 Power: 1.00W
Power: 1W
[modbus] TRACE 2020/11/17 21:09:59 modbus: send 01 03 01 00 00 02 c5 f7
[modbus] TRACE 2020/11/17 21:09:59 modbus: recv 01 03 04 41 d8 14 7b 21 17
[modbus] TRACE 2020/11/17 21:09:59 Sum: 27.01kWh
Energy: 27.0kWh
```

userjp123, 2020-11-18:
So passt es jetzt:

```
- name: charge2
  type: default
  power:
    type: modbus
    device: /dev/ttyUSB1
    baudrate: 9600
    comset: "8E1"
    model: orno3p
    id: 1
    scale: 1000
    register: # manual register configuration
      address: 28
      type: holding
      decode: float32
  energy:
    type: modbus
    device: /dev/ttyUSB1
    baudrate: 9600
    comset: "8E1"
    model: orno3p
    id: 1
    register: # manual register configuration
      address: 256
      type: holding
      decode: float32
  currents:
  - type: modbus
    device: /dev/ttyUSB1
    baudrate: 9600
    comset: "8E1"
    model: orno3p
    id: 1
    register: # manual register configuration
      address: 22
      type: holding
      decode: float32
  - type: modbus
    device: /dev/ttyUSB1
    baudrate: 9600
    comset: "8E1"
    model: orno3p
    id: 1
    register: # manual register configuration
      address: 24
      type: holding
      decode: float32
  - type: modbus
    device: /dev/ttyUSB1
    baudrate: 9600
    comset: "8E1"
    model: orno3p
    id: 1
    register: # manual register configuration
      address: 26
      type: holding
      decode: float32
```

log:
```
Nov 18 09:13:55 hpserver evcc[28051]: [lp-1  ] DEBUG 2020/11/18 09:13:55 detected phases: 1p [15 0.01 0.01]A
```

andig, 2020-11-18:
Ich denke Du hast einen Bug gefunden- das sollte auch ohne die Register gehen.

andig, 2020-11-20:
Jetzt sollte es auch mit vereinfachter Angabe ohne die Registerkonfiguration gehen. Willst Du's nochmal testen?

userjp123, 2020-11-20:
Ich weiß nicht wie es eleganter geht. (Ansonsten bitte Hinweis, mach fast nix mit github)
Ich habe diese Änderungen 
https://github.com/andig/evcc/pull/458/files
(von Hand) in die V0.31 übernommen und make ausgeführt.

Ergebnis: funktioniert nicht (charge2)
```
sudo ./evcc meter
[main  ] INFO 2020/11/20 15:51:33 evcc 0.31 (34e03b3)
[main  ] INFO 2020/11/20 15:51:33 using config file /xxxx/evcc/evcc.yaml

charge
[modbus] TRACE 2020/11/20 15:51:33 modbus: send 01 03 00 1c 00 02 05 cd
[modbus] TRACE 2020/11/20 15:51:33 modbus: recv 01 03 04 3a 83 12 6f 4b 8f
[modbus] TRACE 2020/11/20 15:51:33 Power: 1.00W
Power: 1W
[modbus] TRACE 2020/11/20 15:51:33 modbus: send 01 03 01 00 00 02 c5 f7
[modbus] TRACE 2020/11/20 15:51:34 modbus: recv 01 03 04 42 1c ae 14 53 e2
[modbus] TRACE 2020/11/20 15:51:34 Sum: 39.17kWh
Energy: 39.2kWh

charge2
[modbus] TRACE 2020/11/20 15:51:34 Measurement(0): 0.00
Power: 0W
[modbus] TRACE 2020/11/20 15:51:34 Measurement(0): 0.00
Energy: 0.0kWh
[modbus] TRACE 2020/11/20 15:51:34 Measurement(0): 0.00
[modbus] TRACE 2020/11/20 15:51:34 Measurement(0): 0.00
[modbus] TRACE 2020/11/20 15:51:34 Measurement(0): 0.00
Current L1..L3: 0.0A 0.0A 0.0A
```

andig, 2020-11-20:
Du hast git gecloned? Dann:

    git checkout -b fix/modbus-plugin-values origin/fix/modbus-plugin-values

andig, 2020-11-20:
Alteranativ kannst Du mich per SSH auf Dein System einladen (cpuidle@gmx.de)

userjp123, 2020-11-20:
Sieht gut aus: (erst mal ohne Auto dran)

```
charge
[modbus] TRACE 2020/11/20 17:47:20 modbus: send 01 03 00 1c 00 02 05 cd
[modbus] TRACE 2020/11/20 17:47:20 modbus: recv 01 03 04 3a 83 12 6f 4b 8f
[modbus] TRACE 2020/11/20 17:47:20 Power: 1.00W
Power: 1W
[modbus] TRACE 2020/11/20 17:47:20 modbus: send 01 03 01 00 00 02 c5 f7
[modbus] TRACE 2020/11/20 17:47:20 modbus: recv 01 03 04 42 1c ae 14 53 e2
[modbus] TRACE 2020/11/20 17:47:20 Sum: 39.17kWh
Energy: 39.2kWh

charge2
[modbus] TRACE 2020/11/20 17:47:20 modbus: send 01 03 00 1c 00 02 05 cd
[modbus] TRACE 2020/11/20 17:47:20 modbus: recv 01 03 04 3a 83 12 6f 4b 8f
Power: 1W
[modbus] TRACE 2020/11/20 17:47:20 modbus: send 01 03 01 00 00 02 c5 f7
[modbus] TRACE 2020/11/20 17:47:20 modbus: recv 01 03 04 42 1c ae 14 53 e2
Energy: 39.2kWh
[modbus] TRACE 2020/11/20 17:47:20 modbus: send 01 03 00 16 00 02 25 cf
[modbus] TRACE 2020/11/20 17:47:20 modbus: recv 01 03 04 3c a3 d7 0a d9 b6
[modbus] TRACE 2020/11/20 17:47:20 modbus: send 01 03 00 18 00 02 44 0c
[modbus] TRACE 2020/11/20 17:47:20 modbus: recv 01 03 04 00 00 00 00 fa 33
[modbus] TRACE 2020/11/20 17:47:20 modbus: send 01 03 00 1a 00 02 e5 cc
[modbus] TRACE 2020/11/20 17:47:20 modbus: recv 01 03 04 00 00 00 00 fa 33
Current L1..L3: 0.0A 0.0A 0.0A

```

andig, 2020-11-20:
Perfekt, rein damit! Danke für Deine Geduld :)

closed 2020-11-20 by andig

userjp123, 2020-11-20:
Danke ebenfalls.