evcc-io/evcc #444: Fehlermeldung bei Messages mit eMail

userjp123, 2020-11-16:
**Describe the bug**
Ich bekomme eine Fehlermeldung, beim evcc-Start und bei Mode-Umschaltung, wenn ich bei den Messages eine eMail angebe.
Die Mails werden korrekt gesendet !

**To Reproduce**
Config:
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}

  services:
   - type: email  # der Punkt ist ein Minus
     uri: smtp://xxxxxxxxx

**Expected behavior**
Keine Fehlermeldung bei evcc-Start und Mode-Umschaltung.

**EVCC details:**
Fehlermeldung:
[push  ] ERROR 2020/11/15 18:47:23 invalid template for disconnect: wanted: [connectedDuration], got: [chargeCurrent gridPower mode minSoC chargedEnergy targetSoC chargePower chargeDuration availableVersion]

andig, 2020-11-16:
Bitte Posting formatieren: https://guides.github.com/features/mastering-markdown/

userjp123, 2020-11-16:
**Describe the bug**
Ich bekomme eine Fehlermeldung, beim evcc-Start und bei Mode-Umschaltung, wenn ich bei den Messages eine eMail angebe.
Die Mails werden korrekt gesendet !

**To Reproduce**
Config:
```
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}

  services:
   - type: email  
     uri: smtp://xxxxxxxxx
```
**Expected behavior**
Keine Fehlermeldung bei evcc-Start und Mode-Umschaltung.

**EVCC details:**
Fehlermeldung:
```
[push  ] ERROR 2020/11/15 18:47:23 invalid template for disconnect: wanted: [connectedDuration], got: [chargeCurrent gridPower mode minSoC chargedEnergy targetSoC chargePower chargeDuration availableVersion]
```

goebelmeier, 2020-11-16:
Bitte statt
`      msg: Car disconnected after ${connectedDuration}`
folgendes verwenden:
`      msg: Car disconnected after ${chargeDuration}`

userjp123, 2020-11-16:
Habe ich implementiert und sieht gut aus: Fehlermeldung erscheint nicht mehr.
DANKE !

andig, 2020-11-16:
Naja- es kommt drauf an was man will. Tatsächlich scheint es hier einen wunderbaren Bug zu geben.

andig, 2020-11-16:
Der Fehler: `connectedDuration` wird in den Cache geschrieben, da aber noch nicht verarbeitet bevor schon das Messaging anläuft. Synchronisationspunkt hinzu gefügt.

closed 2020-11-16 by andig